micropython/micropython #9546: CC3200 fails to build

iBoundary, 2022-10-07:
CC mods/modwlan.c
mods/modwlan.c:1297:13: error: initializer element is not constant
     .base = mod_network_nic_type_wlan_base,
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
mods/modwlan.c:1297:13: note: (near initialization for 'mod_network_nic_type_wlan.base')
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
../../py/mkrules.mk:83: recipe for target 'build/LAUNCHXL/release/mods/modwlan.o' failed

micropython version：V1.9

jimmo, 2022-10-07:
Please see the new issue template and include all the relevant information. Compiler version would be helpful.

> micropython version：V1.9

That's a very old version of MicroPython. Do you mean v1.19.1? But judging by the Build-Troubleshooting link I think it's actually the latest, not v1.19 at all.

iBoundary, 2022-10-07:
> Please see the new issue template and include all the relevant information. Compiler version would be helpful.
> 
> > micropython version：V1.9
> 
> That's a very old version of MicroPython. Do you mean v1.19.1? But judging by the Build-Troubleshooting link I think it's actually the latest, not v1.19 at all.

I wrote it wrongly.I got an error when compiling in the latest version.(Error： mciropython version v1.19.1-528)
I found a different definition(cc3200/mods/modwlan.c)：STATIC MP_DEFINE_CONST_OBJ_FULL_TYPE

robert-hh, 2022-10-07:
No problem here. I call it with:

`make BTARGET=application`

Which platform are you working on (Windows, Linux, Mac) and which version of the gcc compiler do you use?

iBoundary, 2022-10-07:
> No problem here. I call it with:
> 
> `make BTARGET=application`
> 
> Which platform are you working on (Windows, Linux, Mac) and which version of the gcc compiler do you use?
Ubuntu
gcc version 7.5.0

micropython version : latest - v1.19.1 - 528 , I compiled successfully with another version(version v1.19.1 - 135，July)

jimmo, 2022-10-07:
OK, I'm guessing this older version of GCC can't figure out that this is actually a constant. It definitely works on newer versions. This particular code is the only place left where we do this in the whole codebase, and I'm in the middle of removing it anyway.

GCC 7.5.0 is pretty old, can you update? Otherwise stick with the 1.19.1 tag for now.

iBoundary, 2022-10-07:
OK, I'll try to upgrade it. Thank you very much.

jonnor, 2024-09-01:
Assuming that the problem was fixed by new compiled. Please reopen with details if not so.

closed 2024-09-01 by jonnor