micropython/micropython #9566: samd: PWM requires duty set before frequency, unlike other ports

scruss, 2022-10-10:
* *Firmware file name*: `SEEED_WIO_TERMINAL-20221006-unstable-v1.19.1-528-gb8982ec5f.uf2`
* *Git commit hash*: gb8982ec5f
* *Port*: samd
* *Board*: [Wio Terminal - Seeed Studio](https://www.seeedstudio.com/Wio-Terminal-p-4509.html)
* *Version information*: `MicroPython v1.19.1-528-gb8982ec5f on 2022-10-06; Wio Terminal D51R with SAMD51P19A`

**Issue**: setting `machine.PWM.freq()` before `machine.PWM.duty_u16()` produces no output.

**Example**

```py3
from time import sleep_ms
from machine import Pin, PWM
pwm = PWM(Pin("BUZZER_CTR", Pin.OUT))
cmaj = [262, 294, 330, 349, 392, 440, 494, 523]

for note in cmaj:
    print(note)
    pwm.freq(note)
    pwm.duty_u16(16383)
    sleep_ms(225)
    pwm.duty_u16(0)
    sleep_ms(25)
```

**Expected result**: play a C Major scale on the built-in buzzer. Equivalent code on rp2 port has expected result.

**Actual result**: only the first note is played

**Workaround**: swapping the frequency and duty lines causes the whole scale to play on SAMD51:

```py3
    pwm.duty_u16(16383)
    pwm.freq(note)
```

This workaround should not be necessary for consistency with other ports.

robert-hh, 2022-10-10:
Thanks for the report. The problem is the duty_u16(0), which is executed in the second loop run before the pwm.freq() call. The new PR  #9556 version runs fine with the test script. Pre-built images are here: https://github.com/robert-hh/Shared-Stuff/tree/master/samd_firmware

robert-hh, 2022-10-10:
Inserting a short sleep like sleep_ms(5) between pwm.freq() and pwm.duty_16() works as well. Seems that the duty cycle must not be updated too fast again. When duty_u16 comes first, it is update together with the frequency change.  If the code waits for the current period to finish, then it works. But the current period may be up to second long. So it can wait either after the freq has been set, or before the duty cycle is changed. In any case, waiting is boring.

robert-hh, 2022-10-10:
A similar problem happens if consecutive frequency or duty cycle changes happen too fast, like in less than a period time. So I have a fix - internal wait for the end of a cycle -  but I do not like it. At 1 Hz, this may last up to 1 sec, during which time at least the scheduler should have a chance to run. And unconditional wait has the faint smell of lock-up. Hmmm.

ricksorensen, 2022-10-11:
On a XIAO I tried a few things.... but was not able to change duty cycle on it's own.  I noticed that in my  early c++ code I had lots of checks on SYNCBUSY for the TCCs- I tried putting some in the duty cycle update to no avail yet.  Here is a REPL log - I am curious why the get duty call returns funny values (e.g. 0 for duy_u16) but as soon as frequency is reset the desired value  appears.   Seems like the write to CCB may not be completing  -- and not just because of time 

( I am using samd_improve branch.  Using A2/D2 with expansion board on XIAO.  Buzzer is A3)
```
>>> 
MPY: soft reboot
MicroPython v1.15-2005-g3c3eb5cb4-dirty on 2022-10-10; Seeed Xiao with SAMD21G18A
Type "help()" for more information.
>>> from machine import Pin, PWM
>>> pwm = PWM(Pin("A2_D2"), freq=1000, duty_u16=3277)
>>> pwm.freq()
1000
>>> pwm.duty_u16()
9
>>> pwm.duty_ns()
50000
>>> pwm.duty_u16(32768)
>>> pwm.freq()
1000
>>> pwm.duty_u16()
0
>>> pwm.duty_ns()
174762666
>>> pwm.freq(pwm.freq())
>>> pwm.duty_u16()
93
>>> pwm.duty_ns()
500000
>>> 

```

robert-hh, 2022-10-11:
Hello @ricksorensen Should happen on the standard branch as well. That seems to be a different topic. The values are retrieved from the unbuffered registers. The other aspect: as long as PWM is not started, reading back registers always returns 0, besides freq(), which returns 48_000_000.

I tried the syncbusy flag as well, but that's way faster than the sequential calls to freq() or duty_u16(). The PWM runs at 48MHz. A bus sync takes worst case just a few clock cycles, so <100ns. The execution of a python function call takes a few µs, so the bus sync has happened since long at the next call of a MP function.

That's all around this buffered update of registers, which is made to have a glitch free frequency change. This PWM drive me crazy!

So the state I have now that fast sequential calls of freq(), duty_xx() are handled well. And the duty_xx calls always report the last value that has been set, whether the device is running or not. freq() will report 0 when the device is stopped, and the actual set freq when it is running, even when the duty cycle is set to 0 and there is no output  pulse.

robert-hh, 2022-10-11:
@ricksorensen @scruss The updated samd_improve branch has fixes for that now.

ricksorensen, 2022-10-11:
@robert-hh , @scruss the new update on samd_improve for PWM looks good.  I managed to hear the annoyting scale  on a XIAO with expansion board (D3) as well as a WIO Terminal  (BUZZER).   Nice addition with `wait_for_register_update`, and stopping PWM on `freq <= 0`    My rather slow logic analyzer captures show good frequencies, duty cycles, and pauses. ,,, Thanks again

robert-hh, 2022-10-11:
Thanks for verifying.

robert-hh, 2022-10-11:
Just a hint: at most places where a Pin object is expected, you can also just use the string or number associated with a Pin. So instead of:
`pwm=PWM(Pin("A2_D2"), freq=1000)`
you can write:
`pwm=PWM("A2_D2", freq=1000)`

scruss, 2022-10-12:
how likely are we to see most of the features of samd_improve in main soon? I am wary of recommending forks to people

robert-hh, 2022-10-12:
I had the impression that Damien wanted to integrate that PR soon, since he asked me to remove some commits which would require more discussions and hold up merging.

closed 2022-12-15 by robert-hh