micropython/micropython #9570: Socket gets stuck while sending data on esp8266

stas-sl, 2022-10-11:
Hi!

I originally noticed this issue while using WebREPL. I was running loops printing sensor values and after a while (few minutes) it got stuck. It was not just one time - it reproduced over and over. So, in order to find the issue, I tried to create minimal examples using sockets and the issue reproduced for them as well.  If you run following examples below for few minutes (usually 2-3), they will hang (or at least for me they do). Sometimes they hang for several seconds (upto 30) and then continue, but afterwards they hang completely. If I use non-blocking sockets it raises EAGAIN error on server. The issue occurs for both sync and async server versions.

Also I thought that maybe my wifi is unstable, but I rewrote same example on C++ (Arduino IDE) and it worked perfectly for long time without any delays and freezes.

**server_async.py** (run on esp)
```python
import uasyncio

port = 3000

async def server(reader, writer):
  cnt = 0
  while True:
    print(cnt)
    await writer.awrite(f'{cnt}\n')
    cnt += 1
    await uasyncio.sleep_ms(100)
  await writer.aclose()

loop = uasyncio.get_event_loop()
loop.create_task(uasyncio.start_server(server, "0.0.0.0", port))
loop.run_forever()
loop.close()
```
or
**server_sync.py** (run on esp)
```python
import time
import socket

port = 3000

server = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
server.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

ai = socket.getaddrinfo("0.0.0.0", port)
addr = ai[0][4]

server.bind(addr)
server.listen(1)

while True:
  client, remote_addr = server.accept()
  client.setblocking(False)
  cnt = 0

  while True:
    print(cnt)
    client.sendall(f'{cnt}\n')
    time.sleep_ms(100)
    cnt += 1

server.close()
```

**client.py** (run on PC)
```python
import time
import socket

socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
socket.connect(('192.168.100.6', 3000))

response = ''
prev = time.time()
while True:
    recv = socket.recv(1024)
    print('duration:', int((time.time() - prev) * 1000), 'ms')
    print(recv.decode(), end='')
    prev = time.time()

socket.close()
```

Version: MicroPython v1.19.1 on 2022-06-18; ESP module with ESP8266
Firmware: esp8266-20220618-v1.19.1.bin

dpgeorge, 2022-10-11:
Can you please share the Arduino/C++ code you used as a comparison, which did work?

stas-sl, 2022-10-11:
Here is the C++ version:

**server.ino**
```cpp
#include <ESP8266WiFi.h>

#define MODE "STA"
#define STASSID "..."
#define STAPSK "..."
#define APSSID "ESP"
#define PORT 3000

WiFiServer server(PORT);
WiFiClient client;

void setup() {
  Serial.begin(115200);

  if (MODE == "STA") {
    WiFi.mode(WIFI_STA);
    WiFi.begin(STASSID, STAPSK);

    Serial.printf("\nConnecting to %s\n", STASSID);
    while (WiFi.status() != WL_CONNECTED) {
      Serial.print('.');
      delay(500);
    }
    Serial.printf("connected, address = %s", WiFi.localIP().toString().c_str());
  } else if (MODE == "AP") {
    Serial.print("Configuring access point...");
    WiFi.softAP(APSSID);
    Serial.printf("AP IP address: %s", WiFi.softAPIP().toString().c_str());
  }

  server.begin();
}

void loop() {
  static uint32_t cnt = 0;
  static char buf[100];

  if (server.hasClient()) {
    client = server.available();
    Serial.println("New client");
  }

  if (client.availableForWrite()) {
    sprintf(buf, "%u\n", cnt);
    Serial.printf("%u\n", cnt);
    client.write(buf, strlen(buf));
    cnt++;
  }

  delay(100);
}
```

I performed some more tests with Arduino/micropython varying AP/STA WiFi modes and here is what I found. The issue seems to occur only using micropython+STA mode. If running micropython+AP mode, both async and sync servers as well as WebREPL work well for long time (at least 10 minutes). So it might seem like there might be something wrong with my WiFi, or combination of micropython+my WiFi. 

At the same time, as I said, I tested similar Arduino code above and it worked without major delays for both STA and AP modes. Though, I observed some delays (2-3 seconds at max) quite often (like every 15-20 seconds), but it never got stuck for longer than 3 seconds. And those delays occurred almost identically for STA/AP modes, so I didn't see any difference in timings/delays between those modes using Arduino implementation.

I also observed another difference between Arduino/micropython. When using Arduino, 95% of all time, if there are no major delays, the client gets each message individually, even if I'm not calling explicitly `client.flush()` after `client.write()`. Though if there some delays, it might get 10-20 messages at once. However when using micropython, messages are usually grouped to 2-3, despite I tried to use `client.sendall()` or `writer.drain()`. This is not a big deal for me, unless micropython stucks when in STA mode.

Here I'm attaching typical outputs when connecting to Arduino/micropython servers, so you be able to see the difference how messages being grouped.

<details>
<summary>Typical client output when connecting to Arduino STA/AP</summary>

````
dur: 28 ms
3168
dur: 99 ms
3169
dur: 109 ms
3170
dur: 112 ms
3171
dur: 80 ms
3172
dur: 224 ms
3173
dur: 2 ms
3174
dur: 76 ms
3175
dur: 101 ms
3176
dur: 103 ms
3177
dur: 108 ms
3178
dur: 90 ms
3179
dur: 106 ms
3180
dur: 101 ms
3181
dur: 99 ms
3182
dur: 258 ms
3183
dur: 57 ms
3184
3185
dur: 93 ms
3186
dur: 106 ms
3187
dur: 138 ms
3188
dur: 53 ms
3189
dur: 99 ms
3190
````
</details>

<details>
<summary>
Typical client output when connecting to Arduino STA/AP
</summary>

```
dur: 298 ms
11
12
13
dur: 270 ms
14
15
dur: 4 ms
16
dur: 530 ms
17
18
dur: 2 ms
19
20
21
dur: 592 ms
22
23
24
dur: 3 ms
25
26
27
dur: 321 ms
28
29
30
dur: 274 ms
31
32
33
dur: 401 ms
34
35
dur: 12 ms
36
37
dur: 135 ms
38
dur: 300 ms
39
40
41
dur: 609 ms
42
43
44
dur: 9 ms
45
46
47
dur: 340 ms
48
49
50
dur: 314 ms
51
52
53
dur: 341 ms
54
55
56
```
</details>

stas-sl, 2022-10-11:
I might be wrong, but my current understanding is that my network is not perfect, maybe some packages get lost, maybe something else, I can't say for sure. And for both Arduino and micropython it results in delays. Even occasional 1-3 seconds delays for Arduino seems like a lot for local network, when my router is 5 meters away. But somehow Arduino recovers from it quite quickly, while micropython hangs forever. Maybe some default socket options are configured differently?

stas-sl, 2022-10-12:
I captured traffic with Wireshark, maybe it might give some hints.

First let's look at Arduino server.

I captured traffic for about 150 seconds. You can see there quite often 20 packets/second - 10 inbound and 10 outbound sent every 100ms. TCP errors are displayed as red bars. Unfortunately those include ALL errors for all WiFI traffic, not only belonging to this TCP conversation, but some of them are indeed from it. For this particular case, there were about 10-15 of them during 150 seconds. However after short pause (about 1 sec) transmission recovers and goes on.

[image]

[image]

This is what TCP errors look like. They all are "TCP Spurious retransmission". From Time column, you can see that it takes about one second to send those repeating packages. And the very next package has bigger size, because it contains all messages combined that server produced during this 1 second delay.

[image]

Let's now look at micropyhon server.

Here I captured traffic for about 1 minute. You can see long delay after at 15 seconds for about 10 seconds. After which traffic continued, but after 45 seconds it stuck completely. Unlike Arduino server here we see that usually we don't have 20 packets/second, but on average only about 10. That is because, as I noticed before, micropython groups messages into groups of 2-3 and sends them together in one packet.

[image]

[image]

Here is that long pause from 13 till 27 second. There are no errors, just long delay for some reason. After that goes quite large packet containing all messages produced during this time.
[image]

And the most interesting observation is despite you can see some red error bars, they all are not from this conversation. So, unlike Arduino, I almost never see retransmission errors with micropython. I've seen it only one time. So this could confirm my previous guess, that both Arduino and micropyhon face same packet losses, Arduino is able to recover from it quickly by sending the packet one more time, but micropython stucks and waits. That long pause probably could be exactly that case, when the packet finally arrived, however not all packets that lucky, and probably instead of waiting them for such long time, micropython should try to resend them quicker.

stas-sl, 2022-10-12:
Running same experiments in AP mode:

Arduino:
[image]

micropython:
[image]

In AP mode there was not a single retransmission error. So both Arduino & micropython were running smoothly without any delays.

Also I tried to host the server on a regular Ubuntu server in the internet (connected via my WiFi router) and I also haven't observed any retransmission errors in that case.

Unfortunately I don't have other devices in my local network where I could deploy the server. But looks like it is combination of my esp8266 chip + my WiFi + micropython that leads to the issue. In order to determine source of the errors I would need another WiFi router, other esp8266 chips or regular servers in my local network, but I don't have them at the moment. The only thing that I could compare is Arduino/micropython on the same chip, same network, and from that comparison, as I said already, you could see that Arduino can recover from those errors, whereas micropython struggles. Why those errors appear at all - that is another question and I can't tell it at the moment.

stas-sl, 2022-10-13:
I performed some more tests and I should correct a bit my previous observations.

1. When using micropython in STA mode, I actually do see TCP retransmissions and seems like it is able to survive them and keep running after they occur. So there must be some other reason why it stucks.
2. The reason why I didn't see retransmissions in AP mode was because my chip was half a meter away from my laptop, but as I moved it to another room, I start to see them as well. But still it didn't stuck a single time.

stas-sl, 2022-11-12:
Checked this on esp32 (MicroPython v1.19.1 on 2022-06-18; 4MB/OTA module with ESP32) and it works without issues. Also I noticed that it doesn't group 2-3 packets into one (unlike esp8266) - which is consistent with Arduino implementation. 

So seems like the issue is only with esp8266 boards.