micropython/micropython #9597: cyw43: Can't set multiplexed pins (e.g. LED) from second core.

jimmo, 2022-10-13:
It appears that cyw43_ll hangs in cyw43_do_ioctl waiting for a response from the cyw43, if invoked from the second core.

Originally reported here: https://github.com/orgs/micropython/discussions/9595

```py
from machine import Pin
import _thread
import time

led = Pin('LED', Pin.OUT)

# To confirm that this isn't related to cyw43 init, first toggle it from the main core.
led.high()
led.low()

def led_thread():
    print("turning on led")
    led.high() #           <------------------- This never returns
    print("led is on")

print("main")
time.sleep_ms(500)
_thread.start_new_thread(led_thread,())
time.sleep_ms(1000)
```

iabdalkader, 2022-10-27:
I'm almost sure this happens because `__WFI()` in CYW43 delay (and in wait on IOCTL). The second core doesn't have any IRQs enabled (and no handlers installed), so when it executes `__WFI()` it never wakes up. I already have a fix (can toggle the LED in a loop in the thread) but it seems to make the processor uninterruptible, will give it another go today. FWIW here's the fix:

```diff
diff --git a/ports/rp2/mpthreadport.c b/ports/rp2/mpthreadport.c
index 252f7b01f..42b3c3ef8 100644
--- a/ports/rp2/mpthreadport.c
+++ b/ports/rp2/mpthreadport.c
@@ -108,6 +108,12 @@ STATIC void core1_entry_wrapper(void) {
     // Allow MICROPY_BEGIN_ATOMIC_SECTION to be invoked from core0.
     multicore_lockout_victim_init();
 
+    // Configure and enable SysTick interrupt to wake-up the core
+    // from __WFI(). Note, i bit is disabled so the core doesn't
+    // jump to handlers, since none are installed.
+    __asm volatile ("cpsid i");
+    SysTick_Config(clock_get_hz(clk_sys)/1000/1000);
+
     if (core1_entry) {
         core1_entry(core1_arg);
     }
```

dpgeorge, 2022-11-09:
Fixed by 345f1d23957be43e1157463a6405ba9cb04bd43d

closed 2022-11-09 by dpgeorge