micropython/micropython #9614: Enable AES CTR mode for Pico W

realvarx, 2022-10-14:
#### Please include a clear and concise description of what the bug is, the expected output, and how to replicate it:
AES CTR mode isn't working on rp2-pico-w builds

#### If you are seeing code being executed incorrectly, please provide a minimal example and expected output:
Try with the code from [this test](https://github.com/micropython/micropython/blob/89b320737652829edbab921e86d7ad3962d86d9e/tests/extmod/ucryptolib_aes128_ctr.py) from this **micropython** repo

Code:
```
import ucryptolib

# 6 is Counter mode (CTR)
crypto = ucryptolib.aes(b"1234" * 4, 6, b"5678" * 4) 
```

Output:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: mode
```

#### Please provide as much information as possible about the version of MicroPython you're running, such as:
 - firmware file name: 
        rp2-pico-w-20221011-unstable-v1.19.1-543-gab317a0d6.uf2 (neither worked with an end of july version)
 - version information shown in the REPL (hit Ctrl-B to see the startup message) : 
        MicroPython v1.19.1 on 2022-10-11; Raspberry Pi Pico W with RP2040

jimmo, 2022-10-14:
This is enabled via the `MICROPY_PY_UCRYPTOLIB_CTR` config option. The only place this is enabled is for the coverage tests (i.e. the coverage configuration for the unix port).

If you want to compile your own firmware, you can add `#define MICROPY_PY_UCRYPTOLIB_CTR (1)` to `ports/rp2/boards/pico_w/mpconfigboard.h` and add `#define MBEDTLS_CIPHER_MODE_CTR` to `ports/rp2/mbedtls/mbedtls_config.h`.

I don't have a good feel for whether this is something that should be enabled by default on more boards. It costs 488 bytes to enable it on Pico W.

jimmo, 2022-10-14:
https://github.com/micropython/micropython/tree/master/ports/rp2#readme
and
https://docs.micropython.org/en/latest/develop/gettingstarted.html

(Use the same instructions as for the stm32 in terms of installing the same arm-none-eabi-gcc compiler)

realvarx, 2022-10-14:
Thank you! I thought it wasnt possible to do it

closed 2022-10-14 by realvarx

brycethorup, 2026-10-28:
Adding a note for anyone else coming across this, the correct define is now `MICROPY_PY_CRYPTOLIB_CTR` not `MICROPY_PY_UCRYPTOLIB_CTR`