micropython/micropython #9634: Raspberry Pico W MicroPython execution freezes a few seconds after disconnecting screen from UART

robert-hh, 2022-10-15:
### Discussed in https://github.com/orgs/micropython/discussions/9633

<div type='discussions-op-text'>

<sup>Originally posted by **cirosantilli** October 15, 2022</sup>
I got my program running fine as explained at: https://stackoverflow.com/questions/66183596/how-can-you-make-a-micropython-program-on-a-raspberry-pi-pico-autorun/74078142#74078142

I'm installing a `main.py` that does:

```
import machine
import time

led = machine.Pin('LED', machine.Pin.OUT)
# For Rpi Pico (non-W) it was like this instead apparently.
# led = Pin(25, Pin.OUT)

i = 0
while (True):
    led.toggle()
    print(i)
    time.sleep(.5)
    i += 1
```

When I power the device on by plugging the USB to my laptop, it seems to run fine, with the LED blinking.

Then, if I connect from my laptop to the UART with:

```
screen /dev/ttyACM0 115200
```

I can see the numbers coming out on my host terminal correctly, and the LED still blinks, all as expected.

However, when I disconnect from screen with Ctrl-A K, after a few seconds, the LED stops blinking! It takes something around 15 seconds for it to stop, but it does so every time I tested.

If I reconnect the UART again with:

```
screen /dev/ttyACM0 115200
```

it starts blinking again.

Also also noticed that after I reconnect the UART and execution resumes, the count has increased much less than the actual time passed, so one possibility is that the Pico is going into some slow low power mode?

Micropython firmware: rp2-pico-w-20221014-unstable-v1.19.1-544-g89b320737.uf2, Ubuntu 22.04 host.

Also asked at:

- https://stackoverflow.com/questions/74081960/raspberry-pico-w-micropython-execution-freezes-a-few-seconds-after-disconnecting</div>

robert-hh, 2022-10-15:
Created from the discussion #9633 with more details.

robert-hh, 2022-10-16:
This situation happens with all ports using USB CDC. So this is the piece of code for RP2, where it blocks. It looks similar in other ports,
```
// Send string of given length
void mp_hal_stdout_tx_strn(const char *str, mp_uint_t len) {
    #if MICROPY_HW_ENABLE_UART_REPL
    mp_uart_write_strn(str, len);
    #endif

    #if MICROPY_HW_ENABLE_USBDEV
    if (tud_cdc_connected()) {
        for (size_t i = 0; i < len;) {
            uint32_t n = len - i;
            if (n > CFG_TUD_CDC_EP_BUFSIZE) {
                n = CFG_TUD_CDC_EP_BUFSIZE;
            }
            while (n > tud_cdc_write_available()) { 
                MICROPY_EVENT_POLL_HOOK    
            }
            uint32_t n2 = tud_cdc_write(str + i, n);
            tud_cdc_write_flush();
            i += n2;
        }
    }
    #endif

    #if MICROPY_PY_OS_DUPTERM
    mp_uos_dupterm_tx_strn(str, len);
    #endif
}
```

It is the `while (n > tud_cdc_write_available()) ` loop, where it hangs. So the question is now, how to proceed. At the STM32 port there is a timeout at that place in the code logic, skipping the send after about 500ms. So we have two options:

1. Adding a small timeout here and skip the transfer on timeout. The timeout has to be chosen based on the time, MICROPY_EVENT_POLL_HOOK will wait if nothing happens.
2. Skip the transfer if data cannot be sent without waiting.

I prefer option 1 to avoid data loss in regular operation.

Note: 
1. Ports using serial transfer like ESP32 usually just send and do not care, if data can be received.
2. If USB is not connected, that data is dropped anyhow.

cirosantilli, 2022-10-16:
BTW, reproduction is much easier/faster by printing out more data to stdout faster:

```
import machine
import time

led = machine.Pin('LED', machine.Pin.OUT)

i = 0
while (True):
    led.toggle()
    print('asdf ' * 10 + str(i))
    time.sleep(.1)
    i += 1
```

robert-hh, 2022-10-16:
I made a PR to add a timeout. See PR #9638.

dpgeorge, 2022-11-09:
Fixed by 8447fef9f9d0db8104d0e600b79870c4aafa9725

closed 2022-11-09 by dpgeorge