micropython/micropython #9635: f-string comma parsed differently than python3.9

jepler, 2022-10-15:
A tuple without enclosing parens can be printed by f-string in Python 3.9:
```
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> f"{1,=}"
'1,=(1,)'
```

In MicroPython, it is a SyntaxError:
```
MicroPython v1.19.1-544-g89b320737-dirty on 2022-10-15; linux [GCC 10.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> f"{1,=}"
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

jimmo, 2022-10-19:
Thanks @jepler 

This confused me initially because we don't do anything special with commas, but of course what's happening is the f-string transform does:

`f"{1,=}"` --> `"1,={}".format(1,,)`

So it would be worse if we didn't lazily include the trailing comma in the format args, because it would be incorrect rather than syntax error.

The fix is quite simple -- enclose each arg to `format()` in parens (two line change, see https://github.com/micropython/micropython/commit/8f52b545495f541e694af9950a7c73406e85201b )   (+16 bytes FWIW)

Did this come up in real code, is it worth fixing?

jepler, 2022-10-19:
What I _really_ wanted was repr-formatted `f"{arg=!r}"` which is a SyntaxError in MP/CP. Since tuple items are repr'd I thought adding the comma was a quick way to get what I wanted, but it didn't work out either.

So I'm not super interested in adding it, I'd rather bank 16 bytes against adding !r to f-strings.

Is it worth documenting as a difference, though?

jimmo, 2022-10-19:
> An adjacent mis-behavior that's more likely to bite someone:

@jepler Just thought of this too last night... :) Yeah I think we should consider fixing this then. I will send a PR with that commit above.

dpgeorge, 2023-01-20:
Fixed by fb8792c095802a9fcc8b38c1d8cbc3f641918433

PR #9678 (2023-01-20): py/lexer: Wrap f-string arguments to format in parens.
Fixes #9635 -- @jepler 

(As pointed out by @jepler, this provides a convenient workaround for the lack of `!r` for debugging, as printing a tuple of a value will implicitly cause the value to be repr'ed. But mostly fixing this for correctness as it can silently lead to the wrong results.)

+16 bytes on PYBV11.

This is important for literal tuples, e.g.

`f"{a,b,}, {c}"` --> `"{}".format((a,b), (c),)`

which would otherwise result in either a syntax error or the wrong result.

_This work was funded through GitHub Sponsors._
closed 2023-01-20 by dpgeorge