micropython/micropython #9652: docs: Pin.off() method is not documented

millosolomillo, 2022-10-17:
`Pin.off()` method works (at least for RP2040 platform) but it is not documented here: 

https://docs.micropython.org/en/latest/library/machine.Pin.html#methods](url)

robert-hh, 2022-10-17:
It's available only at the rp2 and samd port, as you know.

millosolomillo, 2022-10-17:
Already opened a PR at #9653, sorry :sweat_smile:

uPesy, 2022-10-17:
I really think board ports should be more harmonized for basic fonctions, otherwise we lost the theorical advantage of uPython : one code that works on different hardware ...

jonnor, 2024-09-02:
I do not consider toggle to be core. It is trivial to use `pin.value(not pin.value())`. Proposing to close.

projectgus, 2024-09-10:
> I do not consider toggle to be core. It is trivial to use `pin.value(not pin.value())`. Proposing to close.

I agree, but I think we at least should document it on the ports where it's already implemented.

PR #15822 (2024-09-19): doc: Document machine.Pin.toggle() method.
### Summary

Document the `pin.toggle()` method that's available on some ports.

This is an amended version of the commit from #9653. Closes #9652.
closed 2024-09-19 by dpgeorge