micropython/micropython #9659: mpremote: all mpremote commands are followed by a soft reboot

Josverl, 2022-10-17:
according to @jimmo  mpremote should be able to resume a repl session on a connected mcu without a reboot by using the following command pattern:
``` sh
mpremote connect exec "start_something()"
mpremote connect resume exec "continue_here()"
```
where the `resume` is the important part.
see [Discord discussion](https://discord.com/channels/574275045187125269/1029416691488264252/1031558675900407818) 

however when I try this : 
 1. I see that any variables set in the the first `mpremote` commands are lost 
 2. I see that the esp32 performs a soft reset after exiting the `raw REPL` 
     (observed through logic analyzer)

So even though the 2nd mpremote command does not force a reset, by that time a soft reboot has already occurred.
I also see a `soft reset` after: 

- exiting the repl
- eval "42"
- exec "print(42)"

## Actual result on windows : 
```
C:\develop\usb>mpremote exec "foo=bar=42;print(foo)"
42

C:\develop\usb>mpremote resume exec "print(bar)"
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'bar' isn't defined
```
## Expected 
the MCU not to reset between two mpremote commands similar to the behavour on linux 

```
$ mpr exec "foo=bar=42; print(foo)"
42
$ mpr exec "print(bar)"
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'bar' isn't defined
$ mpr exec "foo=bar=42; print(foo)"
42
$ mpr resume exec "print(bar)"
42
```

results as seen by a logic analyzer hooked up to the RX/TX pins of the esp32 
[image]

``` text
\x01\x03
>>> 
>>> 
>>> 
>>> 
raw REPL; CTRL-B to exit
\x02OKx05A\x01foo=bar=43;print(foo)\x04
MPY: soft reboot
raw REPL; CTRL-B to exit
>R\x01\x80\0\x01\x0443
\x04\x04>ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12448
load:0x40080400,len:4124
entry 0x40080680
MicroPython v1.19.1 on 2022-06-18; 4MB/OTA module with ESP32
Type "help()" for more information.
>>> 
>>> 
>>> 
>>> 
>>> 
raw REPL; CTRL-B to exit
\x02R\x01\x80\0\x01\x04\x04Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'foo' isn't defined
\x04>ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12448
load:0x40080400,len:4124
entry 0x40080680
MicroPython v1.19.1 on 2022-06-18; 4MB/OTA module with ESP32
Type "help()" for more information.
>>> 
```
## other observations 
- same behavior is observed with multiple firmware versions, and on different boards 
- esp8266 devices do NOT soft-reset in the same way  ( note: all my esp8266 devices use a different serialport driver) 
- 

## Versions 

 - mpremote : 0.4.0 
 - pyserial: 3.5
 - Python 3.9.13
 - Version: 1.72.2 (user setup)
 - OS: Windows_NT x64 10.0.22621
 - Micropython - v1.19.1
 - MCU - all esp32 
 - serial port driver esp32 : Silicon Laboratories Inc. 11.1.0.53
 - serial port driver esp8266 : wch.cn - No version or other details 
```
> mpremote devs
COM9 01665F52 10c4:ea60 Silicon Labs None # esp32
COM11 0001 10c4:ea60 Silicon Labs None # esp32 
COM20 00000000A 0403:6001 FTDI None # UART debug 
COM21  1a86:7523 wch.cn None # ESP8266
```

Josverl, 2022-10-17:
if you want to take a look at the trace :
[three_device_session.zip](https://github.com/micropython/micropython/files/9805167/three_device_session.zip)

open with : https://www.saleae.com/downloads/

jimmo, 2022-10-18:
@Josverl This is a hard reset, not a soft reset. A soft reset would not print out the esp boot sequence.

The only hint I have is that on many ESP32 boards, they wire up RTS/CTS from the on-board cp210x / ftdi in a way such that esptool.py can trigger the board to reset into the bootloader. I wonder if somehow the way the serial port is being opened on Windows is also triggering this?

Josverl, 2022-10-18:
you are correct , there is both a soft and a hard reset. I was blinded by the first one.
but im sure that  I and students in bootcamps have been able to connect to and disconnect many times to a board without it rebooting.

my esp32s all use a CP210x ,
while my esp8266 all have an CH340 

but that is about the most common UART/USB interface used, 

Could it be that the state of the DTR/RTS signals have an effect ?
One fix that I have needed to add to a [few terminal tools](https://github.com/pycom/pymakr-vsc/issues?q=is%3Aissue+is%3Aclosed+DTR) is to compensate for the fact that that windows and linux actually handle the DTR/RTS signals differently. (neither is wrong, just different) 
(I've not looked into how pyserial handles this yet)

mgsb, 2022-11-13:
I encountered this recently with a new-to-me board, SparkFun MicroMod ESP32. The [schematic](https://cdn.sparkfun.com/assets/2/2/5/9/5/MicroMod_ESP32_Schematic.pdf) documents the DTR/RTS behavior for resetting:

* If DTR is LOW, toggling RTS from HIGH to LOW resets to run mode
* If RTS is HIGH, toggling DTR from LOW to HIGH resets to bootloader

As a test, I changed the Pyboard constructor in `pyboard.py` to assert the DTR and RTS pins:

``` diff
@@ -275,7 +275,11 @@ class Pyboard:
             delayed = False
             for attempt in range(wait + 1):
                 try:
-                    self.serial = serial.Serial(device, **serial_kwargs)
+                    self.serial = serial.Serial(**serial_kwargs)
+                    self.serial.dtr = True
+                    self.serial.rts = False
+                    self.serial.port = device
+                    self.serial.open()
                     break
                 except (OSError, IOError):  # Py2 and Py3 have different errors
                     if wait == 0:
```

This fixed the problem on Linux and Windows, but not macOS (Big Sur, Monterrey).

On Linux, using

``` shell
stty -F /dev/ttyUSB0 -hup
```

also prevented the hard reset.

jimmo, 2022-11-14:
Thanks @mgsb -- From what I've read elsewhere this seems like the right approach, although frustrating about macOS.

I haven't had much chance to look into this in detail but one thing I'd like to understand -- is this a new issue, or has it always been a problem? i.e. was this always causing DTR/RTS to bounce quickly, but there was enough capacitance and/or pull strength that it was never noticed? Are we seeing it now because just a handful of boards have less capacitance on these lines?

@Josverl does this solve it for your board?

cymplecy, 2022-12-22:
Following long debugging session on Discord today (22Dec2022) I can confirm I had this issue on an ESP32-C3FH4 (ESP32-C3 with RBG matrix on board) and that the fix above fixed my issue

Who do I see about getting it incorporated into standard code base? :)

Josverl, 2022-12-22:
I'll give this a test, but the .py fix is like what I've been doing in .js for pymskr, but with the added condition/constraint to only manipulate dtr rts on win32/64

jimmo, 2022-12-22:
> Following long debugging session on Discord today (22Dec2022) I can confirm I had this issue on an ESP32-C3FH4 (ESP32-C3 with RBG matrix on board) and that the fix above fixed my issue

Some more details -- this was Windows, and the behaviour was that there was a reset on _disconnect_.

The C3 is interesting because it's on-chip USB/UART, rather than external, but clearly it's internally wired up the same way with the DTR/RTS pins.

> Who do I see about getting it incorporated into standard code base? :)

I will send a PR. I think it sounds like this change doesn't solve everything, but overall only improves things and doesn't make anything break. But will need some more testing.

closed 2023-01-13 by dpgeorge

andrewleech, 2023-07-31:
> I will send a PR. I think it sounds like this change doesn't solve everything, but overall only improves things and doesn't make anything break. But will need some more testing.

Lucky me found the case where it does break :-D on windows and stm32 with "custom" driver installed (with zadig) to give the ports a personalised name! So yeah not a common situation, though common to every device I make at work... 

I'm not sure how to handle it better here though really, other than adding a extra bit of extra injected code on connection to query `os.uname()` to enable stuff like this per-port. Of course this would fail if the device isn't at repl on connection (though much of mpremote fails in this circumstance).

In theory I might be able to update my driver creation / installation to keep the manufacturer as "Microsoft" to bypass this bit of mpremote code too, though that doesn't help devices / drivers already in the field...

jimmo, 2023-07-31:
@andrewleech What actually broke? The issue here was that it triggered a hard reset of the board due to the ESP32 bootloader-activation-via-DTR/RTS-thing. I can't see how that would apply to an stm32 board?

andrewleech, 2023-07-31:
For me, having `self.serial.dtr = False  # DTR False = gpio0 High = Normal boot` causes no data to be sent from STM to PC, so I get no console content come in. Text I type goes out, but is blocking as the STM is respecting the flow control and waiting for RTS to go high :-)

crackwitz, 2025-07-27:
This problem has reappeared.

This is the current code:

https://github.com/micropython/micropython/blob/ebc9525c953dfbb6c44ab99320b5278e4314dafe/tools/mpremote/mpremote/transport_serial.py#L68-L82

And this is my fix that works for me, against an ESP32-C3 that appears as a COM port with a driver by *Microsoft*:

```python
if portinfo and portinfo[0].manufacturer != "Microsoft":
    # ESP8266/ESP32 boards use RTS/CTS for flashing and boot mode selection.
    # DTR False: to avoid using the reset button will hang the MCU in bootloader mode
    # RTS False: to prevent pulses on rts on serial.close() that would POWERON_RESET an ESPxx
    self.serial.dtr = False  # DTR False = gpio0 High = Normal boot
    self.serial.rts = False  # RTS False = EN High = MCU enabled
else: # my addition
    self.serial.dtr = True
    self.serial.rts = False
```

crackwitz, 2025-07-28:
The device I'm dealing with is [from aliexpress](https://www.aliexpress.com/item/1005005967641936.html) and reports as VID 303A, PID 1001

Josverl, 2025-07-28:
> I did not try other DTR/RTS combinations since the first one I tried worked already

@crackwitz, the challenge is not only fixing this for your board, 
but doing that without breaking mpremote + other boards across a number use different usecases.

crackwitz, 2025-07-29:
I am aware of that, which is why I strive to make perfectly clear what *didn't* try.

andrewleech, 2025-07-29:
> @andrewleech, 
> did you find the root cause for your problem ?
> if it was indeed the RTS held high then perhaps its also possible to prevent that if there is a common VID for STM devices ? 

My particular issue was from having a custom vid pid and separately loaded driver that did not have Microsoft as the listed manufacturer so it didn't match the checks in code.

I tried making changes to this block of code to resolve the problems better but every option I tried had other subtle negative outcomes too, they're in a PR here somewhere I think.

projectgus, 2025-07-31:
This is probably a silly question as I'm not across the whole thing, but in the limit is it reasonable to patch the opposite case? (i.e. don't clear DTR/RTS for drivers/configs where we know that will cause a problem, do it for all other boards which should include most/all ESP32 boards plus a bunch of boards which don't care either way.

Alternatively, if it's too hard to get all of the custom cases then maybe this needs to be an option on `mpremote resume` (i.e. `mpremote resume --esp ...` or similar.

Josverl, 2025-07-31:
not silly , but hard to parse your inverse logic, but I think we may see the same way ahead.

let me try to recap my experience, and intents

- pyboard/pyserial defaults are OK on linux / macOS, and were mostly broken on Windows. ( different world visions between the OS ecosystems) 
- this mostly surfaces as spikes in the DTR/CTSlines at either connection , but more often at disconnection 
- this is most problematic in boards that use these to do special things, such as the ESPxxx boards that connect these lines to some logic and !EN and other strapping Pins
- it is not always simple to see / trace these signals - but on the ESP32 / ESP8266 chips it often is, so I used there for my testing.
- most of these boards have UART-USB convertors on board - with multiple different drivers and driver versions for each of these. 
Naming of these drivers is - not quite consistent, and new drivers are released .

To avoid ESPxx board to reset on disconnect , or hang in bootloader on connect we need the set DTR/CTS to False/False **Only for these boards**, both for UART/USB as well as for USB/CDC , but there is no authoritative list of VIDs / vendor names to do so.

With Pycom I was able to iterate quickly on user feedback - and updates were done in weeks, so we were able to add regexes for vendor strings that were reported quickly.
with mpremote some of the same updates have been stalled for months, or longer. That is the main reason I have chosen for the if not "Microsoft" driver approach.

Now with the modern ESPxxx boards switching to native USB , and less reliance on 3rd party UART drivers I think it is a good tie to switch tactics 

- The default mpremote config should have a list of VIDs-PIDs that require "ESP-Handling" on Windows
- this config should be extendible/overridable  though the mpremote configuration file (currently still broken on windows ) 
- when connecting to a port - mpremote can check if it needs ESP-Handling - and then act accordingly 

*  if / when needed new vids/pids can be added to the default list via PRs, and included in the next release 
*  personal / interim changes can be done by editing the mpremote config file.