micropython/micropython #9672: timer_group: timer_get_config(331): HW TIMER NEVER INIT ERROR

iBoundary, 2022-10-18:
MicroPython v1.19.1-dirty on 2022-10-12; WEMOS LOLIN32 LITE with ESP32-DOWDQ6
Type "help()" for more information.
>>> from machine import Timer
>>> t = Timer(-1)
>>> t
Timer(3ffe59d0; E (36710) timer_group: timer_get_config(331): HW TIMER NEVER INIT ERROR
alarm_en=1, auto_reload=1, counter_en=46545)
>>> t2 = Timer(1)
>>> t2
Timer(3ffe5b30; E (55290) timer_group: timer_get_config(331): HW TIMER NEVER INIT ERROR
alarm_en=1, auto_reload=1, counter_en=28611)
>>> t2 = Timer(0)
>>> t2
Timer(3ffe5cb0; E (60340) timer_group: timer_get_config(331): HW TIMER NEVER INIT ERROR
alarm_en=1, auto_reload=1, counter_en=28611)

robert-hh, 2022-10-18:
This is a message from RTOS, telling that the timer was not inititalized. Quite irritating, but harmless.

iBoundary, 2022-10-18:
> This is a message from RTOS, telling that the timer was not inititalized. Quite irritating, but harmless.
Do you know how to solve this problem?

closed 2022-10-19 by iBoundary