micropython/micropython #9689: `mpremote` utility doesn't find config.py in Windows

jerrens, 2022-10-20:
The `mpremote` utility expects to find the `.config/mpremote/config.py` file relative to the path stored in the `XDG_CONFIG_HOME` environment variable first, then looks in the `HOME` environment variable next if that was not defined. [^1]  

In Windows, `HOME` is not an environment variable defined by default, but the counterpart is `HOMEPATH`.

I am currently able to work around this problem by manually creating a `HOME` env var and setting it's value to `%HOMEPATH%`, but if the `mpremote` utility could be updated to either check the OS first and then use that to determine the appropriate environment variable, or use the suggestion in this StackOverflow answer [^2]:

```
from os.path import expanduser
home = expanduser("~")
```
If you're on Python 3.5+ you can use [pathlib.Path.home()](https://docs.python.org/3/library/pathlib.html#pathlib.Path.home):
```
from pathlib import Path
home = str(Path.home())
```

to correctly find the user's home directory, then support for user-defined commands would be enabled for all OSes.

[^1]: https://github.com/micropython/micropython/blob/f13134e40308c24bcb2c09084f17fc9d54efe5db/tools/mpremote/mpremote/main.py#L330-#L357
[^2]: https://stackoverflow.com/a/4028943/2136313

stinos, 2022-10-21:
> HOMEPATH

For all practical purposes it should be `USERPROFILE`. 

> by manually creating a HOME env var and setting it's value to %HOMEPATH%,

That's strange because HOMEPATH has no drive, i.e. full directory would be `%HOMEDIRVE%\%HOMEPATH%`

That being said: https://github.com/micropython/micropython/pull/9573 has the solution already.

jerrens, 2022-10-21:
Yes.  USERPROFILE is the better choice.

Looks like https://github.com/micropython/micropython/pull/9573 does the trick.  Not sure how I missed that one... I'm normally better at searching than that :-).

Sorry for the duplicate.

closed 2022-10-21 by jerrens