micropython/micropython #9707: ESP32S3 cant import TouchPad from machines

nikhil-robinson, 2022-10-22:
Hello there i am facing problem with importing touchpad function getting the bellow error.

`>>> from machine import TouchPad
Traceback (most recent call last):      
  File "<stdin>", line 1, in <module>   
ImportError: can't import name TouchPad 
>>>`

MCU i am using esp32s3 custom board
Compiled using GENERIC_S3 with 4MB flash enabled.
Idf version 4.4

kago35, 2022-10-24:
Hello,

i have got the same issue.
TouchPad unavailable with ESP32S3

>>> from machine import TouchPad
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name TouchPad

The used plateform is ESP32-S3-DevKitC-1
I have download the micropython binary from micropython.org: 
https://micropython.org/resources/firmware/GENERIC_S3-20220618-v1.19.1.uf2

See the log below:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0xf24
load:0x403b6000,len:0xc2c
load:0x403ba000,len:0x2d60
entry 0x403b61f4

MicroPython v1.19.1 on 2022-06-18; ESP32S3 module with ESP32S3
Type "help()" for more information.
MicroPython v1.19.1 on 2022-06-18; ESP32S3 module with ESP32S3

Type "help()" for more information.

>>> help('modules')
__main__          framebuf          uasyncio/stream   uplatform
_boot             gc                ubinascii         urandom
_onewire          inisetup          ubluetooth        ure
_thread           math              ucollections      uselect
_uasyncio         micropython       ucryptolib        usocket
_webrepl          neopixel          uctypes           ussl
apa106            network           uerrno            ustruct
btree             ntptime           uhashlib          usys
builtins          onewire           uheapq            utime
cmath             uarray            uio               utimeq
dht               uasyncio/__init__ ujson             uwebsocket
ds18x20           uasyncio/core     umachine          uzlib
esp               uasyncio/event    uos               webrepl
esp32             uasyncio/funcs    upip              webrepl_setup
flashbdev         uasyncio/lock     upip_utarfile     websocket_helper
Plus any modules on the filesystem
>>> import machine
>>> help(machine)
object <module 'umachine'> is of type module
  __name__ -- umachine
  mem8 -- <8-bit memory>
  mem16 -- <16-bit memory>
  mem32 -- <32-bit memory>
  freq -- <function>
  reset -- <function>
  soft_reset -- <function>
  unique_id -- <function>
  sleep -- <function>
  lightsleep -- <function>
  deepsleep -- <function>
  idle -- <function>
  disable_irq -- <function>
  enable_irq -- <function>
  bitstream -- <function>
  time_pulse_us -- <function>
  Timer -- <class 'Timer'>
  WDT -- <class 'WDT'>
  SDCard -- <class 'SDCard'>
  SLEEP -- 2
  DEEPSLEEP -- 4
  Pin -- <class 'Pin'>
  Signal -- <class 'Signal'>
  ADC -- <class 'ADC'>
  ADCBlock -- <class 'ADCBlock'>
  I2C -- <class 'I2C'>
  SoftI2C -- <class 'SoftI2C'>
  I2S -- <class 'I2S'>
  PWM -- <class 'PWM'>
  RTC -- <class 'RTC'>
  SPI -- <class 'SPI'>
  SoftSPI -- <class 'SoftSPI'>
  UART -- <class 'UART'>
  reset_cause -- <function>
  HARD_RESET -- 2
  PWRON_RESET -- 1
  WDT_RESET -- 3
  DEEPSLEEP_RESET -- 4
  SOFT_RESET -- 5
  wake_reason -- <function>
  PIN_WAKE -- 2
  EXT0_WAKE -- 2
  EXT1_WAKE -- 3
  TIMER_WAKE -- 4
  TOUCHPAD_WAKE -- 5
  ULP_WAKE -- 6
>>> from machine import Pin
>>> from machine import TouchPad
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name TouchPad
>>>

Hamsanger, 2022-11-01:
Same here - the `TouchPad` class is missing from the machine module in the ESP32-S3 port.  The problem may be related to line 31 in `ports/esp32/machine_touchpad.c`:

```
31 #if CONFIG_IDF_TARGET_ESP32
```
which excludes other targets (`CONFIG_IDF_TARGET_ESP32S2`, `CONFIG_IDF_TARGET_ESP32S3`) from the build.  This #if conditional is not matched until line 145 at the bottom of the file.

mattytrentini, 2023-01-04:
TouchPad was disabled on the ESP32-S2 and S3 because there were some implementation issues due to differences to the original ESP32. The issues have since been resolved (see #8955) so if you use a nightly you'll find TouchPad should work as expected.

closed 2023-01-04 by mattytrentini