micropython/micropython #9713: Periodic machine timer blocked while waiting socket.connect()

flis-mateusz, 2022-10-22:
Hi,

I am using micropython version 1.19.1 on the esp32 platform

The problem is that when connecting to the socket, the timer is blocked. 
After the socket connecting is interrupted, the callback function is called several times at a time

Here's a fragment of code where you can see the problem:
```import machine
import socket
import network
from machine import Timer
import time

sta_if = network.WLAN(network.STA_IF)
if not sta_if.isconnected():
    sta_if.active(True)
    time.sleep_us(200)
    sta_if.connect('xxx', 'xxx')

while not sta_if.isconnected():
    time.sleep(1)

t = Timer(-1)
t.init(period=1000, mode=Timer.PERIODIC, callback=lambda t:print(time.time()))

def connect():
    try:
        print('connecting')
        addr = socket.getaddrinfo('192.168.1.41', 1886)[0][-1]
        s = socket.socket()
        s.connect(addr)
        s.send(bytes('.....\n', 'utf8'))
        s.read()
        s.close()
    except Exception as e:
        print(e)
time.sleep(10)
connect()
```
Output:
```
719797401
719797402
719797403
719797404
719797405
719797406
719797407
719797408
719797409
connecting
719797428
[Errno 113] ECONNABORTED
719797428
719797428
719797428
719797428
719797428
719797428
719797428
719797429
719797430
719797431
719797432
719797433
...
```
As you can see, time.time() should be printed every 1sec, but when you call socket.connect() it stops printing. When the connecting is interrupted, the function is suddenly called several times with the same result

IhorNehrutsa, 2022-10-24:
socket.getaddrinfo() is modal function.
It blocks program execution inside self.

beyonlo, 2022-11-06:
What or how should be used in the source code to no block the `Timer?`

peterhinch, 2022-11-06:
The behaviour of `socket.getaddrinfo()` is a longstanding bugbear, there is a need for a nonblocking version. In [asynchronous MQTT](https://github.com/peterhinch/micropython-mqtt/tree/master/mqtt_as) we work round this by cacheing the result of the DNS lookup so that subsequent connections to the same server are nonblocking. This is not ideal.

An option would be to use hardware with hardware timers and hard IRQ's, e.g. STM32.

beyonlo, 2022-11-06:
In this case, for the `ESP32` that has no hard IRQ's, an alternative to have `Timer` no-blocking, is to create a `thread` just to run the `Timer`. Or put the socket in another thread. Am I correct?

peterhinch, 2022-11-07:
I haven't used threading on ESP32 but it should work. Communication between a thread and `uasyncio` needs care: you must use `ThreadSafeFlag` - see [my tutorial](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#36-threadsafeflag).