micropython/micropython #9746: mpremote cp -r doesn't work

one0410, 2022-10-26:
I'm using esp32-s3 on micropython 1.19.1.

On mpremote [document](https://docs.micropython.org/en/latest/reference/mpremote.html).
If you use cp -r, it assumes you can copy whole directory to esp32.

But when I use following command on Windows,
`
python -m mpremote cp -r main/ :
`

REPL give me an OSError:
<pre>
cp main/d40.py :main/
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 21] EISDIR
</pre>

jimmo, 2022-10-26:
Just checked this is working for me on Linux with the latest mpremote version. Which version of mpremote are you using?

Can you run the following two commands please:

`python -m mpremote ls`

`python -m mpremote ls main`

The main way I can think of that this problem could occur is if there exists a directory on the device called `main/d40.py/`.

one0410, 2022-10-27:
Thank you for the response.
I am also using the latest version of mpremote 0.4.0.
And the main directory is empty on ESP32-S3.
Please refer the following.

<pre>
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls           
ls :
         139 boot.py
           0 lib/
         293 main.py
           0 main/
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls main
ls :main
</pre>

And on my PC, there are many .mpy files in the main directory.
d40.mpy is the first one.
When mpremote wants to copy the first one, it failed.
<pre>
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote cp -r main/ :
cp main/d40.mpy :main/
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 21] EISDIR
</pre>

dpgeorge, 2022-10-27:
Does it work if you first run the following (to remove the main directory on the device):
```
python -m mpremote rmdir :main
```

one0410, 2022-10-27:
YES, I can remove the main directory on the device.

<pre>
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls
ls :
         139 boot.py
           0 lib/
         293 main.py
           0 main/
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls main      
ls :main
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote rmdir :main
rmdir :main
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls         
ls :
         139 boot.py
           0 lib/
         293 main.py
</pre>

Even after above, I can recreate the main directory on the device.
<pre>
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls         
ls :
         139 boot.py
           0 lib/
         293 main.py
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote mkdir :main
mkdir :main
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls
ls :
         139 boot.py
           0 lib/
         293 main.py
           0 main/
</pre>

dpgeorge, 2022-10-27:
Does the recursive copy work after `main` is removed from the device?

one0410, 2022-10-27:
> Does the recursive copy work after `main` is removed from the device?

Still can't work, no matter if "main" directory exists or not on the device.

<pre>
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote ls                       
ls :
         139 boot.py
           0 lib/
         293 main.py
           0 main/
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote rmdir :main
rmdir :main
PS C:\Users\onero\Dropbox\PC (2)\Documents\GitHub\TaidocGateway\bin> python -m mpremote cp -r main/ :
cp main/d40.mpy :main/
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 21] EISDIR

</pre>

hybotics, 2022-10-29:
I am also having the same problem with copying a directory to a device. I also have mpremote v0.4.0. Running on a Raspberry Pi 400 with Ubuntu 22.04.
"mpremote cp -r hybotics_pca9685 :" does not complete. The directory has four files. Sometimes I get "could not complete raw paste: b'\x01'" but most of the time the command never completes and I have to abort it. I cleared everything off the device before I started copying the directory to it. I ended up having to copy each file separately, but this is a real pain to do.

jonnor, 2024-09-01:
Does this issue still exist with latest MicroPython, version 1.23 or later? Many improvements have been made to `mpremote` (and ESP32) over the last 1.5 years.

mattytrentini, 2024-09-02:
Not sure if it's still a problem, but it's worth mentioning #11777 which overhauls `mpremote`'s recursive copy feature (among other improvements).

closed 2025-02-04 by projectgus