micropython/micropython #9800: Working with filesystems - Hybrid (ESP32)

Tangerino, 2022-11-01:
In the example of using alternate partition in ESP32 the example does not work
```python
import esp32, os
p = esp32.Partition.find(esp32.Partition.TYPE_DATA, label='foo')
os.mount(p, '/foo')
```

My partition table
```
ID Type   Subtype   Address  Size     Label    Encripted
 0   0      0         65536  2031616  factory  False
 1   1      2         36864    24576  nvs      False
 1   1      1         61440     4096  phy_init False
 1   1      129     2097152  2097152  vfs      False
 1   1      129     4194304  4194304  data     False
```

The result when running the example
```
>>> import esp32, os
p = esp32.Partition.find(esp32.Partition.TYPE_DATA, label='data')
os.mount(p, '/foo')
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
AttributeError: 'list' object has no attribute 'readblocks'
>>> 
```

jimmo, 2022-11-01:
`esp32.Partition.find` returns a list. The example should be `os.mount(p[0], '/foo')` (and likely something to check that the list wasn't empty etc).

jonnor, 2024-09-02:
Latest documentation seems to have the same problem still. Merge requests welcomed!