micropython/micropython #9842: mpy-ld.py: preprocess c source before trying to find qstr-s

TsXor, 2022-11-05:
Currently, mpy-ld.py only directly search for `MP_QSTR_*` keywords in the source file declared in `Makefile`, in some situations, it will fail to find `MP_QSTR_*` keywords and make it fail to build:
- It's too ugly to put all my source code in only one file, so I put some code which uses `MP_QSTR_*` in another file and includes it. 
- `MP_QSTR_*` expression is constructed with a macro.

I think expanding users' includes and macros with something like [pcpp](https://github.com/ned14/pcpp#:~:text=A%20pure%20universal%20Python%20C%20%28pre-%29preprocessor%20implementation%20very,conventional%20C%20preprocessor%20%28i.e.%20it%27ll%20accept%20similar%20arguments%29.) may help.

TsXor, 2022-11-05:
btw: If such problems are encountered, there's a trick: manually put all `MP_QSTR_*` expressions in comments in the source file declared in `Makefile`

jimmo, 2022-11-05:
> I think expanding users' includes and macros with something like [pcpp](https://github.com/ned14/pcpp#:~:text=A%20pure%20universal%20Python%20C%20%28pre-%29preprocessor%20implementation%20very,conventional%20C%20preprocessor%20%28i.e.%20it%27ll%20accept%20similar%20arguments%29.) may help.

Rather than introducing a third-party dependency it might be easier just to use the exact same mechanism that the regular build uses to extra QSTRs (described here: https://docs.micropython.org/en/latest/develop/qstr.html). I don't know exactly the reason why native modules use the simpler version (probably just for simplicity of implementation and reduced dependencies).

TsXor, 2022-11-05:
> Rather than introducing a third-party dependency it might be easier just to use the exact same mechanism that the regular build uses to extra QSTRs (described here: [...]

Maybe it's because native modules live in runtime.
https://github.com/micropython/micropython/blob/master/py/dynruntime.h#L41-L43