micropython/micropython #9851: Improve BLE on nRF52

0xabe1, 2022-11-06:
The ubluepy module for the nRF is pretty lacking. For one, if a BLE connect fails, it just hangs forever. I saw somewhere that @jimmo had a plan to swap the Nordic SoftDevice for NimBLE which would give the Nordic folks access to the awesomeness that is aioble. 

I spent the better part of today trying to figure that out, but I just lack foundational knowledge on the Cortex-M4 platform. I did manage to get NimBLE compiled with the RAM HCI within the MicroPython build tree, but it doesn’t run because there’s no background task. 

Since the nRF52 and STM32 are both M4s, it looks like we should be able to use the same SysTick mechanism to run the BLE stack, but that’s where I got in over my head with all the IRQ stuff. I don’t quite understand the M4 platform or how MicroPython uses/allocates interrupts. 

Is my analysis reasonable? Is this something somebody could help me understand so I can finish porting? Or is this easy enough that it might get implemented easily by the MicroPython team? I’ve got some nRF52832s and nRF52840s I’d really like to get MicroPython working on. With time I could figure it out, but it looks so similar to the STM32, maybe somebody knows how to get it done more quickly?

Thanks.

jimmo, 2022-11-06:
@0xabe1 See https://github.com/micropython/micropython/pull/8318 for an initial attempt at migrating the nRF port to NimBLE.

I think the more pragmatic approach going forward for the nRF port is to migrate to Zephyr and make the Zephyr port the primary way of supporting nRF chips, and eventually remove the nRF port altogether. The development of the BLE stack for the nRF family (especially the nRF53) is far more active on Zephyr than NimBLE. I am working on this at the moment although I haven't had much time to spend on it in the past couple of weeks.

Supporting aioble on Zephyr on nRF is exactly the goal (and bringing the Zephyr port to feature parity with the existing nRF port at the same time).

0xabe1, 2022-11-06:
Ah, ok. Thanks for the insight into the plan. I won’t waste too much time on NimBLE then. 

The RTOS strategy is probably a good one. Have you seen the nRF7002? Nordic is adding a wifi companion chip to their portfolio next year. Probably get Wi-Fi and networking for free with Zephyr too.

iabdalkader, 2022-11-10:
> it looks like we should be able to use the same SysTick mechanism to run the BLE stack

I have `pendsv` working in `nrf` port, take a look maybe there's something you can borrow https://github.com/openmv/micropython/tree/openmv-1.19/ports/nrf 

But why not just use `mp_sched_schedule(_node)` ? See https://github.com/micropython/micropython/blob/master/ports/rp2/mpbthciport.c and https://github.com/micropython/micropython/blob/master/ports/rp2/mpnimbleport.c  https://github.com/micropython/micropython/blob/master/extmod/network_ninaw10.c#L104

jonnor, 2024-07-13:
There seems to be some progress on this the last weeks in #8318 (PR already mentioned above, but nearly 2 years ago, so thought it relevant to update).

jonnor, 2025-03-24:
@jimmo do you have any code lying around for Bluetooth support on Zephyr?

bikeNomad, 2025-07-10:
I built a BLE-enabled Zephyr build for the XIAO BLE NRF52840 SENSE board. It was able to run the `aioble` `temp_sensor.py` example, but I haven't tried much else yet.

In the board-specific config, I have:
```
# Enable Bluetooth
CONFIG_BT=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_CENTRAL=y
CONFIG_BT_DEVICE_NAME="XIAO nRF52840 Sense"
CONFIG_BT_GATT_DYNAMIC_DB=y
CONFIG_BT_DEVICE_NAME_DYNAMIC=y
CONFIG_BT_GATT_CLIENT=y
CONFIG_BT_L2CAP_TX_MTU=252
CONFIG_BT_BUF_ACL_RX_SIZE=256
CONFIG_BT_GATT_ENFORCE_SUBSCRIPTION=n
```