micropython/micropython #9856: Unable to Publish Message to AWS IoT Cloud

vivekmystrey, 2022-11-06:
MCU: ESP32-S3
Version: GENERIC_S3-20220618-v1.19.1

With umqtt.robust2 unable to publish messages to AWS IoT Cloud and there is no error too, while with umqtt.robust it works properly in ESP32-C3 for version v1.17.

Have used micropython version - v.1.19 and his error is common with both ESP32-C3 and ESP32-S3.

And, below is my code snippet:

`#AWS MQTT client cert example for esp8266 or esp32 running MicroPython 1.9
from umqtt.robust2 import MQTTClient
import time
import network
import machine
from machine import Pin, ADC
from time import sleep
import usocket as socket
import ujson

#Certs for ESP8266
CERT_FILE = "/flash/ESP32-Core.cert.pem"
KEY_FILE = "/flash/ESP32-Core.private.key"

#if you change the ClientId make sure update AWS policy
MQTT_CLIENT_ID = "basicPubSub"
MQTT_PORT = 8883

MQTT_KEEPALIVE_INTERVAL = 45
THING_NAME = "ESP32-Core1"
SHADOW_UPDATE_TOPIC = "/sdk/test/Python"

RESPONSE_RECEIVED = False

#if you change the topic make sure update AWS policy
MQTT_TOPIC = "esp8266-topic"

#Change the following three settings to match your environment
MQTT_HOST = "*******************-ats.iot.us-east-1.amazonaws.com"
WIFI_SSID = "*************"
WIFI_PW = "************"

mqtt_client = None

def sub_cb(topic, msg, retained, duplicate):
    print((topic, msg, retained, duplicate))

def connect_mqtt():
    try:
        with open(KEY_FILE, "r") as f:
            key = f.read()

        print("MQTT received KEY")

        with open(CERT_FILE, "r") as f:
            cert = f.read()

        print("MQTT received CERTIFICATE")

        mqtt_client = MQTTClient(client_id=MQTT_CLIENT_ID, server=MQTT_HOST, port=MQTT_PORT, keepalive=5000, ssl=True, ssl_params={"cert":cert, "key":key, "server_side":False})
        mqtt_client.connect()
        mqtt_client.set_callback(sub_cb)
        print("MQTT is connected")
        a = {"message": "Hello from IoT console"}
        try:
            mqtt_client.publish(SHADOW_UPDATE_TOPIC, ujson.dumps(a))
            print("Publish Message")
        except Exception as e:
            print('Cannot Publish' + str(e))

    except Exception as e:
        print('Cannot connect to MQTT ' + str(e))
        raise

def connect_wifi(ssid, pw):
    wlan = network.WLAN(network.STA_IF)

    if(wlan.isconnected()):
        wlan.disconnect()
    nets = wlan.scan()

    if not wlan.isconnected():

        wlan.active(True)
        wlan.connect(WIFI_SSID, WIFI_PW)
        while not wlan.isconnected():
            pass
    print("Connected as:", wlan.ifconfig())

try:
    print("Connecting to WIFI...")
    connect_wifi(WIFI_SSID, WIFI_PW)
    print("Connecting to MQTT...")
    connect_mqtt()
    print("OK")

except Exception as e:
    print(str(e))`

And, below is the output of the script:

`>>> %Run -c $EDITOR_CONTENT
Connecting to WIFI...
Connected as: ('192.***.*.*', '255.255.255.0', '192.***.*.*', '192.***.*.*')
Connecting to MQTT...
MQTT received KEY
MQTT received CERTIFICATE
MQTT is connected
Publish Message
OK`

The publish event is successful, however, still don't see the message in AWS. Please advise, what is wrong here? Why the publish event is successful when the message is not sent?

Below is the snippet of ESP32-S3 filesystem:

[image]

jimmo, 2022-11-07:
> With umqtt.robust2 unable to publish messages to AWS IoT Cloud and there is no error too. While, with umqtt.robust it works properly for below script with ESP32-C3 for version v1.17 and message is received successfully in AWS.

Sorry I can't quite figure out what the exact issue is and what does and doesn't work. I don't know what `umqtt.robust2` is -- it's not an "official" library from micropython-lib. Also it sounds like you're saying your code _is_ working with (what I assume is the micropython-lib version) umqtt.robust.

Either way, I think this is the wrong place for this issue, this is the issue tracker for the development of micropython itself. If there's an issue with umqtt.robust, then raise an issue with https://github.com/micropython/micropython-lib, or otherwise you need to follow up with umqtt.robust2.

closed 2022-11-07 by jimmo

vivekmystrey, 2022-11-07:
Hi Jimmo,

It is problem with the stable v1.19 version, had tried with the unstable v1.19 version and it works without any issues.