micropython/micropython #9866: nrf: download_ble_stack.sh failing to download SD binaries.

jimmo, 2022-11-07:
Something has changed in `https://www.nordicsemi.com/api/sitecore/Products/MedialibraryZipDownload2` that now returns a 403 page with  "www.nordicsemi.com needs to review the security of your connection before proceeding".

The URLs are still correct and they work in the browser, but I guess something is triggering abuse protection. Not sure exactly, when I copy the successful requests from Chrome devtools as curl commands, they fail in curl.

This is causing the nRF build to fail in CI.

@glennrub Do you have any idea what might have changed or how we can work around this? Thanks

glennrub, 2022-11-09:
@jimmo 
I tried to run the script myself, and it works for me (wget version 1.19.4). However, I'll check a bit if it is possible do download using an alternate method.

dpgeorge, 2022-11-10:
The nrf CI of recent PRs seem to work again...

jimmo, 2022-11-15:
OK thanks @glennrub -- looks like this was just a temporary thing. Sorry for the noise!

closed 2022-11-15 by jimmo