micropython/micropython #9889: Support for Pico with SSD1306

newhinton, 2022-11-08:
I would like to see micropython to add support for ssd1306 for the pico. 
It is already supported for esp's ([Documentation](https://docs.micropython.org/en/latest/esp8266/tutorial/ssd1306.html)) so it would be great to have it there also. 

There are alternative implementations ([eg. here](https://github.com/stlehmann/micropython-ssd1306)) that work, but lack some functionality. Thanks!

robert-hh, 2022-11-08:
There is no reason why it should not work with a pico. Only the pin numbers and the selection of SPI or I2C are different, but the driver itself is port independent.

newhinton, 2022-11-08:
I have installed [v1.19.1-628-gecd4d5439 (2022-11-08) .uf2](http://www.micropython.org/resources/firmware/rp2-pico-w-20221108-unstable-v1.19.1-628-gecd4d5439.uf2) from the website, and when i try to import ssd1306 via the pico-w-go extension of vscode, i get a `module not found` error. I can use mip to load it, but that is unreliable.

robert-hh, 2022-11-08:
It is located here: https://github.com/micropython/micropython-lib/tree/master/micropython/drivers/display/ssd1306
You can simply copy it to your pico using e.g. mpremote or thonny.