micropython/micropython #9890: mp_obj_t type is not a C++ constant expression

nomis, 2022-11-08:
Despite only containing constant values, the following causes a C++ compiler to allocate the struct in RAM and generate a static initialiser function:
```
const struct {
    mp_obj_t y = MP_ROM_INT(42);
} x;
```

Forcing the compiler to treat it as a constant expression explains what the problem is:
```
constexpr const struct {
    mp_obj_t y = MP_ROM_INT(42);
} x;
```
```
in constexpr expansion of ‘<unnamed struct>()’
error: value ‘85’ of type ‘mp_obj_t {aka void*}’ is not a constant expression
```

C++ forbids the use of pointers in constant expressions unless they refer to a static object, a function or are nullptr. There's no way around this.

Object representations A, B and C use a `void*` type for `mp_obj_t`. This makes it impossible to initialise an `mp_obj_t` value at compile time for them in C++ code. Changing `mp_obj_t` to `uintptr_t` makes this work but then `mp_const_obj_t` is no longer `const` in any representation. Perhaps the type could be changed to `uintptr_t` only if `__cplusplus` is defined?

stinos, 2022-11-09:
> Perhaps the type could be changed to uintptr_t only if __cplusplus is defined?

On 32bit architectures that would double the size of the object representation so probably not desired to have it by default. And you can always `#define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_D)` yourself. Just wondering: do you have a concrete example of where it's beneficial to have a constexpr value like this?

jimmo, 2022-11-09:
@stinos 

> On 32bit architectures that would double the size of the object representation so probably not desired to have it by default.

Isn't `sizeof(uintptr_t)` always `sizeof(void*)` ? So no change here.

> Just wondering: do you have a concrete example of where it's beneficial to have a constexpr value like this?

I had assumed the issue isn't with `constexpr` specifically, but rather that because it's not handling the constness, it puts the data in RAM not ROM. (The `constexpr` is only relevant to get the compiler to explain itself)

@nomis 

> Changing mp_obj_t to uintptr_t makes this work but then mp_const_obj_t is no longer const in any representation.

To make sure we're talking about the same thing, I'm interpreting what you're saying as "there's no such thing as a "uintptr_to_const_t" -- i.e. a uintptr_t equivalent of `const void *`. And of course `const uintptr_t` means something different, and not what we want.

> Perhaps the type could be changed to uintptr_t only if __cplusplus is defined?

I don't really like the idea of losing the safety check of what we currently use `mp_const_obj_t` for, but just doing this for C++ builds might be reasonable. (And certainly preferable to putting stuff in RAM that should be in ROM!)

stinos, 2022-11-09:
> Isn't sizeof(uintptr_t) always sizeof(void*) ? So no change here.

Those are the same size yes, but I think this is about
```
#if MICROPY_OBJ_REPR == MICROPY_OBJ_REPR_D
typedef uint64_t mp_obj_t;
typedef uint64_t mp_const_obj_t;
#else
typedef void *mp_obj_t;
typedef const void *mp_const_obj_t;
#endif
```
so assumed nomis meant `uint64_t` not `uintptr_t`

nomis, 2022-11-09:
> I had assumed the issue isn't with `constexpr` specifically, but rather that because it's not handling the constness, it puts the data in RAM not ROM. (The `constexpr` is only relevant to get the [...]

I'm doing something [a bit silly by parsing a string into a constant tuple with the compiler](https://github.com/nomis/aurora-coriolis/blob/2f1447c0b36e5396cef52eec43e53374c21223f8/src/version.cpp#L155-L165) but the issue with things in C++ going into RAM and being initialised at runtime is still a problem.

Note: many of the other `#defines` don't work in C++ because they initialise structs in ways that C++ doesn't support, so there's limited scope for defining `mp_*_t` things already. I need to resort to a separate C file for those.

> To make sure we're talking about the same thing, I'm interpreting what you're saying as "there's no such thing as a "uintptr_to_const_t" -- i.e. a uintptr_t equivalent of `const void *`. And of [...]

Yes, `const uintptr_t` causes insurmountable problems. Aside from it being const any union with a const field expects it to be initialised in C++.

> Those are the same size yes, but I think this is about
> 
> ```
> #if MICROPY_OBJ_REPR == MICROPY_OBJ_REPR_D
> ```

There are no issues with representation D because it doesn't use `void *`.

nomis, 2022-11-13:
It's inconvenient to be unable to define a globals table for a module in C++ because any constants defined in C++ header files are unavailable in C sources to add as constants to module.

dlech, 2023-02-17:
> I don't really like the idea of losing the safety check of what we currently use `mp_const_obj_t`

On the other hand, `mp_obj_t` currently has no type safety with regards to `MP_OBJ_TO_PTR()` and `MP_OBJ_FROM_PTR`. Independent of the C++ use case (investigating a strange issue that turned out to be a compiler bug), I tried changing the type of `mp_obj_t` from `void *` to `uintptr_t` and found dozens of places where we missed these macros because we use object representation A only. These cases seem much more frequent than cases where `mp_const_obj_t` would catch a problem so given a choice between the two, I would prefer the former.

And since `mp_obj_t` can be encoded small ints/qstrs/const objects that are not valid pointers, using `void *` doesn't seem like the correct type anyway. So I think there is some motivation for considering changing the type definition globally, not just for C++.