micropython/micropython #9891: ESP32-C3 fatal error on installing micropython

ctvarlan, 2022-11-08:
### Discussed in https://github.com/orgs/micropython/discussions/9881

<div type='discussions-op-text'>

<sup>Originally posted by **ctvarlan** November  7, 2022</sup>
Hello everyone,
The first try to put micropython on an older ESP32-C3. 
The command is:

_>esptool.py --chip esp32c3 --port COM4 --baud 460800 write_flash -z 0x0 esp32c3-usb-20220117-v1.18.bin_

and the messages are:

_esptool.py v4.3
Serial port COM4
Connecting....
Chip is ESP32-C3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:43:10:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size..._

_A fatal error occurred: esp32c3-usb-20220117-v1.18.bin requires chip revision 3 or higher (this chip is revision 2). Use --force to flash anyway._

I tried --force and the result is the same.

What should I do?

Thank you for your attention,
</div>

jonnor, 2024-09-02:
According to the discussion thread the flashing could be performed when using esptool 3.3 - and there is a patch for changing minimum revision for C3 to 2 instead of 3. Which involves changing CONFIG_ESP32C3_REV_MIN_3=y to CONFIG_ESP32C3_REV_MIN_2=y in the board configuration. That should probably be used by people who need to use that chip revision.

Unless revision 2 is very common, I propose we close this as a wont fix.

dpgeorge, 2024-09-02:
I haven't seen many reports of older revisions (less than rev 3) of ESP32C3 not working with MicroPython, so they must be relatively rare parts.

Rev 3 moved some WiFi code into ROM allowing the firmware to be smaller, and I think it's reasonable for us to require that revision in our firmware builds.

If users need to support revision 1 or 2 then they can build the firmware themselves.

closed 2024-09-02 by dpgeorge