micropython/micropython #9920: uname attribute missing from os library for Sparkfun SAMD51 Thingplus build

RetiredWizard, 2022-11-11:
After importing the os library I was expecting **os.uname()** to return the board name but the attribute doesn't seem to exist in the library.

I am using the latest nightly build:
MicroPython v1.19.1-645-g451ded8d7 on 2022-11-10; Sparkfun SAMD51 Thing Plus with SAMD51J20A

    >>> import os
    >>> os.uname()
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    AttributeError: no such attribute
    >>> dir(os)
    ['__class__', '__name__', 'remove', '__dict__', 'VfsFat', 'VfsLfs1', 'VfsLfs2', 'chdir', 'dupterm', 'getcwd', 'ilistdir', 'listdir', 'mkdir', 'mount', 'rename', 'rmdir', 'stat', 'statvfs', 'umount', 'unlink', 'urandom']

robert-hh, 2022-11-11:
os.uname() was intentionally not included, since it's planned to generally remove support for it in MicroPython. You can use sys.platform and sys.implementation instead, which provides the same information about the board.

RetiredWizard, 2022-11-11:
Thanks!!! I'll start pulling os.uname from my code :).  While working on this board I also noticed the /lib directory wasn't included in the default sys.path list.

    >>> import sys
    >>> sys.path
    ['', '.frozen']

Is that intentional as well?

robert-hh, 2022-11-11:
> I also noticed the /lib directory wasn't included in the default sys.path list.

No, that's not intentional, but easy to fix, either in the C code or in boot.py.

RetiredWizard, 2022-11-11:
I've updated my app to insert the path but if you want me to open an issue let me know. If it's of help and you point me in the direction of the  C module that defines the list, I can probably get a pull request in on it as well. I assume it's somewhere in the SAMD port directories.

dpgeorge, 2022-11-11:
Most ports do add `lib` to `sys.path`.  There's actually common code in `py/runtime.c:mp_init()` to populate `sys.path` but it doesn't add `lib` because that's a little bit port-specific.

For samd, probably best to add this to the path in `ports/samd/modules/_boot.py` (similar to mimxrt).

robert-hh, 2022-11-11:
@RetiredWizard Just like Damien told, _boot.py is a proper place. You can make a PR for that, or I can include that into the next "service-pack" type PR, which is usually a collection of tiny changes.

RetiredWizard, 2022-11-11:
It's easy enough to update the path on the APP side until you have a chance to push the service-pack.

robert-hh, 2022-11-11:
While you're at it, you could try if the flashbdev.py driver works on your Sparkfun board. I have that board too, but replaced he 512k chip by a 8M chip, and do not want to solder it back. The driver is here: https://github.com/robert-hh/SPI_Flash
The script to mount the flash is:
```
import os
from flashbdev import FlashBdev
from machine import SPI, Pin

spi=SPI(0, sck="FLASH_SCK", mosi="FLASH_MOSI", miso="FLASH_MISO", baudrate=24_000_000)
cs = Pin("FLASH_CS", Pin.OUT, value=1)

flash=FlashBdev(spi, cs)
try:
    vfs = os.VfsLfs1(flash, progsize=256)
except OSError as e:
    print("Mount failed with error", e)
    print("Recreate the file system")
    os.VfsLfs1.mkfs(flash, progsize=256)
    vfs = os.VfsLfs1(flash, progsize=256)

os.mount(vfs, "/flash")
```

RetiredWizard, 2022-11-11:
When I first loaded your script I got the "Mount failed with error/Recreate the file system", unfortunately I didn't make note of the error number (sorry, getting too late for me :) ). The file system seemed to create fine I copied a file to it, soft rebooted and ran your script again (no errors this time) and was able to access the file fine.

Looks good to me...

robert-hh, 2022-11-11:
The error message on the first attempt is fine and expected, because there was no file system initially.

RetiredWizard, 2022-11-11:
flashbdev is very useful on this board :D. 

Thanks for pointing me to it, I'll definitely use it. Are you planning on adding the functionality into the port for this board?

RetiredWizard, 2022-11-11:
Would Pin 17 be better named "LED" instead of "D13"?

robert-hh, 2022-11-11:
The pin labelling had to be reworked. D13 is the LED on Arduino boards. flashbdev is primarily make for samd21 devices. I plan to put it into frozen bytecode as default. samd51 devices usually have the flash at the QSPI interface. for these, a C code class is needed. That's planned, but not now.

closed 2022-11-12 by RetiredWizard

robert-hh, 2022-11-12:
@RetiredWizard About D13 vs. LED. There is a board pin called `D13`. Internally the LED is as well connected to it. In that case, `D13` is preferred.

RetiredWizard, 2022-11-12:
@robert-hh That makes sense, thanks!