micropython/micropython #9921: The memory of ESP32S3 is too large when using microython

Zhengqinze05, 2022-11-11:
I use the board of the ESP32. When I turn on microython, there will be more than 100 k of internal memory consumption. I have other functions to add. This more than 100 k of memory will cause me to run out of memory. What can I do to reduce the memory consumption of esp32 when using microython

sosi-deadeye, 2022-11-11:
Try to [compile](https://docs.micropython.org/en/latest/develop/optimizations.html#frozen-bytecode) your MicroPython-Sourcecode to .mpy. If I'm right, the code is frozen and does not consume memory if it's imported because it's already available in the structure which MicroPython requires.

peterhinch, 2022-11-11:
See [this doc](http://docs.micropython.org/en/latest/reference/constrained.html) for general advice.

YanMinge, 2022-11-16:
can we put the bytecode into SPI RAM, looks like we don't need to use internal ram to store bytecode

peterhinch, 2022-11-17:
The doc I referenced above describes compiling MicroPython code as part of the firmware. The bytecode is then stored in Flash. The MicroPython runtime then accesses the bytecode from Flash with the result that the code occupies minimal RAM.