micropython/micropython #9924: unix port: cannot import from machine module

Korhm, 2022-11-11:
Hello,

On my Ubuntu 64-bit machine, I built the unix port, but I'm not able to import from `machine` module

My PC:
```bash
➜  ~ uname -a
Linux LinuxDamien 5.15.0-52-generic #58-Ubuntu SMP Thu Oct 13 08:03:55 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

Here what I did (I had troubles to build, but after installing properly `libffi7` and `libffi-dev`, everything ran fine)
```
sudo apt update
sudo apt install libffi7 libffi-dev

git clone https://github.com/micropython/micropython.git

cd micropython/port/unix
make submodules
make
```

Then, I run the tests from the unix port README. Everything is OK
```python
./build-standard/micropython
MicroPython v1.19.1-660-gc8913fdbf on 2022-11-11; linux [GCC 11.3.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 
>>> list(5 * x + y for x in range(10) for y in [4, 2, 1])
[4, 2, 1, 9, 7, 6, 14, 12, 11, 19, 17, 16, 24, 22, 21, 29, 27, 26, 34, 32, 31, 39, 37, 36, 44, 42, 41, 49, 47, 46]
```
But... 
```python
>>> from machine import Pin
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name Pin
```

I also have an ESP8266 NodeMcu board on which I run `esp8266-20220618-v1.19.1.bin`, and this command works perfectly on it

My code on the ESP8266 also use neopixel lib. On the unix port, I had to install the lib with mip
```python
>>> import mip
>>> mip.install('neopixel')
Installing neopixel (latest) from https://micropython.org/pi/v2 to /home/damien/.micropython/lib
Exists: /home/damien/.micropython/lib/neopixel.mpy
Done
```

But I cannot install machine with mip
```python
>>> mip.install('machine')
Installing machine (latest) from https://micropython.org/pi/v2 to /home/damien/.micropython/lib
Package not found: https://micropython.org/pi/v2/package/6/machine/latest.json
Package may be partially installed
```
Or
```bahs
./build-standard/micropython -m mip install machine 
Installing machine (latest) from https://micropython.org/pi/v2 to /home/damien/.micropython/lib
Package not found: https://micropython.org/pi/v2/package/6/machine/latest.json
Package may be partially installed
```

Did I miss something ?

Can you please help me on how to make `machine`work on the unix port. I would like to run some unittests of a custom lib

dlech, 2022-11-11:
The unix port doesn't have a `machine` module. You can make your own for your tests that simulates the real hardware. Just make a `machine.py` file and add the directory where it is located to `sys.path` or use the `MICROPYPATH` environment variable when running your tests.

Korhm, 2022-11-11:
Ho...
Thank you for your very fast answer @dlech  ! :) 
I'll create fake modules for my unittests (machine, network...)

dpgeorge, 2022-11-14:
There is a `machine` module for the unix port with some basic functionality in micropython-lib: https://github.com/micropython/micropython-lib/tree/master/unix-ffi/machine

closed 2022-11-14 by dpgeorge