micropython/micropython #9927: rp2 v1.19.1-660-gc8913fdbf (2022-11-11) .uf2 nightly not recognized under Windows 10

scruss, 2022-11-12:
There are a couple of reports that rp2040 boards flashed with [v1.19.1-660-gc8913fdbf (2022-11-11) .uf2](https://micropython.org/resources/firmware/rp2-pico-20221111-unstable-v1.19.1-660-gc8913fdbf.uf2) are not creating virtual serial ports under Windows and thus won't connect to Thonny or other serial terminals:

* [Thonny, pico and micropython - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=342797#p2053651)
* [PICO W/Thonny won't connect after UF2 install](https://forums.raspberrypi.com/viewtopic.php?p=2053606#p2053598)

The common factors are

1.  rp2040
2. Windows 10
3. v1.19.1-660-gc8913fdbf (2022-11-11) .uf2 nightly build

This could be related to USB changes in [rp2: Allow enabling USB device without enabling USB-CDC. · micropython/micropython@c8913fd](https://github.com/micropython/micropython/commit/c8913fdbfadd43c879bba4d6d565be8b644f1feb)

dpgeorge, 2022-11-14:
Thanks for the report.  Looking at dmesg under Linux, I can see that the latest firmware has the following error message (which wasn't present in prior firmware):
```
usb 1-1.4: config 1 has an invalid interface number: 1 but max is 0
usb 1-1.4: config 1 has 2 interfaces, different from the descriptor's value: 1
```

RetiredWizard, 2022-11-14:
On Windows 8, with the Adafruit Feather RP2040, after installing the nightly build: [v1.19.1-660-gc8913fdbf (2022-11-11) .uf2], the COM port was not available and there was a new "Universal Serial Bus controller" entry labeled: "Unknown USB Device (Device Descriptor Request Failed)".

The previous nightly build: [v1.19.1-646-g0698dd72e (2022-11-11) .uf2] worked fine.

dpgeorge, 2022-11-14:
See #9942 for a fix.

RetiredWizard, 2022-11-14:
I built the latest bits with #9942 included and a COM port now shows up and the Device Manager doesn't display any "Unknown" USB devices. Seems to be working on Windows 8.

dpgeorge, 2022-11-14:
Fixed by dbb0019666a3270b0ed032e2cb9479eadc599c46

closed 2022-11-14 by dpgeorge