micropython/micropython #9929: ESP32 - BLE scan - asyncio in thread: wifi not works.

straga, 2022-11-12:
ESP32 rev1 and rev3. Board without psram.
Try with ESP-IDF, 4.4.1 and 4.4.3, micropython from master

Suppose running asyncio in the thread. Task call ble.scan. Wifi works not good. At first look ok, But if you try from FTP download/upload some or other. Wifi stops working normally.

[image]

If run asyncio without thread. All works normally.
[image]

Firmware:
[4m_straga_core.zip](https://github.com/micropython/micropython/files/9994533/4m_straga_core.zip)

main.py and ftp.py for test.
[main_ble_ftp.zip](https://github.com/micropython/micropython/files/9994537/main_ble_ftp.zip)

straga, 2022-11-20:
ESP32-C3: same. One-time use of WIFI and BLE is not stable. Wifi work is not correct.

straga, 2022-11-23:
- ESP32 with psram - works, WIFI and BLE same time not problem with uasyncio in the thread too.

- ESP32 without psram - not works normally.

- ESP32-C3 - panic kernel with nimble.

QGB, 2023-05-27:
wifi scan ,ble reboot
```
r=M.wl()

***ERROR*** A stack overflow in task ble has been detected.
Core  0 register dump:
MEPC    : 0x40380796  RA      : 0x4038c2e4  SP      : 0x3fc965a0  GP      : 0x3fc93a00
TP      : 0x3fcb27dc  T0      : 0x4005890e  T1      : 0x74732041  T2      : 0x00000001
S0/FP   : 0x3fc965b4  S1      : 0x80000003  A0      : 0x3fc965b4  A1      : 0x3c121fe4
A2      : 0x002e6465  A3      : 0x3fc965f0  A4      : 0x00000001  A5      : 0x3fc9b000
A6      : 0x7265766f  A7      : 0x206b6361  S2      : 0x3fc9b430  S3      : 0x00000001
S4      : 0x3fcc7db8  S5      : 0x3fcdd518  S6      : 0x00000000  S7      : 0x3c12e318
S8      : 0x000001ab  S9      : 0x3fc9b90c  S10     : 0x3fc9b90c  S11     : 0x00000000
T3      : 0x202a2a2a  T4      : 0x524f5252  T5      : 0x452a2a2a  T6      : 0xffffffff
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc965a0: 0x00000001 0x00000072 0x3fcd3930 0x4038e998 0x3fc9b90c 0x452a2a2a 0x524f5252 0x202a2a2a
3fc965c0: 0x74732041 0x206b6361 0x7265766f 0x776f6c66 0x206e6920 0x6b736174 0x656c6220 0x73616820
3fc965e0: 0x65656220 0x6564206e 0x74636574 0x002e6465 0x00000000 0x00000000 0x00000000 0x00000000
3fc96600: 0x00000780 0x3fca0bf4 0x3fc9b464 0x4038dbb4 0x3fc9b42c 0x60023000 0x3fcc7db8 0x00000001
3fc96620: 0x00001881 0x80000003 0x3fcc7da8 0x4038c360 0x40380214 0x80000003 0x3fcc7da8 0x403801f8
3fc96640: 0x00000001 0x3fc96648 0xffffffff 0x3fc9fccc 0x3fc9fccc 0x00000001 0x3fc9665c 0xffffffff
3fc96660: 0x3fcc4c30 0x3fcc4c30 0x00000000 0x3fc96670 0xffffffff 0x3fc96670 0x3fc96670 0x00000000
3fc96680: 0x3fc96684 0xffffffff 0x3fc96684 0x3fc96684 0x00000000 0x3fc96698 0xffffffff 0x3fc96698
3fc966a0: 0x3fc96698 0x00000000 0x3fc966ac 0xffffffff 0x3fc966ac 0x3fc966ac 0x00000000 0x3fc966c0
3fc966c0: 0xffffffff 0x3fc966c0 0x3fc966c0 0x00000000 0x3fc966d4 0xffffffff 0x3fc966d4 0x3fc966d4
3fc966e0: 0x00000000 0x3fc966e8 0xffffffff 0x3fc966e8 0x3fc966e8 0x00000000 0x3fc966fc 0xffffffff
3fc96700: 0x3fc966fc 0x3fc966fc 0x00000000 0x3fc96710 0xffffffff 0x3fc96710 0x3fc96710 0x00000000
3fc96720: 0x3fc96724 0xffffffff 0x3fc96724 0x3fc96724 0x00000000 0x3fc96738 0xffffffff 0x3fc96738
3fc96740: 0x3fc96738 0x00000000 0x3fc9674c 0xffffffff 0x3fc9674c 0x3fc9674c 0x00000000 0x3fc96760
3fc96760: 0xffffffff 0x3fc96760 0x3fc96760 0x00000000 0x3fc96774 0xffffffff 0x3fc96774 0x3fc96774
3fc96780: 0x00000000 0x3fc96788 0xffffffff 0x3fc96788 0x3fc96788 0x00000000 0x3fc9679c 0xffffffff
3fc967a0: 0x3fc9679c 0x3fc9679c 0x00000000 0x3fc967b0 0xffffffff 0x3fc967b0 0x3fc967b0 0x00000000
3fc967c0: 0x3fc967c4 0xffffffff 0x3fc967c4 0x3fc967c4 0x00000000 0x3fc967d8 0xffffffff 0x3fc967d8
3fc967e0: 0x3fc967d8 0x00000001 0x3fc967ec 0xffffffff 0x3fcde1c8 0x3fcde1c8 0x00000000 0x3fc96800
3fc96800: 0xffffffff 0x3fc96800 0x3fc96800 0x00000001 0x3fc96814 0xffffffff 0x3fcc9ea0 0x3fcc9ea0
3fc96820: 0x00000000 0x3fc96828 0xffffffff 0x3fc96828 0x3fc96828 0x00000001 0x3fc9683c 0xffffffff
3fc96840: 0x3fcc6494 0x3fcc6494 0x00000000 0x3fc96850 0xffffffff 0x3fc96850 0x3fc96850 0x00000000
3fc96860: 0x3fc96864 0xffffffff 0x3fc96864 0x3fc96864 0x00000004 0x3fc96878 0xffffffff 0x3fcc7474
3fc96880: 0x3fcd7fb0 0x00000000 0x3fc9688c 0xffffffff 0x3fc9688c 0x3fc9688c 0x00000000 0x3fc96898
3fc968a0: 0x00000000 0x00000000 0x00000000 0x3fc968b0 0xffffffff 0x3fc968b0 0x3fc968b0 0x00000000
3fc968c0: 0x3fc968c4 0xffffffff 0x3fc968c4 0x3fc968c4 0x00000001 0x00000001 0x00000000 0x0001ffff
3fc968e0: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc968ec 0x00000000 0x00000000 0x00000000
3fc96900: 0x3fc96904 0xffffffff 0x3fc96904 0x3fc96904 0x00000000 0x3fc96918 0xffffffff 0x3fc96918
3fc96920: 0x3fc96918 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000
3fc96940: 0x00000000 0x3fc9cdb8 0x3fc9ce20 0x3fc9ce88 0x00000000 0x00000000 0x00000001 0x00000000
3fc96960: 0x00000000 0x00000000 0x42102646 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc96980: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 6884c813663ec719

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380762
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6f4
load:0x403d0000,len:0x28d8
entry 0x403ce000
WebREPL server started on http://0.0.0.0:8266/
Started webrepl in normal mode
```

straga, 2023-06-03:
ESP32-C3: not work together BLE and WIFI too.
```
import bluetooth
class BLESimple:
    def __init__(self, ble):
        self._ble = ble
        self._ble.active(True)
        self._ble.irq(self._irq)
        self.done = False

    def _irq(self, event, data):
        if event == 5:
            addr_type, addr, adv_type, rssi, adv_data = data
            print("addr_type: {}, addr: {}, adv_type: {}, rssi: {}, adv_data: {}".format(addr_type, addr, adv_type, rssi, adv_data))

        elif event == 6:
            # done scan, need scan again
            print("done scan")
            self.done = True

    def scan(self):
        self._ble.gap_scan(2000, 30000, 30000)

```

```
>>> ble = bluetooth.BLE()
>>> central = BLESimple(ble)
>>> central.scan()

addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -35, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -66, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -26, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -44, adv_data: <memoryview>

But if try connet to WIFI, not connected to wifi and 
E (78580) esp-sha: Failed to allocate buf memory

```

straga, 2023-06-04:
ESP32 PSRAM MicroPython v1.20.0-151-ga61d40e5e -  ESP-IDF v4.3.5
Run uasyncio in thread 
```

stack size : _thread.stack_size(8 * 1024)

addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -63, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -41, adv_data: <memoryview>Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e9278  PS      : 0x00060530  A0      : 0x800e9729  A1      : 0x3ffe9de0
A2      : 0xffffffff  A3      : 0x3ffe9e20  A4      : 0x000034da  A5      : 0x3f82fee0
A6      : 0x3f404660  A7      : 0x3f404660  A8      : 0x800e9274  A9      : 0x3ffe9dc0
A10     : 0x3f82fee0  A11     : 0x3f82fee0  A12     : 0x00060b20  A13     : 0x3ffea0ac
A14     : 0x3f831e90  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x000034da  LBEG    : 0x400951be  LEND    : 0x400951c9  LCOUNT  : 0x00000000

Backtrace:0x400e9275:0x3ffe9de0 0x400e9726:0x3ffe9e20 0x400e8484:0x3ffe9e50 0x40085e52:0x3ffe9e80 0x400e20c2:0x3ffe9f20 0x400e8453:0x3ffe9f40 0x40085e52:0x3ffe9f70 0x400e20c2:0x3ffea010 0x400e8453:0x3ffea030 0x40085e52:0x3ffea060 0x400e20c2:0x3ffea100 0x400e210b:0x3ffea120 0x400e2172:0x3ffea150 0x400e1c56:0x3ffea170 0x400e81fe:0x3ffea190 0x400e834e:0x3ffea1b0 0x40085c7d:0x3ffea1d0 0x400e1d54:0x3ffea270 0x400e81fe:0x3ffea2a0 0x40085bcd:0x3ffea2c0 0x400e1d54:0x3ffea360 0x400e81fe:0x3ffea3b0 0x4010d2ef:0x3ffea3d0 0x400dd04d:0x3ffea470

stack size : _thread.stack_size(12 * 1024)

>>> addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -30, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 2, rssi: -65, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -28, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -28, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -30, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 2, rssi: -66, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -30, adv_data: <memoryview>
BLE scan complete.
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -77, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -81, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -30, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -28, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 2, rssi: -64, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -28, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -67, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -28, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -60, adv_data: <memoryview>
BLE scan complete.
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -64, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -28, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -44, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -28, adv_data: <memoryview>
addr_type: 1, addr: <memoryview>, adv_type: 3, rssi: -62, adv_data: <memoryview>
Guru Meditation Error: Core  1 panic'ed (InstructionFetchError). Exception was unhandled.

Core  1 register dump:
PC      : 0x3f82cc10  PS      : 0x00060b30  A0      : 0x80085e55  A1      : 0x3ffeae30
A2      : 0x3f831090  A3      : 0x00000006  A4      : 0x3f82cc10  A5      : 0x3ffeaeac
A6      : 0x00000100  A7      : 0x3f82f860  A8      : 0x800e8487  A9      : 0x3ffeae10
A10     : 0x3f831090  A11     : 0x3ffeb538  A12     : 0x00060b20  A13     : 0x00000000
A14     : 0x3f82f0a8  A15     : 0x3f82f0bc  SAR     : 0x00000000  EXCCAUSE: 0x00000002
EXCVADDR: 0x3f82cc10  LBEG    : 0x400839fd  LEND    : 0x40083a05  LCOUNT  : 0x00000026

Backtrace:0x3f82cc0d:0x3ffeae30 |<-CORRUPTED

```

naftaly-fontyn, 2023-10-17:
Similar issue 
Simple program to setup wifi connect to AP , deactivate and the try to setup BLE crashes.
I am using a ESP32 WROOM mini
<sample program>
import time
import network
import bluetooth
# import ntptime
from micropython import const

_IRQ_CENTRAL_CONNECT = const(1)
_IRQ_CENTRAL_DISCONNECT = const(2)
_IRQ_GATTS_WRITE = const(3)
_IRQ_GATTS_READ_REQUEST = const(4)
_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)
_IRQ_PERIPHERAL_CONNECT = const(7)
_IRQ_PERIPHERAL_DISCONNECT = const(8)
_IRQ_GATTC_SERVICE_RESULT = const(9)
_IRQ_GATTC_SERVICE_DONE = const(10)
_IRQ_GATTC_CHARACTERISTIC_RESULT = const(11)
_IRQ_GATTC_CHARACTERISTIC_DONE = const(12)
_IRQ_GATTC_DESCRIPTOR_RESULT = const(13)
_IRQ_GATTC_DESCRIPTOR_DONE = const(14)
_IRQ_GATTC_READ_RESULT = const(15)
_IRQ_GATTC_READ_DONE = const(16)
_IRQ_GATTC_WRITE_DONE = const(17)
_IRQ_GATTC_NOTIFY = const(18)
_IRQ_GATTC_INDICATE = const(19)
_IRQ_GATTS_INDICATE_DONE = const(20)
_IRQ_MTU_EXCHANGED = const(21)
_IRQ_L2CAP_ACCEPT = const(22)
_IRQ_L2CAP_CONNECT = const(23)
_IRQ_L2CAP_DISCONNECT = const(24)
_IRQ_L2CAP_RECV = const(25)
_IRQ_L2CAP_SEND_READY = const(26)
_IRQ_CONNECTION_UPDATE = const(27)
_IRQ_ENCRYPTION_UPDATE = const(28)
_IRQ_GET_SECRET = const(29)
_IRQ_SET_SECRET = const(30)

def bt_irq(event, data):
    if event == _IRQ_CENTRAL_CONNECT:
        # A central has connected to this peripheral.
        conn_handle, addr_type, addr = data
    elif event == _IRQ_CENTRAL_DISCONNECT:
        # A central has disconnected from this peripheral.
        conn_handle, addr_type, addr = data
    elif event == _IRQ_GATTS_WRITE:
        # A client has written to this characteristic or descriptor.
        conn_handle, attr_handle = data
    elif event == _IRQ_GATTS_READ_REQUEST:
        # A client has issued a read. Note: this is only supported on STM32.
        # Return a non-zero integer to deny the read (see below), or zero (or None)
        # to accept the read.
        conn_handle, attr_handle = data
    elif event == _IRQ_SCAN_RESULT:
        # A single scan result.
        addr_type, addr, adv_type, rssi, adv_data = data
    elif event == _IRQ_SCAN_DONE:
        # Scan duration finished or manually stopped.
        pass
    elif event == _IRQ_PERIPHERAL_CONNECT:
        # A successful gap_connect().
        conn_handle, addr_type, addr = data
    elif event == _IRQ_PERIPHERAL_DISCONNECT:
        # Connected peripheral has disconnected.
        conn_handle, addr_type, addr = data
    elif event == _IRQ_GATTC_SERVICE_RESULT:
        # Called for each service found by gattc_discover_services().
[... 95 lines cut ...]
ble.irq(bt_irq)
print(ble.active(True))
# ble.gap_scan()

<result on consol>
MPY: soft reboot
Wifi Connected
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40113c5a  PS      : 0x00060d30  A0      : 0x800dd295  A1      : 0x3fff05f0  
A2      : 0x3fff0740  A3      : 0x3fff07e0  A4      : 0x0000008f  A5      : 0x000005d1  
A6      : 0x3f4047dc  A7      : 0x00000001  A8      : 0x00000c33  A9      : 0x3fff05d0  
A10     : 0x3fff07e0  A11     : 0x3ffe7b78  A12     : 0x000000bb  A13     : 0x00000040  
A14     : 0xffffffff  A15     : 0x00000003  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000c33  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40113c57:0x3fff05f0 0x400dd292:0x3fff0610 0x400dff4c:0x3fff0630 0x400e7405:0x3fff0660 0x400e74a2:0x3fff0680 0x400eb596:0x3fff06b0 0x400eb618:0x3fff0730 0x400eb6ba:0x3fff07d0 0x400eb8c9:0x3fff0840 0x400f3861:0x3fff08a0 0x4011a999:0x3fff08e0 0x4011ac6e:0x3fff0900 0x4011ac77:0x3fff0920 0x400954c1:0x3fff0940 0x40085b55:0x3fff0960 0x400ff7ea:0x3fff0980

ELF file SHA256: 2d2ed3fbef6a52cf

Rebooting...
MicroPython v1.21.0 on 2023-10-05; Generic ESP32 module with ESP32
Type "help()" for more information.
>>>