micropython/micropython #10005: W5100-EVB-Pico failed to enumerate USB port on Linux

qashar, 2022-11-17:
Dear all,
I got an issue while working with the board in the subject. Basically the Micropython firmware from the download page is not able to connect through USB under Linux. I am able to enter the bootsel mode, upload the firmware but nothing appears on lsusb after the boards reboots itself. This does not happens when using Windows or OSX. Again on Linux the plain Raspberry Pico firmware does not show this problem. I prefer not use the 

Here's the kernel log when the plain Raspberry Pico FW is loaded (Mass storage, load fw, then seen as board in FS mode):
> [164457.016923] usb 1-2: new full-speed USB device number 86 using xhci_hcd
> [164457.017134] usb 1-2: Device not responding to setup address.
> [164457.225136] usb 1-2: Device not responding to setup address.
> [164457.432902] usb 1-2: device not accepting address 86, error -71
> [164457.560946] usb 1-2: new full-speed USB device number 87 using xhci_hcd
> [164457.561156] usb 1-2: Device not responding to setup address.
> [164457.769126] usb 1-2: Device not responding to setup address.
> [164457.976923] usb 1-2: device not accepting address 87, error -71
> [164457.977075] usb usb1-port2: unable to enumerate USB device
> [164522.968740] usb 1-2: new full-speed USB device number 88 using xhci_hcd
> [164523.118421] usb 1-2: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
> [164523.118427] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [164523.118431] usb 1-2: Product: RP2 Boot
> [164523.118435] usb 1-2: Manufacturer: Raspberry Pi
> [164523.118438] usb 1-2: SerialNumber: E0C9125B0D9B
> [164523.120300] usb-storage 1-2:1.0: USB Mass Storage device detected
> [164523.121550] scsi host6: usb-storage 1-2:1.0
> [164524.145564] scsi 6:0:0:0: Direct-Access     RPI      RP2              3    PQ: 0 ANSI: 2
> [164524.146252] sd 6:0:0:0: Attached scsi generic sg1 type 0
> [164524.146890] sd 6:0:0:0: [sdb] 262144 512-byte logical blocks: (134 MB/128 MiB)
> [164524.147296] sd 6:0:0:0: [sdb] Write Protect is off
> [164524.147302] sd 6:0:0:0: [sdb] Mode Sense: 03 00 00 00
> [164524.147674] sd 6:0:0:0: [sdb] No Caching mode page found
> [164524.147697] sd 6:0:0:0: [sdb] Assuming drive cache: write through
> [164524.177696]  sdb: sdb1
> [164524.210014] sd 6:0:0:0: [sdb] Attached SCSI removable disk
> [164534.858799] usb 1-2: USB disconnect, device number 88
> [164534.859804] blk_update_request: I/O error, dev sdb, sector 260 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
> [164534.859816] Buffer I/O error on dev sdb1, logical block 259, lost async page write
> [164534.896006] FAT-fs (sdb1): unable to read boot sector to mark fs as dirty
> [164535.228698] usb 1-2: new full-speed USB device number 89 using xhci_hcd
> [164535.478547] usb 1-2: New USB device found, idVendor=2e8a, idProduct=0005, bcdDevice= 1.00
> [164535.478555] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [164535.478559] usb 1-2: Product: Board in FS mode
> [164535.478563] usb 1-2: Manufacturer: MicroPython
> [164535.478566] usb 1-2: SerialNumber: e660d441570e7e2e
> [164535.480518] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
> [165092.843126] usb 1-2: USB disconnect, device number 89

And here's the kernel log when the W5100-EVB-PICO FW is loaded (Mass storage, load fw, then **USB failed to enumerate**):
> [164401.161130] usb 1-2: new full-speed USB device number 83 using xhci_hcd
> [164401.310616] usb 1-2: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
> [164401.310623] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [164401.310627] usb 1-2: Product: RP2 Boot
> [164401.310631] usb 1-2: Manufacturer: Raspberry Pi
> [164401.310634] usb 1-2: SerialNumber: E0C9125B0D9B
> [164401.312614] usb-storage 1-2:1.0: USB Mass Storage device detected
> [164401.313156] scsi host6: usb-storage 1-2:1.0
> [164402.322147] scsi 6:0:0:0: Direct-Access     RPI      RP2              3    PQ: 0 ANSI: 2
> [164402.322759] sd 6:0:0:0: Attached scsi generic sg1 type 0
> [164402.323948] sd 6:0:0:0: [sdb] 262144 512-byte logical blocks: (134 MB/128 MiB)
> [164402.324593] sd 6:0:0:0: [sdb] Write Protect is off
> [164402.324599] sd 6:0:0:0: [sdb] Mode Sense: 03 00 00 00
> [164402.325063] sd 6:0:0:0: [sdb] No Caching mode page found
> [164402.325079] sd 6:0:0:0: [sdb] Assuming drive cache: write through
> [164402.354537]  sdb: sdb1
> [164402.370844] sd 6:0:0:0: [sdb] Attached SCSI removable disk
> [164453.541431] usb 1-2: USB disconnect, device number 83
> [164453.542478] blk_update_request: I/O error, dev sdb, sector 260 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
> [164453.542489] Buffer I/O error on dev sdb1, logical block 259, lost async page write
> [164453.580727] FAT-fs (sdb1): unable to read boot sector to mark fs as dirty
> [164453.916961] usb 1-2: new full-speed USB device number 84 using xhci_hcd
> **[164454.865447] usb 1-2: device descriptor read/all, error -71
> [164454.992904] usb 1-2: new full-speed USB device number 85 using xhci_hcd
> [164455.700957] usb 1-2: device descriptor read/64, error -71
> [164456.181134] usb 1-2: Device not responding to setup address.
> [164456.389133] usb 1-2: Device not responding to setup address.
> [164456.600939] usb 1-2: device not accepting address 85, error -71
> [164456.601052] usb usb1-port2: attempt power cycle**

Looking at the error, I thought the board could be broken, but I have been using that with other machines without problem.
Also the issue is limited to USB communication because some python code loaded before still runs fine (tested communicating through ethernet).

Many thanks for any advice or suggestion on where to look on the code and try to solve the problem.

Have a nice day!

dpgeorge, 2022-11-18:
In the working case (the first case above), these lines look suspicious:
```
[164534.859804] blk_update_request: I/O error, dev sdb, sector 260 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[164534.859816] Buffer I/O error on dev sdb1, logical block 259, lost async page write
[164534.896006] FAT-fs (sdb1): unable to read boot sector to mark fs as dirty
```
Did you properly unmount the drive before resetting?

Try rebooting your Linux PC, sometimes its USB drivers can get into a broken state.

qashar, 2022-11-18:
Hi, thanks for the quick reply.
I reboot the PC and still the same outcome, see log below.

I don't fully understand your question: the drive unmount by itself after copying the fw, so there is no "properly unmount" or "reset" to do.

In addition I tried also on another Linux PC and I got the same behavior.

Log:
(connect a board with Pico firmware to Linux PC Debian 5.10.149-2 (2022-10-21) x86_64)
> [   60.236989] usb 1-2: new full-speed USB device number 4 using xhci_hcd
> [   60.508151] usb 1-2: New USB device found, idVendor=2e8a, idProduct=0005, bcdDevice= 1.00
> [   60.508159] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [   60.508163] usb 1-2: Product: Board in FS mode
> [   60.508166] usb 1-2: Manufacturer: MicroPython
> [   60.508169] usb 1-2: SerialNumber: e660d441570e7e2e
> [   60.539257] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
> [   60.539569] usbcore: registered new interface driver cdc_acm
> [   60.539570] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

(disconnect the board, reconnect with bootsel)
> [  199.579185] usb 1-2: USB disconnect, device number 4
> [  206.321968] usb 1-2: new full-speed USB device number 5 using xhci_hcd
> [  206.470923] usb 1-2: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
> [  206.470931] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  206.470935] usb 1-2: Product: RP2 Boot
> [  206.470938] usb 1-2: Manufacturer: Raspberry Pi
> [  206.470942] usb 1-2: SerialNumber: E0C9125B0D9B
> [  206.492474] usb-storage 1-2:1.0: USB Mass Storage device detected
> [  206.492761] scsi host6: usb-storage 1-2:1.0
> [  206.492858] usbcore: registered new interface driver usb-storage
> [  206.499241] usbcore: registered new interface driver uas
> [  207.495102] scsi 6:0:0:0: Direct-Access     RPI      RP2              3    PQ: 0 ANSI: 2
> [  207.495388] sd 6:0:0:0: Attached scsi generic sg1 type 0
> [  207.495806] sd 6:0:0:0: [sdb] 262144 512-byte logical blocks: (134 MB/128 MiB)
> [  207.496100] sd 6:0:0:0: [sdb] Write Protect is off
> [  207.496102] sd 6:0:0:0: [sdb] Mode Sense: 03 00 00 00
> [  207.496407] sd 6:0:0:0: [sdb] No Caching mode page found
> [  207.496413] sd 6:0:0:0: [sdb] Assuming drive cache: write through
> [  207.514673]  sdb: sdb1
> [  207.538837] sd 6:0:0:0: [sdb] Attached SCSI removable disk

(load board specific firmware in the mass storage, the rp2 restarts)
> [  280.201701] usb 1-2: USB disconnect, device number 5
> [  280.203605] blk_update_request: I/O error, dev sdb, sector 260 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
> [  280.203609] Buffer I/O error on dev sdb1, logical block 259, lost async page write
> [  280.218218] FAT-fs (sdb1): unable to read boot sector to mark fs as dirty
> [  280.641258] usb 1-2: new full-speed USB device number 6 using xhci_hcd
> [  281.629710] usb 1-2: device descriptor read/all, error -71
> [  281.757266] usb 1-2: new full-speed USB device number 7 using xhci_hcd
> [  282.497303] usb 1-2: device descriptor read/64, error -71
> [  283.002059] usb 1-2: Device not responding to setup address.
> [  283.209584] usb 1-2: Device not responding to setup address.
> [  283.417381] usb 1-2: device not accepting address 7, error -71
> [  283.417528] usb usb1-port2: attempt power cycle
> [  283.829393] usb 1-2: new full-speed USB device number 8 using xhci_hcd
> [  283.829690] usb 1-2: Device not responding to setup address.
> [  284.038040] usb 1-2: Device not responding to setup address.
> [  284.245334] usb 1-2: device not accepting address 8, error -71
> [  284.373388] usb 1-2: new full-speed USB device number 9 using xhci_hcd
> [  284.373694] usb 1-2: Device not responding to setup address.
> [  284.582028] usb 1-2: Device not responding to setup address.
> [  284.789393] usb 1-2: device not accepting address 9, error -71
> [  284.789535] usb usb1-port2: unable to enumerate USB device

Plaque-fcc, 2022-11-19:
Is this your only board of this type? If not, does this happen to all boards with your test environment?
Please consider clarifying which version of the firmware produces the errors you're observing and which one doesn't.
Also, if it happens to you systematically, please document the steps you perform and the outcome you expect and have.
Nobody knows your GNU/Linux distro flavour and version, and libs versions, some specific (very old or bizzare) setups can behave unexpectedly and not obviously.
Btw, have you tried `picotool` to flash the firmware onto your RP2040 board? If yes, does it produce the very same result?

qashar, 2022-11-26:
@Plaque-fcc (sorry for the late reply)
- This is the only board I can connect to this setup. I will try to get another one, but no guarantee
- I tried to load the following firmware
1. rp2-pico-20220618-v1.19.1.uf2 last stable release
2. rp2-pico-20221103-unstable-v1.19.1-613-g2d406f222, nightly from Nov 6
3. W5100S_EVB_PICO-20220618-v1.19.1 last stable release
4. W5100S_EVB_PICO-20221103-unstable-v1.19.1-613-g2d406f222, nightly from Nov 6
    As said before the problem showed up with W5100S fw, but not with rp2-pico ones, in the same way for both stable and nightly 
- The problem happens systematically as described above, I believe I described accurately the steps: I just load the firmware and try to connect to the board using the serial through USB. Which step is not clear to you?
- The Linux Distro is Debian 11, Linux PC Debian 5.10.149-2 (2022-10-21) x86_64
- I still haven't had time to install picotool and try that, I will let u know

qashar, 2023-02-21:
Hi again,
just to to close this and put some log for future ref.

The problem disappeared after using a more recent firmware (nightly):
- W5100S_EVB_PICO-20221103-unstable-v1.19.1-613-g2d406f222, failed to enumerate
- W5100S_EVB_PICO-20230207-unstable-v1.19.1-854-g35524a6fd, no issue

Thanks everybody for the support and for Micropython.

R

closed 2023-02-21 by qashar

jimmo, 2023-02-23:
> The problem disappeared after using a more recent firmware (nightly):

@qashar Thanks, yes the issue you're seeing was fixed a few months ago. Sorry it's taken a while to get the next release out.