micropython/micropython #9993: uasyncio: Behaviour differs from CPython.

peterhinch, 2022-11-17:
Consider this script:
```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio

evt = asyncio.Event()

async def foo():
    try:
        await evt.wait()
    finally:
        print('Finally')

async def main():
    t = asyncio.create_task(foo())
    await asyncio.sleep(1)
    t. cancel()
    print('canned')
    await asyncio.sleep(1)

asyncio.run(main())
```
On MP the outcome is:
```
MicroPython v1.19.1-617-g43dd3ea74 on 2022-11-08; linux [GCC 9.4.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import rats
canned
Finally
>>> 
```
On CPython:
```
$ python3
Python 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rats
Finally
canned
>>> 
```
MP behaviour also confirmed on Pyboard.

dpgeorge, 2022-11-17:
I'm using Python 3.10.8 and it behaves the same as MicroPython for this example, printing "canned" before "Finally".

jimmo, 2022-11-17:
I see `Finally`, `canned` on 3.8.12, and `canned`, `Finally` on 3.10.8 (and MicroPython stm32/unix).

dpgeorge, 2022-11-17:
It would be interesting to understand why CPython changed this behaviour.  But at the very least it shows that precise ordering of output on asyncio is not a strict concern for CPython.

fivdi, 2022-11-19:
Here's what I see with Python 3.9.2:
```
$ python --version
Python 3.9.2
$ python b.py 
Finally
canned
$ 
```
And here's what I see with MicroPython v1.19.1-705-gac5934c96-dirty on an ESP32C3:
```
$ mpremote connect /dev/ttyACM0 run b.py 
canned
Finally
$ 
```

The behaviour of MicroPython and Python are actually quite different here. When Python is used, there's a `RuntimeError` that gets swallowed. When MicroPython is used, there is no error. This can be seen with the following variant of the program where three lines of code have been added:

```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio

evt = asyncio.Event()

async def foo():
    global evt             # added
    try:
        await evt.wait()
    except Exception as e: # added
        print(f'{e}')      # added
    finally:
        print('Finally')

async def main():
    t = asyncio.create_task(foo())
    await asyncio.sleep(1)
    t. cancel()
    # Can be fixed with .sleep(0) here
    print('canned')
    await asyncio.sleep(1)

asyncio.run(main())
```

With the modified variant here's what I see with Python 3.9.2:
```
$ python --version
Python 3.9.2
$ python a.py 
Task <Task pending name='Task-2' coro=<foo() running at /home/pi/src/junk/a.py:11>> got Future <Future pending> attached to a different loop
Finally
canned
$ 
```
And here's what I see with MicroPython v1.19.1-705-gac5934c96-dirty on an ESP32C3:
``` 
$ mpremote connect /dev/ttyACM0 run a.py 
canned
Finally
$ 
```

Python doesn't like the fact that `await evt.wait()` is being called in an event loop that is not the event loop in which `evt` was created. MicroPython doesn't object to this.

The following variant of the program which avoids the global `evt` also avoids the problems and gives identical output in both Python and MicroPython, that is "canned" followed by "Finally".

```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio

async def foo(evt):
    try:
        await evt.wait()
    except Exception as e:
        print(f'{e}')
    finally:
        print('Finally')

async def main():
    evt = asyncio.Event()
    t = asyncio.create_task(foo(evt))
    await asyncio.sleep(1)
    t. cancel()
    # Can be fixed with .sleep(0) here
    print('canned')
    await asyncio.sleep(1)

asyncio.run(main())
```

fivdi, 2022-11-20:
> It would be interesting to understand why CPython changed this behaviour. But at the very least it shows that precise ordering of output on asyncio is not a strict concern for CPython.

I took another look at this to see if I could figure out why CPython changed its behaviour.

In Python 3.8.10 the Event constructor sets it _loop attribute. This can be seen [here](https://github.com/python/cpython/blob/v3.8.10/Lib/asyncio/locks.py#L247-L265):

```python
class Event:
    """Asynchronous equivalent to threading.Event.
    Class implementing event objects. An event manages a flag that can be set
    to true with the set() method and reset to false with the clear() method.
    The wait() method blocks until the flag is true. The flag is initially
    false.
    """

    def __init__(self, *, loop=None):
        self._waiters = collections.deque()
        self._value = False
        if loop is None:
            self._loop = events.get_event_loop()
        else:
            self._loop = loop
            warnings.warn("The loop argument is deprecated since Python 3.8, "
                          "and scheduled for removal in Python 3.10.",
                          DeprecationWarning, stacklevel=2)
```

Then there is this bug report https://bugs.python.org/issue42392 (https://github.com/python/cpython/issues/86558) which states the following:

> asyncio.Lock and other primitives should no longer accept the `loop` parameter. They should also stop storing the current loop in the `self._loop` attribute. Instead, they should use `get_running_loop()` whenever they need to access the loop.

This bug was fixed in Python 3.10.0 with PR https://github.com/python/cpython/pull/23420.

The Event constructor now looks like [this](https://github.com/python/cpython/blob/v3.10.0/Lib/asyncio/locks.py#L157-L169):

```python
class Event(mixins._LoopBoundMixin):
    """Asynchronous equivalent to threading.Event.
    Class implementing event objects. An event manages a flag that can be set
    to true with the set() method and reset to false with the clear() method.
    The wait() method blocks until the flag is true. The flag is initially
    false.
    """

    def __init__(self, *, loop=mixins._marker):
        super().__init__(loop=loop)
        self._waiters = collections.deque()
        self._value = False
```

@peterhinch: I guess this means that there isn't an issue with MicroPython. There is however a bug in Python prior to version 3.10.0.