micropython/micropython #10024: Improved watchdog interface

H-Sachse, 2022-11-20:
I'd like it if it was possible to have more control over the watchdog.

- allow access to the scratch registers
- allow changing the timeout on the fly (current workaround seems to be reinstanciating a WDT object.)
- add read only property for reading the configured timeout.
- add read only property for reading the tick count.
- allow disabling the watchdog (for e.g. dropping into REPL programmatically)
if one of these features is not available on other controllers they could raise NotImplementedError to keep the watchdog interface similar across boards.

robert-hh, 2022-11-20:
Since the properties of the WDT hardware of various device vary a lot, such an interface set would end up in a list of various not-supported features. So nothing would be gained.
Disabling the watchdog is for instance mostly impossible. The only common functions of hardware watchdogs is enabling and feeding. Some allow changing the timeout. Some allow reading the tick count or raising an early warning alarm. Reading back the configured timeout is obsolete. The code that has set the timeout can remember it. If a feature is not available, there will be already an AttributeError raised, without the need to spend code size for raising an different error. And AFAIK, MicroPython does not support properties for it's internally implemented modules.

karfas, 2022-11-27:
A WDT.deinit() function was proposed (and discussed) in #6631.

The conclusion there: You **must not** disable any serious, mission-critical watchdog (to the extent that the watchdog should not have any possibility to do that).

It's better to start the device in some kind of debug mode and don't enable the WDT in the first place.

H-Sachse, 2022-11-27:
Interesting discussion in #6631, but not helpful for the e.g. boards with shorter WDT timeouts like the RPi Pico-W with about 8 seconds. I have decided to write my own WDT extension class which allows to manipulate the Watchdog registers directly. Hackish but so what.

Also the argument made about not disabling the watchdog does not hold true for the PR2040 because you only need to manipulate one bit in one register to disable the watchdog, if I understand the datasheet correctly. This can easily happen even through SEU or an erroneous register write. 

For real mission critical apps you need an actual hardware watchdog, which really cannot be disabled by software, but then you need a hardware way to disable it for debugging.

closed 2022-11-27 by H-Sachse