micropython/micropython #10060: Raspberry Pi Pico W: WiFi connection issues after UART reads

mjoldfield, 2022-11-23:
I've had trouble getting a Pico W to connect to WiFi after the UART has
received data.

For example the code below often doesn't connect to the network if 
serial data is being sent to pin 2 of the Pico (GPIO 1) when it boots. 
If you remove this connection the code works as expected and connects
to the WiFi network within 10s.

Annoyingly, it's all a bit flaky: sometimes the code does work, and 
sometimes it crashes with an out of memory error. The most common 
outcome is that the WiFi simply fails to connect though.

If you ignore the UART until the WiFi network is connected, it all works 
happily.

I'm running "MicroPython v1.19.1 on 2022-11-18."

Please does anyone have any advice for how to investigate this.

```
import network
import rp2
import time

from machine import UART

uart = UART(0)
for i in range(10):
    print(uart.read(6))

rp2.country('GB')
wlan = network.WLAN(network.STA_IF)
wlan.active(True)

wlan.connect("XXXXXX", "YYYYY")
for i in range(300):
    c = wlan.ifconfig()
    print(f"{i} {c}")
    if c[0] != '0.0.0.0':
        break
    time.sleep(1)
```

To generate the serial data, I'm using another Pico running this code:
```
from machine import UART

uart = UART(0)
while True:
    uart.write("Hello\n")
```

robert-hh, 2023-01-06:
Thr other pico keeps the UART irq busy. Try adding a suitable delay between the writes, or write just a few times.

mjoldfield, 2023-01-06:
Thanks for looking at this. 

The real application is a serial to network gateway, and sadly I don't have control over the serial device.