micropython/micropython #10083: CPython difference when unpacking kwargs

dlech, 2022-11-25:
The following shows a difference with CPython. It works fine if the dicts have no common keys:
```python
def foo(**x):
    print(x)
foo(**a, **b)  # a and b are dicts
```
With these dicts, behaviour differs:
```python
a = {'one': 1, 'two': 2}
z = {'one': 1, 'five': 5}
```
MP:
```python
>>> foo(**a, **z)
{'two': 2, 'one': 1, 'five': 5}
```
CPython:
```python
>>> foo(**a, **z)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: __main__.foo() got multiple values for keyword argument 'one'
```

_Originally posted by @peterhinch in https://github.com/micropython/micropython/discussions/10058#discussioncomment-4239261_

dlech, 2022-11-25:
I see three possibilities here:
1. Make MicroPython behavior match CPython behavior.
2. Same as 1 but put it behind the `MICROPY_CPYTHON_COMPAT` flag to save space on constrained systems.
3. Document this as a CPython difference.

dpgeorge, 2022-11-26:
Is this easy to fix?  There is already code there to check for multiple keyword arguments of the same name.

(At least if you write valid code it works fine, the same as CPython.  It's only if you have an error with duplicate args that MicroPython doesn't match.)

closed 2022-11-28 by dpgeorge