micropython/micropython #10091: PEP448 dict merge syntax not supported

peterhinch, 2022-11-26:
While the PEP584 syntax works fine, the PEP448 version
```python
d = {**a, **b}
```
works under CPython V3.11.0 but not under MP. Is this deliberate?

dpgeorge, 2022-11-26:
We have not implemented the syntax for multiple `**` in dict comprehension.  That requires modifying the grammar and compiler, it's a bit involved.  It may be that we never implement this (I'm not sure it's exactly PEP448 because that did not include unpacking generalisations to dicts initially).

PEP584 was a lot simpler, no change to the syntax/grammar needed, it was just implementing the `|` and `|=` binary operators on dits.

sosi-deadeye, 2022-11-26:
Quote from PEP448:

> The keys in a dictionary remain in a right-to-left priority order, so {**{'a': 1}, 'a': 2, **{'a': 3}} evaluates to {'a': 3}. There is no restriction on the number or position of unpackings.

Most people don't know this syntax and that is why it is rarely asked for.
The Pipe-Operator to merge dicts feels natural and is better known.

dpgeorge, 2022-11-26:
So it might be best that we document this is a difference between MicroPython and CPython, and note that it won't be implemented (and encourage people to use `|` instead).

dlech, 2022-11-26:
Related discussion: https://github.com/micropython/micropython/issues/1329#issuecomment-1084815423