micropython/micropython #10099: Software timer misfiring after gc.collect()

intx82, 2022-11-26:
Hi guys

My board is RP-Pico with firmware: 

```
MicroPython v1.19.1 on 2022-10-14; Raspberry Pi Pico with RP2040
Type "help()" for more information.
```

I found an issue related to garbage collector which break a timer if a timer was created as a local variable.
For example:

```
def test():
    Timer(period=10000, mode=Timer.ONE_SHOT, callback=lambda t: print(t))

>>> test()
>>> Timer(mode=ONE_SHOT, period=268687648, tick_hz=1000000)

>>>
>>> test()
>>> gc.collect()
...
# and after 10secs there is no print message
```

It's sounds like, garbage collector freed timer, and this should happen because timer was a local variable. But from the other side, it's a mistake due to the timer never fired up. 

Is this bug or feature? Which behavior is right?

As for me, more delicate solution will be clearing already finished timers.

If it's a bug, i will try to make some pull request to fix it.

dlech, 2022-11-26:
This makes sense since the return value of `Timer(...)` was never assigned to anything, therefore the GC is can collect it because it is not reachable.

intx82, 2022-11-27:
but creating local variable is not a solution, it also will be freed at the next gc.collect call

```
>>> def test(idx):
...     t = Timer(period=10000, mode=Timer.ONE_SHOT, callback=lambda t: print(t, idx))
...
...
...
>>>
>>> test(0)
>>> Timer(mode=ONE_SHOT, period=268687648, tick_hz=1000000) 0

>>> test(0)
>>> test(1)
>>> gc.collect()
>>>
...
```

My current solution is creating a semaphore (counter) which increments on creating a new timer and decrements on it's firing, when it equals to zero, the garbage collector can be called from a main loop. But it's a kludge.

dpgeorge, 2022-11-27:
`Timer` objects have a finaliser, so if they are collected by the GC then they automatically cancel themselves.

If you don't want tho GC to collect the timer then you need to store it in a "global" variable, or at least a container (eg a list) that lives for as long as you need the timer to stay alive.

For example you can do:
```py
timers = []

def test(...):
    t = Timer(...)
    timers.append(t)
```

intx82, 2022-11-27:
Thanks, this solves my issue. More completely, it will look like:

```
timers = []

def test(**kwargs):
    old_cb = kwargs['cb']
    def _cb(t):
        timers.remove(t)
        old_cb(t)

    kwargs['cb'] = _cb
    t = Timer(**kwargs)
    timers.append(t)
```

Because, on other way, timers list will be expanded at every timer start, and after firing have a broken link inside. Anyway, i never saw such behavior in CPython, but i don't have such cases (tasks) in CPython. 

Thanks guys, i will close the issue. If someone thinks, that this is a not normal GC behavior, please re-open it.

closed 2022-11-27 by intx82