micropython/micropython #10102: TouchPad (GPIO_2) not working esp-wroom-32

manobendro, 2022-11-27:
Can't use touch pin 2 (GPIO 2) as a touchpad it gives a value error.
 
esp32 module: esp-wroom-32
from tested: esp32-idf3-20210202-v1.14.bin , esp32-20220618-v1.19.1.bin

fivdi, 2022-11-27:
This test program work for me on an ESP32-DevKitC-v4 with and ESP-WROOM-32:
```python
from machine import TouchPad, Pin
import time

tp = TouchPad(Pin(2))

print("Touch the pad")
time.sleep(3)
for i in range(10):
    print(f'{i}: {tp.read()}')
    time.sleep(1)

print("Don't touch the pad")
time.sleep(3)
for i in range(10):
    print(f'{i} - {tp.read()}')
    time.sleep(1)
```
Here's the output of the test program:
```
Touch the pad
0: 112
1: 79
2: 45
3: 20
4: 28
5: 47
6: 72
7: 100
8: 114
9: 127
Don't touch the pad
0 - 479
1 - 474
2 - 475
3 - 477
4 - 478
5 - 478
6 - 478
7 - 477
8 - 477
9 - 477
```
Can you post code which can be used to reproduce the error and explain more precisely what "gives a value error" means. Also GPIO2 is a strapping pin. Make sure it's not connected to anything else on your board.

manobendro, 2022-12-24:
I find out the problem. GPIO2 was Pull_Down using 10k resistor.

closed 2022-12-24 by manobendro