micropython/micropython #10105: micropython firmata

crousekimo, 2022-11-27:
I try to compile micropython firmata for pico 2040 w.
I follow the steps of Appendix A.
https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf
However, the generated firmata.uf2 cannot be connected by using putty.
May I know what happened?

robert-hh, 2022-11-27:
So 
- building firmware.uf2 succeeded and 
- you could upload the firmware

Then you tried to connect with Putty. Does Putty complain about the wrong com port? Do you use the right COM port? Did you disable flow control with Putty?

crousekimo, 2022-11-27:
I also compile pico 2040 firmata.
pico 2040 firmata can be connected.
I think that I connect the right com port.
You can watch the following video.
https://youtu.be/JTmaR_DVhJU
The generated uf2 can be downloaded as follows.
https://drive.google.com/file/d/10-pn1HLAkeQuAvg5AOkFi0KchrHPM-Ab/view?usp=sharing

robert-hh, 2022-11-27:
With your firmware image I get a REPL prompt at my Pico W device.

robert-hh, 2023-01-06:
The created image is fine. Build worked.

closed 2023-01-06 by robert-hh

AstroHuster, 2023-05-11:
@crousekimo ,
The google drive file is gone. Can you please post it again?

jimmo, 2023-05-12:
@AstroHuster Official firmware builds are available at https://micropython.org/download/rp2-pico-w/