micropython/micropython #10108: urequests not working in main.py

PhaestusFox, 2022-11-27:
I have been trying to make a sensor that uploads its data to my server every few minutes, It seems to work if I run the code from the thonny editor but when I run the code from the main.py with no computer attached it gets stuck at urequests.post()

light turns off for 10 seconds when running from Thonny but never goes off when run directly from USB power

```
import network
import time
from machine import I2C
from machine import Pin
import gc
import urequests

led = Pin("LED", Pin.OUT)

name = "GreenHouse"
ssid = "TelstraF39C34"
psk = "twfxm4t2fd"

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(ssid, psk)

i2c = I2C(id=1,scl=Pin(27),sda=Pin(26),freq=400000)

led.on()

tel_txt = "name:{name}\ntime:{time}\ntemperature_h:{tmp:.2f}\nhumidity:{hmd:.2f}"
server_ip = "http://192.168.0.100/{}"

board_pin = Pin(22, Pin.OUT)
board_pin.high()

class SHT30:
    def __init__(self, name, addr=68):
        self.name = name
        self.addr = addr
        self.temp = 0
        self.humd = 0
    def read(self, bus):
        bus.writeto(self.addr, b'\x24\x00')
        time.sleep_ms(20)
        res = bus.readfrom(self.addr, 6)
        self.temp = -45 + 175 * (((res[0] << 8) | res[1]) / 65535)
        self.humd = 100 * (((res[3] << 8) | res[4]) / 65535)
        #print("temp={:.2f}\nhumd={:.2f}".format(self.temp, self.humd))
    def send(self):
        urequests.put(url=server_ip.format("telemetry"),data = tel_txt.format(name=self.name, time=time.time(), tmp=self.temp, hmd=self.humd))

sht = SHT30(name)

gc.collect()

if not wlan.isconnected():
    wlan.connect(ssid, psk)

sht.read(i2c)
print("Sent Data: ", sht.send())
led.off()
time.sleep(10)
led.on()
```

code is a little doubled up because it was split across the boot and main but get the same problem no matter where I put the code

fivdi, 2022-11-27:
Does it help if the program waits for the connection by replacing this:
```python
if not wlan.isconnected():
    wlan.connect(ssid, psk)
```
with this:
```python
while not wlan.isconnected():
    print("Waiting to connect:")
    time.sleep(1)
```

karfas, 2022-11-27:
I am pretty sure there are TWO problems.

a) you don't wait until the connect() at the beginning of the program succeeds.
connect() runs in the background!
Your current code starts connect(), does a few milliseconds something else, then calls connect() again.
Just wait a few seconds for the isconnected() and **then** make the http request.
I would also call disconnect() before retrying connect, as I'm pretty sure that the behaviour of multiple connect() calls is not consistent across different ports (is this defined somewhere ?) so the outcome is most likely undefined in your case.

b) "but it works with thonny" is an error on it's own. Just get your code working **without** a fancy IDE - use pyremote, rshell, putty, whatever. Inconvenient, yes, but any code working without thonny will run in thonny, too - but not necessary the other way around.