micropython/micropython #10109: Calling method with multiple named arguements does not raise SyntaxError

dlech, 2022-11-27:
This is good, thank you!  It actually fixes a bug unrelated to double `**` usage.  Eg the following would previously pass without error:
```py
def f(**x):
    print(x)

f(a=1, a=2)
```

Can you please add a test for that case as well?

_Originally posted by @dpgeorge in https://github.com/micropython/micropython/issues/10095#issuecomment-1328186336_

dlech, 2022-11-27:
When I made the test case, I found that CPython raises `SyntaxError` on the line `f(a=1, a=2)`, so raising TypeError at runtime is not correct for this case.

jonnor, 2024-09-02:
In recent MicroPython the example raises TypeError. Is it critical that it raises SyntaxError and not TypeError?

```
MicroPython v1.24.0-preview.276.g1897fe6227.dirty on 2024-09-02; linux [GCC 14.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> def f(**x):
...     print(x)
>>> 
>>> f(a=1, a=2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function got multiple values for argument 'a'
```