micropython/micropython #10123: On Teensy4.1 writing strings longer than 1000ch to a file on SD card disconnects the card

JuhaS2, 2022-11-28:
On Teensy4.1, connection to the SD card breaks randomly if writing 1000 or more chars with one `f.write()` call. Writing at the about same bytes/second speed works fine if each write call writes just 100 characters. As Teensy uses SDIO to connect the card instead of more common SPI, this issue is probably relevant only for the mimxrt port.

Code to reproduce the error:
```
import os
import machine

# bytes to write:
lorem1000 = 'Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor incididunt ut labore et dolore magna aliqua. Et sollicitudin ac orci phasellus egestas tellus rutrum tellus. Amet mauris commodo quis imperdiet massa tincidunt nunc pulvinar. Faucibus nisl tincidunt eget nullam. Quis hendrerit dolor magna eget est lorem. Dui ut ornare lectus sit. Arcu felis bibendum ut tristique et egestas. Malesuada bibendum arcu vitae elementum curabitur vitae. Lorem donec massa sapien faucibus et molestie ac feugiat sed. A scelerisque purus semper eget duis. Leo in vitae turpis massa sed elementum tempus. Urna et pharetra pharetra massa massa. Eget gravida cum sociis natoque penatibus et. Est ullamcorper eget nulla facilisi etiam dignissim diam quis. Risus quis varius quam quisque id diam vel quam elementum. Justo laoreet sit amet cursus sit amet. Facilisis gravida neque convallis a cras semper auctor neque. Sed nisi lacus sed viverra tellus in hac habitasse platea. Felis bibendum ut tris'

print('Mounting SD card.')
sd = machine.SDCard()
os.mount(sd, '/sdcard')

print('Creating empty test.txt file.')
with open('/sdcard/test.txt', 'w') as fid:
    fid.write('')

print('Writing 100.000 chars in 100ch writes')
with open('/sdcard/test.txt', 'a') as fid:
    for i in range(1000):
        fid.write(lorem1000[:100])
print('   Success')

print('Writing 100.000 chars in 1000ch writes')
with open('/sdcard/test.txt', 'a') as fid:
    for i in range(100):
        fid.write(lorem1000)
print('   Success')

print('Writing 100.000 chars in 10000ch writes')
with open('/sdcard/test.txt', 'a') as fid:
    for i in range(10):
        fid.write(lorem1000*10)
print('   Success')
```
Which outputs:
```
Mounting SD card.
Creating empty test.txt file.
Writing 100.000 chars in 100ch writes
   Success
Writing 100.000 chars in 1000ch writes
   Success
Writing 100.000 chars in 10000ch writes
Traceback (most recent call last):
  File "<stdin>", line 54, in <module>
OSError: 28
```

The probability of connection break seems to depend on the length of the messages in single writes. In my tests 100ch messages were written without errors for over 5 hours. With 1000ch messages, the error happens usually within 2-30 minutes of continuous writing.  With 10.000ch messages error happens almost always on first or second write call.

Tested on two different SD cards both with FAT32 file system.

Once connection breaks, `sd.present()` returns `False` and all file operations with the SD card fail until system is restarted. Despite a lot of testing, deinit, init, umount, and mount calls are not able to reconnect the card without restart. 

Our MicroPython is compiled by ourselves to include ulab, but should otherwise be vanilla. Version information shown in the REPL: "MicroPython v1.19.1-682-gd1ed0f161-dirty on 2022-11-16; Teensy 4.1 with MIMXRT1062DVJ6A"

robert-hh, 2022-11-28:
That#s indeed strange, When we had set up the driver, we had a stress test running, writing files in random sizes up to 256k. There were quite many changes since then, and we had a library version change.
But I can create similar results. The funny thing is, that is seems stable if the data block which is written is a multiple of 512 bytes. Funny because it locks up at the driver level, but the driver gets data always as multiples of 512 bytes.
Writing a large block of any size only once is fine as well.

Rolling back to an old version looks stable. e.g. commit d84ebc31f7f00edb47dd417dff6adfd47a880c1

robert-hh, 2022-11-29:
I have a fix for that problem in https://github.com/robert-hh/micropython/tree/mimxrt_sp12. I rand your test code in a loop 1000 times with success, and other tests as well.
That branch includes as well a commit to return RTC the subsecond value as 0.

JuhaS2, 2022-11-30:
Thank you for the quick fix. I can confirm the new version works well with our hardware. I can now successfully write big files and the rtc works as expected.

robert-hh, 2022-12-01:
Fixed by PR #10132

closed 2022-12-14 by robert-hh