micropython/micropython #10150: Compil for ESP32 with a USER_C_MODULES / usqlite

l-atome, 2022-12-03:
Hello, 
I try to compile MicroPython for ESP32 with the usqlite module ()  for an education purpose ... 
It's seems that this module was build with an old version of MicroPython ... Actually, I have the error :
ERROR: Call to MP_REGISTER_MODULE(MP_QSTR_usqlite, usqlite_module, 1) should be MP_REGISTER_MODULE(MP_QSTR_usqlite, usqlite_module)
I'd made the change but i have new errors ...

I try to find old version of the micropython source to try to downgrade my build .... Where can i find them ?
Thanks
Thomas

robert-hh, 2022-12-03:
All versions of MicroPython are still in the repository. Just clone it and run `git tag`, which will list all the versions.
Then you can checkout that version, e.g. with `git checkout v1.17`.

l-atome, 2022-12-07:
thank you very much, your comment was a great help to me...
I downgraded micropython, then gcc, then python and finally managed to insert SQLite in a MicroPython firmware for ESP32

Offering my grade 11 and 12 students, an object (not virtual) on which they can manage sockets, manage a database for 5$ ... Great ...

Ideal for NSI teachers (french curriculum grade 11 and 12) : poo, graph, recursivity, divide and conquer, dynamic programming, tcp, ssl, sql, threat ... The use of ESP32 is going to be great!!!

Translated with www.DeepL.com/Translator (free version)

l-atome, 2022-12-07:
thank you very much, your comment was a great help to me...
I downgraded micropython, then gcc, then python and finally managed to insert SQLite in a MicroPython firmware for ESP32

Offering my grade 11 and 12 students, an object (not virtual) on which they can manage sockets, manage a database for 5$ ... Great ...

Ideal for NSI teachers (french curriculum grade 11 and 12) : poo, graph, recursivity, divide and conquer, dynamic programming, tcp, ssl, sql, threat ... The use of ESP32 is going to be great!!!

dpgeorge, 2022-12-12:
I made a PR to usqlite to make it compile with the latest MicroPython: https://github.com/spatialdude/usqlite/pull/14

spatialdude, 2023-01-02:
Thanks @dpgeorge I have merged the PR, with an additional change to avoid a recursion problem - https://github.com/spatialdude/usqlite/pull/14#issuecomment-1368654511

jonnor, 2024-09-01:
usqlite is a third party module, and issue with building it should be handled in their issue tracker.

Exceptions would be if there is antyhing broken in the user C module system itself (independent of the module being built).

closed 2024-09-01 by jonnor