micropython/micropython #10196: PWM channels do not shut down on exit

itswally1, 2022-12-10:
the problem is the PWM channels do not shut down after closing one program and starting another. 
I am using esp32-d2wd-20220618-v1.19.1.bin on an ESP wroom-32 and using Thonny to program the Micro controller. Sample code follows:
# https://microcontrollerslab.com/servo-motor-raspberry-pi-pico-micropython/
===================================
from time import sleep
from machine import Pin, PWM

pwmL = PWM(Pin(19))
pwmR = PWM(Pin(18))
pwmL.freq(50)
pwmR.freq(50)

try:

    while True:
        for position in range(2000,8000,50):
            pwmL.duty_u16(position)
            pwmR.duty_u16(position)
            print(position)
            sleep(0.1)
        for position in range(8000,2000,-50):
            pwmL.duty_u16(position)
            pwmR.duty_u16(position)   
            sleep(0.1)
except:
    pwmL.duty_u16(0)
    pwmR.duty_u16(0)
    print('stopped')
=======================
After stopping the code above i Open a new tab with the following code:
from machine import Pin, PWM
from time import sleep

pwmLed = PWM(Pin(15))
pwmRed = PWM(Pin(14))

pwmLed.freq(1000)
pwmRed.freq(1000)

try:
    while True:
        for duty in range(65025):
            pwmLed.duty_u16(duty)
            pwmRed.duty_u16(duty)
            sleep(0.0001)
        for duty in range(65025, 0, -1):
            pwmLed.duty_u16(duty)
            pwmRed.duty_u16(duty)
            sleep(0.0001)
except:
    pwmLed.duty_u16(0)
    pwmRed.duty_u16(0)
    print('stopped')
===========================
When I run the second program a servo from the first program will run along with the LEDs fading and brightening.
I run the same programs on a Raspberry pi pico and each routine will terminate and start independently with no interaction on the PWM channels.
The only way I can prevent this is by unplugging the USB and then plugging it back in insuring no external power is supplied to the ESP wroom-32. this charismatic is not seen on any other micro controller I have used.
My guess is the PWM channels are not completely shut down unless the power is removed. 

So why does Micropython on the ESPwroom 32 require a power down to close everything out while Micropython on the other Micro controllers terminate with the except conditions?
It is a pain to unplug and plug in the USB while testing various sections of a program. that have been created as tabs in Thonny.

Also note if I terminate the offending PWM channels before starting the while loop of the second program the offending channels are dormant.

Your thoughts on how to terminate all PWM channels on program termination with out unplugging the USB and removing power when using stop and then loading a new program from a tab in Thonny. 
By the way the problem occurs if the programs are saved to the micro controller and it seems it is only a problem on the ESPwroom-32.

Again your thoughts..

robert-hh, 2022-12-10:
Did you try deinit() to stop a channel?

Kyuchumimo, 2022-12-11:
This behavior is normal. When you set a pin high or a PWM signal, it remains persistent, unless you revert the changes.
Instead of using pwmL.duty_u16(0), use pwmL.deinit(), and in the exception use KeyboardInterrupt

```
except KeyboardInterrupt:
    pwmL.deinit()
    pwmR.deinit()
    print('stopped')
```

itswally1, 2022-12-11:
Thanks for the tip. After your response I  reviewed the docs and found that:
PWM.deinit()
Turn off PWM. After using the PWM, you need to log out deinit() 
=====================
Surprisingly I guess I was lucky on the raspberry pi pico this was not necessary. I recently switched to the ESPwroom32 because of the wifi and bluetooth availability included cheap price.
Again Thanks for the solution.

itswally1, 2022-12-12:
Thanks for your quick responce Regards,Don From: KyuchumimoSent: Saturday, December 10, 2022 8:32 PMTo: micropython/micropythonCc: itswally1; AuthorSubject: Re: [micropython/micropython] PWM channels do not shut down on exit (Issue #10196) This behavior is normal. When you set a pin high or a PWM signal, it remains persistent, unless you revert the changes.Instead of using pwmL.duty_u16(0), use pwmL.deinit(), and in the exception use KeyboardInterruptexcept KeyboardInterrupt:    pwmL.deinit()    pwmR.deinit()    print('stopped')

itswally1, 2022-12-12:
Thanks that worked.Regards,Don  From: Robert HammelrathSent: Saturday, December 10, 2022 12:46 PMTo: micropython/micropythonCc: itswally1; AuthorSubject: Re: [micropython/micropython] PWM channels do not shut down on exit (Issue #10196) Did you try deinit() to stop a channel?

itswally1, 2022-12-13:
Thanks again however I found there are two types of ESP wroom-32The one with the USB-cCconnector does not respond to the KeyboardInterrupt: deinit while the one with the mini usb connector does. My only conclusion is the two different drivers on the board are affecting the outcome.Your thoughts..Regards,D Sent from Mail for Windows From: KyuchumimoSent: Saturday, December 10, 2022 8:32 PMTo: micropython/micropythonCc: itswally1; AuthorSubject: Re: [micropython/micropython] PWM channels do not shut down on exit (Issue #10196) This behavior is normal. When you set a pin high or a PWM signal, it remains persistent, unless you revert the changes.Instead of using pwmL.duty_u16(0), use pwmL.deinit(), and in the exception use KeyboardInterruptexcept KeyboardInterrupt:    pwmL.deinit()    pwmR.deinit()    print('stopped')