micropython/micropython #10205: RPI Pico W: Pin("LED", Pin.OUT).on() crashes/hangs

axxel, 2022-12-12:
My Raspberry Pi Pico W worked for a day or so (on rp2-pico-w-20221209-unstable-v1.19.1-740-gbf49a087b.uf2). Then something happened and I don't know what. The following script makes it hang/crash:
```
from machine import Pin
led = Pin("LED", Pin.OUT)
# all good until here
led.on()
# never reaching this line and the led stays off
```

If I replace `"LED"` with `0` it does not crash and the GPIO_0 pin becomes high as expected.

Another hint that something with the wifi module is off:
```
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
print(wlan.active()) # prints: 'False'
```

I saw #8904 but I concluded that I have another issue, since it is not related to it being connected to a Mac/Linux USB stack, the mentioned fix there does not help and I ran into it with the latest build.

For a lack of any other idea, a wild guess would be that my external power supply semi-fired it (although I examined it with an oscilloscope and everything looks as it is supposed to from what I can tell). I would appreciate any ideas for further debugging, including using my osci to peak in on the communication between the uC and the wifi module to find out if there is indeed a hardware defect.

robert-hh, 2022-12-17:
At the PICO_W the LED is connected to the WiFi module. So if that is off, the LED will not go on. What it the state of GPIO23, which switches WiFi on and off?

axxel, 2022-12-17:
Thanks for your reply. I would assume that GPIO23 would be set to `on` implicitly when creating a `Pin("LED")` object? Anyway, I now explicitly set pin 23 high but that does not change anything. I fear I really did break the unit: I found out that GP27_A0 is dead (always low). I consider this to be a confirmation of my 'broken hardware' theory. It is conceivable that this pin has been subjected to more than 3v3 during my experimentation and thereby killed the whole WiFi module via its connection to pin 23?

Anyway. I give up on that unit and would like to apologize for wasting your time.

closed 2022-12-17 by axxel