micropython/micropython #10221: Item deletion support for `bytearray` object

AmirHmZz, 2022-12-14:
Is there any reason that `list` supports item deletion but `bytearray` doesn't?

```Python
# list example
>>> ls= [1, 2, 3, 4]
>>> del ls[-2:]
>>> ls
[1, 2]

# bytearray example
>>> barr = bytearray(b"1234")
>>> del barr[-2:]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'bytearray' object doesn't support item deletion

```

**CPython** already supports item deletion for `bytearray`. I've also searched [_MicroPython differences from CPython_](http://docs.micropython.org/en/latest/genrst/index.html) but I didn't find anything about this one.

jimmo, 2022-12-14:
See objarray.c, there's a TODO for this: https://github.com/micropython/micropython/blob/master/py/objarray.c#L435

FWIW, if you need to be able to do this, you can use slice assignment as an alternative:

```py
>>> b = bytearray(range(5))
>>> b
bytearray(b'\x00\x01\x02\x03\x04')
>>> b[2:3] = b''
>>> b
bytearray(b'\x00\x01\x03\x04')
```

AmirHmZz, 2022-12-14:
@jimmo Thanks for your answer. Is slice assignment as efficient as slice deletation? I just want to delete a slice from beginning of bytearray.

AmirHmZz, 2022-12-14:
@jimmo I have also another related question but I don't want to create a whole issue or discussion for it:
I have a bytearray which I extend it on every X event and delete a slice from the beginning of it. I use this bytearray as an in-memory ttl-cache which all records have the same ttl. So every second I delete a slice from the beginning of it in order to remove expired items. Can I pin this bytearray to a special part of RAM in order to make it easier to find next free blocks on each extend?

jimmo, 2022-12-15:
> @jimmo Thanks for your answer. Is slice assignment as efficient as slice deletation? I just want to delete a slice from beginning of bytearray.

All slice operations have the cost of having to allocate a slice object (although I've been trying to fix that recently -- #10160). So slice assignment and slice deletion are very similar -- the main cost (after the slice allocation) is likely dominated shifting the data down to fill the gap. But your original question was about item deletion, so that will be cheaper overall.

> I have a bytearray which I extend it on every X event and delete a slice from the beginning of it. I use this bytearray as an in-memory ttl-cache which all records have the same ttl. So every second [...]

This really sounds like you want a collections.deque?

AmirHmZz, 2022-12-15:
@jimmo `collections.deque` doesn't let me search in records. That also allocates lots of RAM per item. After my measurements `deque` takes RAM as much as `list` does. but `bytearray` allocates much less RAM.

jimmo, 2022-12-15:
Maybe a ring buffer instead? Fixed allocation, just move the start and end indices as you add/remove bytes. Use slice assignment to memcpy ranges into the ringbuffer.

But to answer your question, no, there's no way to reliably hint the allocator in this way. (I say "reliably" because there are lots of things where it tries to grow allocations into adjacent free regions and you could over-allocate and then shrink, and then hope that the region is still free when you next need to allocate).

But the best thing to do is just to not grow/shrink at runtime and have a fixed allocation.

AmirHmZz, 2022-12-15:
@jimmo Thanks for your answer. Where can I read about ring buffer and the way to use it?

jimmo, 2022-12-15:
> @jimmo Thanks for your answer. Where can I read about ring buffer and the way to use it?

It's just a data structure (more specifically, a way of using a fixed allocated array/buffer to support double-ended operations).

Not micropython-specific, although see https://github.com/micropython/micropython/pull/9458 for a proposal to add a simple one.

AmirHmZz, 2022-12-15:
@jimmo Thanks for your help. That means a lot to me.
I will let this issue open until slice deletion gets implemented.