micropython/micropython #10226: RP2: 1us glitch on SCK pin when switching from SPI function to Pin.OUT

pdg137, 2022-12-15:
On the RP2040, several sets of pins can be selected for each of the two SPI modules.  Sometimes we need to disable the SPI function of a pin, while leaving it in the state it was in.  For example the following code should leave pin 14 (SCK) low between `write()` and `sck(1)`:

```python
from machine import SPI, Pin

sck = Pin(14)
mosi = Pin(15)
spi = SPI(id=1, baudrate=100000, polarity=0, phase=0, sck=sck, mosi=mosi)
spi.write(bytearray([0xAA]))
sck.init(mode=Pin.OUT, value=0)
sck(1)
```

However, the *first* time it runs after reset, there's a ~1 microsecond glitch during which pin 14 floats upward. The glitch causes unreliable behavior with SPI devices if you don't take measures to avoid it (the workaround is to run `sck.init(mode=Pin.OUT, value=0)` at least once *before* using SPI.)

I tested this on the latest master branch, MicroPython v1.19.1-716-g227ea47f4, with a Raspberry Pi Pico having a 1k pull-up on pin 14.  Here is a trace of the output:

[image]

We can zoom in on it:

[image]

I believe the source of the problem is here:

https://github.com/micropython/micropython/blob/master/ports/rp2/mphalport.h#L99-L103

```c
static inline void mp_hal_pin_output(mp_hal_pin_obj_t pin) {
    gpio_set_function(pin, GPIO_FUNC_SIO);
    gpio_set_dir(pin, GPIO_OUT);
    machine_pin_open_drain_mask &= ~(1 << pin);
}
```

The SPI alternate function is *disabled* before the pin is set to an output.

I don't quite understand what's going on here, but I believe that switching the order of these operations would let it work as intended.  Is there a reason to do them in this order?  Or is there some other problem?

dpgeorge, 2022-12-15:
I think that would be fine to switch the order of those operations in `mp_hal_pin_output()`.  That would configure the GPIO to be output before switching the mux from SPI to GPIO.

pdg137, 2022-12-26:
I can trigger the glitch every time by pre-initializing the pin as an input:

```python
from machine import SPI, Pin

sck = Pin(14)
mosi = Pin(15)

sck.init(mode=Pin.IN)
spi = SPI(id=1, baudrate=100000, polarity=0, phase=0, sck=sck, mosi=mosi)
spi.write(bytearray([0xAA]))
sck.init(mode=Pin.OUT, value=0)
sck(1)
```
(with 1k pull-up; I think my last one must have really been 10k)
[image]

Here are similar glitches in the other functions:

```python
from machine import SPI, Pin

sck = Pin(14)
mosi = Pin(15)

sck.init(mode=Pin.OUT, value=1)
spi = SPI(id=1, baudrate=100000, polarity=0, phase=0, sck=sck, mosi=mosi)
spi.write(bytearray([0xAA]))

sck.init(mode=Pin.IN, pull=Pin.PULL_DOWN)
sck.init(mode=Pin.OUT, value=1)
```
(with no pull-up)
[image]

```python
from machine import SPI, Pin

sck = Pin(14)
mosi = Pin(15)

sck.init(mode=Pin.OUT, value=0)
spi = SPI(id=1, baudrate=100000, polarity=1, phase=0, sck=sck, mosi=mosi)
spi.write(bytearray([0xAA]))

sck.init(mode=Pin.OPEN_DRAIN, value=0)
```
(with 10k pull-up)
[image]

That last one is weird though. It looks like `mp_hal_pin_open_drain` overwrites the value argument and always puts a zero to the output:

https://github.com/micropython/micropython/blob/master/ports/rp2/mphalport.h#L108

Is that intentional?

dpgeorge, 2023-01-12:
> Is that intentional?

No, that's a bug.

commit b208cf23 (2023-01-12): rp2/mphalport: Change order of pin operations to prevent glitches.

When switching from a special function like SPI to an input or output,
there was a brief period after the function was disabled but before the
pin's I/O state was configured, in which the state would be poorly defined.
This fixes the problem by switching off the special function after fully
configuring the I/O state.

Fixes #10226.
PR #10337 (2023-01-12): rp2: Change order of pin operations to prevent glitches.
Fixes #10226. When switching from a special function like SPI to an input or output, there was a brief period after the function was disabled but before the pin's I/O state was configured, in which the state would be poorly defined. This fixes the problem by switching off the special function after fully configuring the I/O state.

Test code:
```python
from machine import SPI, Pin

sck = Pin(14)
mosi = Pin(15)

sck.init(mode=Pin.IN)
spi = SPI(id=1, baudrate=100000, polarity=0, phase=0, sck=sck, mosi=mosi)
spi.write(bytearray([0xAA]))
sck [... cut]
closed 2023-01-12 by dpgeorge