micropython/micropython #10248: Pico W - Issue with LED & Wifi Module control

yojoebosolo, 2022-12-16:
When setting pin 23 to high and low during a script (wifi module on and off), the LED control is lost and the script often fails completely.  

Example below will run once, then crash.

`import time
import machine

#setup pins
wifi_pin = machine.Pin(23, machine.Pin.OUT)

def flash_led():
    print("Flashing LED")
    wifi_pin.high()
    time.sleep(2)
    led = machine.Pin("WL_GPIO0", machine.Pin.OUT)
    time.sleep(1)
    print('starting flash')
    led.on() # led on
    print('ON 1')
    time.sleep(0.2)
    led.off() # led off
    print('OFF 1')
    time.sleep(0.2)
    led.on() # led on
    print('ON 2')
    time.sleep(0.2)
    led.off() # led off
    print('OFF 2')
    time.sleep(0.2)
    wifi_pin.low()
    time.sleep(2)
    print('wifi off')

def main():
    while True:
        flash_led() # purely for debugging restarts

main()

`

robert-hh, 2022-12-17:
That is probably related to the fact, that the LED is connected to the WiFi module. So the question is more, what happens at the WiFi module if it is switched on and off in a row.

yojoebosolo, 2022-12-17:
Yes absolutely. I believe that it causes intermittent issues with other pins/functionality also, for example the ADC seems affected and gives sporadic results, however the LED is the easiest example to prove the bug.

robert-hh, 2022-12-17:
Digging a little bit though the code, you should use wlan.active(True) and wlan.active(False) to switch WiFi on and off. That includes the other activities to make the WiFi module's state reliable.
Some effects between ADC and WiFi should have a different reason, because ADC is not driven by the WiFi module.

yojoebosolo, 2022-12-17:
So, i've read that in order to put the Pico into machine.deepsleep, you need to turn off the wifi module (pin 23), I haven't tested this just yet, but that's my thinking behind turning off the pin and not simply deactivating it with wlan.active(False). 

Secondly, in order to read the vsys voltage (with ADC3) you must set "machine.Pin(29, machine.Pin.IN", which also comprises part of the wireless module (communication/SPI). Which causes this weird buggy error. So I think that basically setting any of the following pins to Low/IN (and back to high when needed) will cause the wifi module to break and will crash the code, sometimes entirely 'bricking' the pico until you upload a flash_nuke.uf2.

   ``` 
    machine.Pin(29, machine.Pin.OUT).high() # Must be low to read ADC3 voltage but must be High for wifi to work.
    machine.Pin(25, machine.Pin.OUT).high()
    machine.Pin(24, machine.Pin.OUT).high()
    machine.Pin(23, machine.Pin.OUT).high()
```
    
here is an exert from the datasheet which might explain to someone who understands it better than I do: 

```A few RP2040 GPIO pins are used for internal board functions:
GPIO29 OP/IP wireless SPI CLK/ADC mode (ADC3) to measure VSYS/3
GPIO25 OP wireless SPI CS - when high also enables GPIO29 ADC pin to read VSYS
GPIO24 OP/IP wireless SPI data/IRQ
GPIO23 OP wireless power on signal```

robert-hh, 2022-12-17:
> So, i've read that in order to put the Pico into machine.deepsleep, you need to turn off the wifi module (pin 23), I haven't tested this just yet, but that's my thinking behind turning off the pin and not simply deactivating it with wlan.active(False).

The code behind wlan.active() switches GPIO23 as well, in addition to other housekeeping actions.

The SPI transfer to the CYW43 module is done with the PIO. The GPIO24, GPIO25, and GPIO29 are assigned as being controlled by the PIO. So you cannot use them for other purposes, unless your re-init the WiFI module, which seems to be done by calling wlan.active(True).

yojoebosolo, 2022-12-17:
OK, I've solved it, I think.... 

```
def read_onboard_voltage():
    **pin_29.init(machine.Pin.IN)** # to read the ADC 3 voltage correctly
    time.sleep(1)
    ADC_raw = onboard_voltage_pin.read_u16()
    ADC_voltage = ((ADC_raw * 3) / 65535) * 3.3
    print(f'Got voltage ADC3: {ADC_voltage}')
    print(pin_29)
    **pin_29.init(mode=machine.Pin.ALT, pull=machine.Pin.PULL_DOWN, alt=7)** # to set back correctly for wifi module to work
```

closed 2022-12-17 by yojoebosolo