micropython/micropython #10293: print() slows down program, when VCP is opened once and then closed

andreamerello, 2022-12-21:
After updating from micropython 1.13 to 1.19 I started to sometimes experience huge slow downs in my program (I even thought it was stuck). I've tracked down this to `print()` (and possibly other VCP-related funcion) starting to waste a lot of time under specific conditions. 

This happens on both pyb v1.1 and pyb SF2 (both stm32)

More specifially, this happens when:
- I start the board (I connect it to USB, so it powers up).
- I open the VCP on my PC (i.e. `screen /dev/ttyACM0 115200`).
- I close the VCP (i.e. exit from screen program).
- I wait some time (e.g. 15 seconds); I guess that buffer fills up.

If I re-open the VCP (i.e. I run `screen` agian) the program starts again to run smoothly.

I've reduced the problem to this minimal example:
```
import pyb
import time

l = pyb.LED(1)

while True:
    l.toggle()
    time.sleep(0.1)
    print('Hello')
```

The led blinks normally for about 15 seconds on my pyb SF2 after opening-then-closing the VCP, then it slows down a lot. Note that this does not happen if the VCP is never opened after POR.

I've bisected this to 581d43b (stm32/usbd_cdc_interface: Check and handle CDC TX wrap-overflow.).

I've noticed that before this commit, I experience just a small glitch after - say - 15 seconds after I detach the VCP, but, after that, all continues to work normally.

I may try to investigate more, and eventually issue a PR; any pointer or suggestion is welcome..

dpgeorge, 2022-12-21:
This is somewhat expected behaviour: if the PC is connected to the VCP then the VCP has flow control and will wait for the PC to be ready before it sends it VCP data.  If the PC is very busy then that will block the board / slow it down.

In your case, I think when you exit screen it does not disconnect VCP, so the board thinks the VCP channel is still open (it technically is) and because the PC is not draining the incoming characters the board waits and eventually times out and overflows the VCP buffer.  It does this over and over so that's why it runs slower.

Try to use a different terminal program that cleanly closes the VCP when done.

andreamerello, 2022-12-22:
Ah, while I understood what happens in general, I didn't realize that this could depend by how the VCP is closed. Thanks for pointing this out. Indeed I've tried with several programs and it seems that it really depends by it, as you suggested. More specifically, also manually clearing DTR after "screen" has been closed seems to prevent the issue, so it really seems that this behaviour is due to how the terminal program leaves the control flow signals.

IMO control flow is certainly useful not to overflow the reader while sending out the buffer content (during normal operation, when the buffer is partially filled) but in case the buffer become full, then IMO whether we want to block or not, it may depend by the specific case. I'm not surprised that an output primitive would block in general, but here IMO this makes the device too fragile, considering also that there isn't a preemptive scheduler.

When prints start to fail (i.e. timeout) the application has no way to detect this (unless I'm missing something) and react by giving up trying to print; on the other hand, I would design a task dedicated to manage communication if I had a RTOS (so that other function in other tasks could survive), but here we haven't.

So, indeed I understand your technical point of view, but still I see a real-world issue with the current behaviour. 

(I'm quite new to micropython world, so it is quite possible that there is a way to handle this, that I'm failing to find out; if that's the case, then sorry in advance :) )

robert-hh, 2022-12-22:
Write to VCP does not block forever, as you have noticed. There is a timeout.  Giving up trying to print does not sound like a solution, because then you have to define a way to tell, when printing is possible again.

andreamerello, 2022-12-22:
Yes, this would require something to detect the VCP is alive again. This might be done by checking for flow control lines or simply re-enabling prints just after receiving something (e.g. a newline). 

BTW The timeout at each print, unfortunately, doesn't even mitigate the issue in practice, as the program will stall for seconds every time.

dpgeorge, 2022-12-22:
The problem is, in the case of screen not fully disconnecting the VCP, the board still thinks that the host PC is listening and so tries its best to do flow control so the PC does not lose any incoming data.

> When prints start to fail (i.e. timeout) the application has no way to detect this (unless I'm missing something) and react by giving up trying to print; on the other hand, I would design a task [...]

Could you outline in a bit more detail how you would design it to work if you had an RTOS?  In MicroPython we have asyncio, and also IRQs, so they can be used to do most things that you'd do with an RTOS.

BTW, you can actually disable flow control on stm32 boards with:
```py
pyb.USB_VCP(0).init(flow=0)
```
You can also enable just CTS or RTS flow control by passing in the `pyb.USB_VCP.CTS` and `pyb.USB_VCP.RTS` constants (they can be or'd as well).

andreamerello, 2022-12-23:
> Could you outline in a bit more detail how you would design it to work if you had an RTOS? In MicroPython we have asyncio, and also IRQs, so they can be used to do most things that you'd do with an RTOS.

I could just implement a task (let's call it "foo") that (among other things), reads from a queue and writes to VCP. A helper function (e.g. something like `vcp_print()`) just puts stuff in the queue, with a non blocking primitive; other tasks would use it for their prints. In case the VCP blocks, only the "foo" task would be blocked, while other tasks could go on (and possibly know what is happening (when you put in the queue you know whether this has been done or not) and/or setting proper timeout, in case they are doing something more important than just outputting log messages). 

I'm new to the whole asyncio thing; I recently gave it a look, but my understanding was that it only works with specifically designed "awaitable" primitives - or something like that. Is the VCP primitive (i.e. `print()`) able to yield if I use asyncio? 

Maybe the whole point is here: I'm not aware of proper methods to handle this, so that I'm seeing a problem where there isn't indeed :)

> BTW, you can actually disable flow control on stm32 boards with:
> 
> ```python
> pyb.USB_VCP(0).init(flow=0)
> ```

I found this on the documentation, but it didn't work for me. I thought I cannot re-init the VCP in case it is already bound to the stdout (i.e. micropython uses it as its console), and that it is being (re-)inited when the USB VCP mode is set i.e. I tried to do that both before and after `pyb.usb_mode()`. But maybe I'm missing something...

(I thought also that conceptually I didn't want to disable the flow control indeed, so that little glitches on the receiver side could be tolerated i.e. the pyb sw buffer would fill a bit without overflowing, rather I conceptually wanted just to avoid to block in case that buffer overflows - but I think this is not a real-world issue, because the PC probably have RX buffers big enough not to loose anything, so if I could disable control flow, that would be enough for me).

andreamerello, 2022-12-23:
Thinking about this a bit more: I would say that my point is that IMHO the VCP APIs should either be non-blocking or block yielding.

It looks like they are just stalling the CPU (I see a lot of `RuntimeError: schedule queue full` from several timers) without yielding.

dpgeorge, 2022-12-23:
> I could just implement a task (let's call it "foo") that (among other things), reads from a queue and writes to VCP. A helper function (e.g. something like `vcp_print()`) just puts stuff in the [...]

You could do this in MicroPython without threading:
1. Using asyncio, create an asyncio task that handles copying data from the queue to the VCP object.
2. Without asyncio, have a function `vcp_print()` that writes if the VCP will not block, otherwise saves to a buffer.  Then have a `machine.Timer(-1)` software timer that runs periodically to try and flush any remaining data in the buffer (when the VCP will not block).

If you use the raw `pyb.USB_VCP().write()` method then that can be made non blocking by turning off flow control.  And you can use `select.poll` to poll the `USB_VCP()` object to see if it's writable.  Basically the same way that you'd do it in C.

The built-in `print()` is just a (very) convenient wrapper on all this, which is blocking, and is usually OK for most use cases.  But in cases like yours you may need to make it more sophisticated.

> Is the VCP primitive (i.e. `print()`) able to yield if I use asyncio?

The `USB_VCP` object can be made non blocking and is pollable, so works with asyncio.  `print()` on the other hand is always synchronous and therefore blocking.

> I found this on the documentation, but it didn't work for me. I thought I cannot re-init the VCP in case it is already bound to the stdout

Ah, you're right, it's a bit unintuitive:
- With CTS enabled, USB_VCP.write will only write as much data that will fit in the buffer.  It's fully non-blocking.
- With CTS disabled (the default when USB_VCP is attached to the REPL), it will always fill the buffer and potentially overwrite existing data in the buffer if it's full.  But it will wait up to 500ms for the buffer to be drained, so that's where the blocking behaviour comes from.

And, as mentioned, `print()` sits on top of this and is synchronous and must wait for all data to be written.  So if CTS is enabled then print becomes fully blocking, waiting to write all data (each underlying call to USB_VCP.write is non-blocking, but it sits in a loop waiting to write everything).  When CTS is disabled (the default for REPL) then `print()` will make progress even if the buffer is full.  But that progress is slow (500ms wait for each call to USB_VCP.write).

In summary: if you can't guarantee that the host PC cleanly closes the USB VCP connection then you'll need to implement your own printing function that has the semantics you need (and that's entirely possible in MicroPython using USB_VCP.write and/or polling).

andreamerello, 2022-12-23:
@dpgeorge Thank you a lot for your explanation. With these information I've been able to implement an acceptable workaround (I've just replaced the default print with one that try to poll before invoking the "real" print).

I wish I had found these information before.. Unless I've missed them, I would propose to try to integrate them in the doc. Would be OK for you if I try to do that and eventually issue a PR ?

Since you explained that `print()` indeed behave like this by design, then what I reported is definitely not a bug (I've been also tricked by the fact that this behavior changed after micropython 1.13). Nevertheless I wonder whether an option (somewhere, somehow) to choose whether `print()` should block or not could be of any interest (If this is the case, I may try to hack a bit to see if I could add it. That would be a good occasion for me to start putting my hands under the hood of micropython and try to understand something more about it...)