micropython/micropython #10333: Loop speed, Numba

neilger, 2022-12-26:
Here's RP2040 timing for a minimal benchmark that does a scatter-gather calculation: https://gitlab.cba.mit.edu/pub/pi

2.6 MFlop (Arduino, floats, 250 MHz, -O3)  
0.21 MFlop (MicroPython, 250 MHz)  

Here's another one that tests pin-core timing: https://pub.pages.cba.mit.edu/ring/

9.17 MHz (Arduino, SIO)  
0.58 MHz (Arduino, digitalRead/Write)  
0.039 MHz (MicroPython)  

For the latter result I did try direct memory access for GPIO, but that didn't help -- the limitation was the loop rather than the I/O.

I've seen some discussion of JIT in the issues, but I haven't seen Numba mentioned (https://numba.pydata.org)  -- is there any hope of speeding of critical loops?

robert-hh, 2022-12-26:
Depending on the CPU clock freq and optimization, I get frequencies up to ~300kHz @250Mhz CPU clock with the pin test. Without the native decorator, the toggle speed is about 150 kHz @250Mhz CPU clock.

```
from machine import Pin

@micropython.native
def run():
    p0 = Pin(0, Pin.OUT)
    p1 = Pin(1, Pin.IN)
    while True:
        p0(1 - p1())
 ```

neilger, 2022-12-26:
@robert-hh thanks for pointing out @micropython.native and the pin functions, I had missed both.

The ring test with p2(not p1()) and @micropython.native gives me 350 kHz; without @micropython.native it's 209 kHz.

The pi test sped up less with @micropython.native, from 0.26 to 0.31 MFlop, presumably because of the overhead of the floating point emulation.

I'm going to close this issue because this is the kind of function decorator I was asking about.

closed 2022-12-26 by neilger

robert-hh, 2022-12-26:
There is also the `@micropython.viper` function decorator, which allows for much faster code, but with a very restricted language coverage. The speed of viper code is close to assembler code, and having something like pointers, it allows access to memory (and ports) in one clock cycle. Toggling in the MHz range is possible. I did not make an example, because it is quite different to plain Python. For some information about it, see: https://docs.micropython.org/en/latest/reference/speed_python.html#the-viper-code-emitter

peterhinch, 2022-12-27:
It is also possible to use [inline assembler](http://docs.micropython.org/en/latest/reference/asm_thumb2_index.html). Note that the docs are for Arm Thumb V7 while RP2 is V6. Aside from the lack of FP I have found the differences to be minor.