micropython/micropython #10335: m.iMX RT:  Hanging of very simple benchmark

rkompass, 2022-12-26:
Attempting to run [simple_benchmarks.py ](https://github.com/kevinjwalters/circuitpython-examples/blob/master/benchmarks/simple_benchmarks.py) crashed (hung) MP on my Teensy 4.1 .
 I reduced the program maximally to:
```
import time

def timeit(func, arg):
    t1 = time.ticks_us()
    func(arg)
    t2 = time.ticks_us()
    return time.ticks_diff(t2, t1) / 1e6

def bm_001AdditionA(n):
    for idx in range(n):
        if idx % 50 == 0:
            a = 1.0
            b = 1.1
            c = 3.456789e-4
            d = 5.678901e4
            f = 1.5
        a += 1.01
        b = a + b
        c = c + a + a + a
        d = a + b + 0.9999
        e = a + b + c + d
    return (a, b, c, d, e)

print('Start..', end='')
for _ in range(5):
    timeit(bm_001AdditionA, 50000)
    time.sleep(0.1)                                            # including this leads to a hang
    # print('.', end="")                                       # but including this runs through
print('Finish')
```
and still it hangs.
Strangely including the `print('.', end="")` leads to ordinary run through.
Commenting that statement out still runs through, but as soon as `time.sleep(0.1)` is uncommented it hangs again.
A combination of timing and printing leading to a crash..

For reproducing I installed the latest MP:
MicroPython v1.19.1-782-g699477d12 on 2022-12-20; Teensy 4.1 with MIMXRT1062DVJ6A.

rkompass, 2022-12-27:
```
import time

print('Start..', end='')
time.sleep(5)
print('Finish')
```
It's only a question of program duration. After > 4 seconds the print is hanging.

robert-hh, 2022-12-27:
I cannot replicate it, using both variants
```
MicroPython v1.19.1-785-g56481555e on 2022-12-15; Teensy 4.1 with MIMXRT1062DVJ6A
Type "help()" for more information.
>>> 

```

robert-hh, 2022-12-27:
In the simple test, can you do something that shows, whether the code stops or the print IO, like blinking the LED.

rkompass, 2022-12-27:
I have i.MX RT 1062.
I changed the tests to:
```
import time

def test2():
    print('Start..')
    time.sleep(2)
    print('Finish')

def test5():
    print('Start..')
    time.sleep(5)
    print('Finish')
```
and tried `screen /dev/ACMtty0` to check whether it's a problem related to Thonny.
`test2()` runs and finishes fine.
`test5()` hangs, but after pressing enter after a few more seconds it either completes with printing `Finish` and returns to the `>>>` promt, repeating that promt properly, or it returns to an empty line and a single promt without the `Finish`.

Now with LED:
```
from machine import Pin
import time

p2 = Pin(2, Pin.OUT)

def test2():
    print('Start..')
    p2(1)
    time.sleep(2)
    p2(0)
    print('Finish')

def test5():
    print('Start..')
    p2(1)
    time.sleep(5)
    p2(0)
    print('Finish')
```
`test2()` works as expected, `test5()` switches LED on, and after 5 seconds switches off, but does not print. It reacts to ctrl-D, however, with soft reboot.

rkompass, 2022-12-27:
If i omit the `print('Finish')` it's still the same problem: the program does not return to the REPL.

robert-hh, 2022-12-27:
Could it be an USB problem? I added a Pin toggle into the loop of `bm_001AdditionA(n)`. Each iteration of the loop takes 10µs. But on my Teensy it runs fine and terminates as expected with the final print() statement.
It is a little bit difficult to find the problem if the error does not occur. And even if, It has to happen with a board that I can attach to my Segger DEBUG interface, such that I get information about the actual state. I have two 1052 board which I can use. the 1052 CPU is similar to the 1062, only fewer peripherals. 
With one of these, I got the assert error again, this time at a different place of the VM.
`Start..Assertion '*q < pool->len' failed, at file ../../py/qstr.c:133`
The assertion error is only shown if the firmware was created with the DEBUG=1 option. Otherwise the code simply stalls.

robert-hh, 2022-12-27:
So it seems like I can replicate the error, if I paste a chunk of data into USB before starting the bench test. That would indicate a buffer overflow in the stdin chain (USB data handling). Not clear why the time in the simple test matters.

rkompass, 2022-12-27:
Simply tell me what to try out. Do you have a firmware with debug option for download?
I tried MP1.18 and even the Teensy 4.0 firmware from [https://micropython.org/download/](https://micropython.org/download/) with similar results.
Is there a description how to debug with Segger? I have Segger device somewhere, don't know if it is the right one.

robert-hh, 2022-12-27:
That's interesting about MP1.18. So you went back in the repository. Teensy 4.0 and 4.1 should not be a difference. They use the same MCU, and the ETH phy and the additional interfaces doe not matter here.
The Teensy boards cannot be connected to the Segger devices. The SWCLK and SWDIO signals of the MCU are connected to the Teensy bootloader device. You can compile with the DEBUG option set by

make DEBUG=1 BOARD=TEENSY41 -j5

rkompass, 2022-12-27:
Out of curiosity I tried Circuitpython (Adafruit CircuitPython 7.3.3 on 2022-08-29; Teensy 4.1 with IMXRT1062DVJ6A) and there is no hanging:
```
import board
import digitalio
import time

p2 = digitalio.DigitalInOut(board.D2)
p2.direction = digitalio.Direction.OUTPUT

def test2():
    print('Start..')
    p2.value = True
    time.sleep(2)
    p2.value = False
    print('Finish')

def test5():
    print('Start..')
    p2.value = True
    time.sleep(5)
    p2.value = False
    print('Finish')
```
both test2() and test5() return to the REPL.

rkompass, 2022-12-27:
Perhaps I will compile tonight..

robert-hh, 2022-12-27:
https://github.com/robert-hh/Shared-Stuff/blob/master/firmware_TEENSY41_DEBUG.hex

robert-hh, 2022-12-27:
I can reproduce an error with the following steps:

1. Reboot
2. Open "pye" and paste text into it from the console. It must be more than
   about 500 characters. Close pye. No need to save.
3. Import the file with the benchmark test form a file
4. Run the benchmark test.

It will stop with assert error messages, or silent with a crash window
in the debugger. Sample assert message:
`Assertion 'def_args == NULL || def_args[0] == MP_OBJ_NULL || mp_obj_is_type(def_args[0], &mp_type_tuple)' failed, at file ../../py/emitglue.c:178` or
`Assertion 'exc_sp >= exc_stack' failed, at file ../../py/vm.c:674`
The assert fail places vary.

It will run fine if I either:

a) I call gc.collect() between step 2 and 3. gc.collect() may as well be added
   at the start of the benchmark script or forced in pye with Ctrl-E
b) Paste a smaller amount of data in step 1.
c) swap steps 2 and 3, such that the benchmark script is already imported
   when I do the paste operation.
d) I use the upysh command "newfile" in step 1 or a simple slow input script.
e) Use the RT1010 device for testing, which has a much smaller RAM,such
   that automatic gc.collect happens in step 1.

Test done with the Seeed Arch mix board, which has 32MByte of heap space.
Terminal emulator 'picocom', which unlike Thonny does not send data unless
told so.

No clue what's going on. Pye is a plain Python script, which however stresses
memory management a lot by handling chunks of data with text.

Some mem_info figures after step 1:
```
>>> micropython.mem_info()
stack: 468 out of 23552
GC: total: 32785984, used: 2238672, free: 30547312
 No. of 1-blocks: 74208, 2-blocks: 3638, max blk sz: 64, max free sz: 1909193
```
And after a gc.collect, this changes to:
```
>>> micropython.mem_info()
stack: 468 out of 23552
GC: total: 32785984, used: 9984, free: 32776000
 No. of 1-blocks: 132, 2-blocks: 19, max blk sz: 64, max free sz: 1909182
```

rkompass, 2022-12-27:
Thanks for the debug firmware. Behaviour is exactly the same as above.
What is pye? Can I debug the firmware now? If yes, how?
I found [https://www.youtube.com/watch?v=RRN_ngx01vM](https://www.youtube.com/watch?v=RRN_ngx01vM) and
[https://github.com/micropython/micropython/pull/8767](https://github.com/micropython/micropython/pull/8767)
Can I use pdb with the debug firmware?

robert-hh, 2022-12-28:
pye is an on-board editor, which is frozen into the image I have created. You can import it like any other python script:
from pye import pye
pye("filename")
I used it in my test to stress the memory management. You could use any other code.

Sorry, I do not know how do use external debuggers with Teensy boards. I used boards with an appropriate interface instead and the Ozone debugger. There are people who added software shims to connect to gdb. In that case you would need to create the firmware yourself.

The issue stays strange. It looks like a problem with memory management, since most of the code is from Python core, except of the USB driver with TinyUSB, the timer interrupt and the flash memory interface, which are port-specific.I tried your code with other ports, but these have way less RAM, and as soon as garbage collection is run once before starting you benchmark code, it works - at least here.

Do you still use Thonny for testing?

robert-hh, 2022-12-28:
I have a simple script which mimics some of what pye is doing. It simple gets characters from stdin, echoes them and adds them to a list of lines. Only you have to paste more text into it before the 1-blocks in mem_info get above a certain limit (1000 lines instead of 30), and then your bench test fails.

```
import sys
content = []

while True:
    l=""

    c = sys.stdin.read(1)
    sys.stdout.write(c)
    if c == '\n':
        content.append(l)
        l=""
    elif c == '\x04':
        break
    else:
        l += c
```

robert-hh, 2022-12-28:
Or even more simple the script below, which sorts out USB I/O as culprit:
```
def store(n=1000):
    content = []
    for x in range(n):
        l=""
        for _ in range(50):
            l = l + "#"
        content.append(l)
```
If I run that script before importing and running the benchmark script, MP crashes. Test board: MIMXRT1050_EVKB.
The slightly modified benchmark test script is below. The core is unchanged.
```
import time
import micropython

def timeit(func, arg):
    t1 = time.ticks_us()
    func(arg)
    t2 = time.ticks_us()
    return time.ticks_diff(t2, t1) / 1e6

def bm_001AdditionA(n):
    for idx in range(n):
        if idx % 50 == 0:
            a = 1.0
            b = 1.1
            c = 3.456789e-4
            d = 5.678901e4
            f = 1.5
        a += 1.01
        b = a + b
        c = c + a + a + a
        d = a + b + 0.9999
        e = a + b + c + d
    return (a, b, c, d, e)

def run(n=5):
    t=0
    print('Start..', end='')
    for _ in range(n):
        t += timeit(bm_001AdditionA, 50000)
        micropython.mem_info()
    print('Finish', t/n)
```

robert-hh, 2022-12-28:
And the execution log is:
```
MicroPython v1.19.1-782-g699477d12 on 2022-12-28; i.MX RT1050 EVKB-A1 with MIMXRT1052DVL6B
Type "help()" for more information.
>>> import store
>>> store.store(1000)
stack: 740 out of 23552
GC: total: 32785984, used: 2500992, free: 30284992
 No. of 1-blocks: 63158, 2-blocks: 16026, max blk sz: 264, max free sz: 1892812
>>> import bench
>>> bench.run()
Start..stack: 732 out of 23552
GC: total: 32785984, used: 10503568, free: 22282416
 No. of 1-blocks: 563198, 2-blocks: 16030, max blk sz: 264, max free sz: 1392648
stack: 732 out of 23552
GC: total: 32785984, used: 18503632, free: 14282352
 No. of 1-blocks: 1063200, 2-blocks: 16031, max blk sz: 264, max free sz: 892644
stack: 732 out of 23552
GC: total: 32785984, used: 26503696, free: 6282288
 No. of 1-blocks: 1563202, 2-blocks: 16032, max blk sz: 264, max free sz: 392640
stack: 732 out of 23552
GC: total: 32785984, used: 1726320, free: 31059664
 No. of 1-blocks: 107476, 2-blocks: 24, max blk sz: 64, max free sz: 999895
Assertion 'exc_sp >= exc_stack' failed, at file ../../py/vm.c:674
```

rkompass, 2022-12-28:
Could it be we are talking about two different issues here? The memory problem and the problem of print hanging after about 5 seconds? For the latter only the sleep(5) is necessary. And it hangs not completely: sometimes an <enter> key leads to a reappearing REPL promt. I might try to print to a file or something else to check whether it is the printing or a problem of USB communication? Try to do the REPL through a serial line perhaps? The problem was avoided with print('.') 's  in intervals of about a second. Are there other ways to reactivate the printing, that do not go as far as using the USB line?
```
def test5a():
    p2(1)
    f = 0.1
    for _ in range(1500000):
        f += 1.01
    p2(0)
    print('Finish')
```
hangs too, so there is no need to use a sleep(5) to elicit the problem. It's just the delayed print.

rkompass, 2022-12-28:
[https://github.com/hathach/tinyusb/issues/179](https://github.com/hathach/tinyusb/issues/179) gives me the idea: perhaps I should try to dump the USB? Could it be the USB has to be woken up after a few seconds? And fails?

robert-hh, 2022-12-28:
I never had the a problem with the version, which just sleeps. That's why I asked about Thonny, which may do some background activities. It is surely a good suggestion to log USB. 
Nevertheless, there seems to be a problem with the dynamic memory management and this special use situation reveals it. I had before some crashes like these, but very rare and not reproducible. Other users mentioned them too, so it is not obvious.
In your new test script,  you allocate at least 1.5 million memory objects. So it could produce a memory error. But it does not with my Teensy board here.

robert-hh, 2022-12-28:
b.t.w.  I tried REPL though the serial line, with no change. And the last test setup does not rely at all on USB.

rkompass, 2022-12-28:
I activated REPL over UART like this:
```
from machine import UART
import os
uart = UART(6, baudrate=115200)
uart.write('-- uart works --')
os.dupterm(uart)
```
after connecting with a terminal program to /dev/ttyACM0
and with another terminal to the FTDI USB-serial converter, which was connected to TX6/RX6.
I tried all this with a pyboard 1.1 first.
On the pyboard after `os.dupterm(uart)` both terminals mirrored all inputs and outputs in parallel.
With the teensy after `os.dupterm(uart)` the serial line mirrored all in- and outputs on USB (i.e. /dev/ttyACM0) but not the other way round: What was communicated on serial did not appear on USB.
This is a difference to pyboard, and I have not found any docs as how dupterm should behave at all.
But for our issue that behaviour is helpful:
After passing test2() and test5() (from above, with led and sleep) in, I noticed that
test2() passes on both USB and the serial
test5() passes only on the serial but halts on USB. An <Enter> keypress in USB however returns to the >>> prompt, omitting the final print('Finish').
So I conclude what I observed is a problem of USB communication, possibly connected to TinyUSB, as there were problems in the past: [https://forums.raspberrypi.com/viewtopic.php?t=334842](https://forums.raspberrypi.com/viewtopic.php?t=334842), [https://forums.raspberrypi.com/viewtopic.php?t=306430](https://forums.raspberrypi.com/viewtopic.php?t=306430).
Am I right in the assumption that the RP2040 also uses TinyUSB? If yes: How were the problems solved there?

In another attempt I tried to make a function that takes a few seconds to process without allocating memory.
Here it is:
```
def test5a():
    p2(1)
    f2 = array('d', (0.0 for _ in range(100)))
    print('free:', gc.mem_free())
    for i in range(1699800):  # runs with 1699599 and prints same free memory
        f2[i%100] = 3.5
    p2(0)
    print('free:', gc.mem_free())
    print('Finish')
```
I observe that it is the same critical time below 5 seconds that leads to the hanging. Below that there is no used memory reported.

robert-hh, 2022-12-28:
> With the teensy after os.dupterm(uart) the serial line mirrored all in- and outputs on USB (i.e. /dev/ttyACM0) but not the other way round: What was communicated on serial did not appear on USB.

In my set-up using dupterm with REPL everything that is received or sent appears on both channels. That's the way it's implemented. See mphalport.c, function mp_hal_stdout_tx_strn(). There is a check for tud_cdc_connected() before writing to USB. So if the connection was lost, only write to UART happens. 

The RP2040 and SAMD use TinyUSB as well and are implemented almost identical.

Tested with both a Teensy 4.1 and a RT1052 Dev Board.

P.S.: With the changed test5a script, there are indeed no memory or assert failures, since it does not allocate more RAM. The problem occurred after the sequence:
1. frequent allocation of RAM for growing strings, but not until automatic garbage collection runs.
2. frequent allocation of RAM for floats, such that automatic garbage collections start.
3. then it crashes during one the next attempts to allocate RAM.

2. does not happen with the new test5a().

rkompass, 2022-12-28:
I only noticed now: even when started on USB and the second print hangs there, it is printed on the serial, and there also the REPL prompt returns. This is with test5() and with test5a().

It seems more clear now that there are two different issues: The memory and the USB.
> So if the connection was lost, only write to UART happens.
Given my observation with asymmetric mirroring of in- and outputs then, could this implicate that the cdc connection was lost in the first place? Why then the printing in the REPL and within the first 3 seconds of running program?

I will try to reproduce the memory issue soon...

rkompass, 2022-12-28:
Do you know what MICROPY_EVENT_POLL_HOOK and __WFE() do?
The difference between the RP2040 and MiMXRT versions of mp_hal_stdout_tx_strn is that RP2040 executes
MICROPY_EVENT_POLL_HOOK `while (n > tud_cdc_write_available())` and the  MiMXRT  __WFE() .

robert-hh, 2022-12-29:
It may be a USB connection issue. Which PC OS do you run? Is the device directly connected to the PC or though a USB hub. Is a modem manager active on the PC?

> Do you know what MICROPY_EVENT_POLL_HOOK and __WFE() do?

MICROPY_EVENT_POLL_HOOK is a macro defined in mpconfigport.h. It is usually called in wait loops and checks for tasks to be scheduled. One of the instructions in MICROPY_EVENT_POLL_HOOK is usually either the __WFE() or __WFI() instruction. Both are ARM instructions and set the CPU to low power mode until an event or interrupt happens, which is at least the timer tick interrupt every 1 ms. You can drop __WFE() for test, if you like. But that should not make a difference, because is the code hangs there, the serial print will either be delayed or not happen.

For the memory issue I'll open a separate issue item.

rkompass, 2022-12-29:
You are right. With my sons laptop and Win10 there is no problem with `test5()`.
But somewhere there must be a difference in USB communication, as pyboard and RP2 had no problem executing this on my NUN running Manjaro Linux,

rkompass, 2022-12-29:
The above mentioned [simple_benchmarks.py ](https://github.com/kevinjwalters/circuitpython-examples/blob/master/benchmarks/simple_benchmarks.py) also runs to completion on WIn10. After adding "mimxrt" to the list of MP platforms.

robert-hh, 2022-12-29:
> But somewhere there must be a difference in USB communication, as pyboard and RP2 had no problem executing this on my NUC running Manjaro Linux. I have a problem with RP2 boards and the USB hub that I use. The rp2 boards register either not at all, or 40 seconds after plugging them into the hub, or after unplugging and re-plugging the hub.

No clue. But the boards have different VID and PID numbers:
```
Teensy: ID f055:9802 MicroPython Teensy 4.1
RP2:    ID 2e8a:0005 MicroPython Board in FS mode
SAMD51: ID f055:9802 MicroPython Board in FS mode
Pyboard:ID f055:9801 MicroPython Pyboard Virtual Comm Port in FS Mode
NRF:    ID f055:9802 MicroPython Board in FS mode
```

robert-hh, 2022-12-31:
The initial question seems to be settled as being related to the PC connection. So this topic can either be closed or changed to a discussion item.

rkompass, 2022-12-31:
I still think it's an issue, given that it occurs only on the MP port to mimxrt. And it should be solved somehow, by an expert in USB and TinyUSB perhaps. And we seem to have a policy that allows many issues to remain open (?!).
I would suggest to delete our lengthy conversations and just give a short summary.

rkompass, 2022-12-31:
Btw. is there a way to "ping" the USB connection without sending actual data? This being done in the background could serve as a workaround.
Sending something with print('.') every second had avoided the stalling.

robert-hh, 2022-12-31:
As far as I can tell, nothing that is available with a Python API.  May I repeat one of my earlier questions: do you have a modem manager active on your PC? These may interfere with the the ttyACMx devices. Or is there some kind of power saving active in your PC which shuts down USB when not used?

rkompass, 2022-12-31:
USB autosuspend was the culprit on my PCs side. I followed [https://forum.manjaro.org/t/how-to-disable-at-boot-autosuspend-usb-port/70283](https://forum.manjaro.org/t/how-to-disable-at-boot-autosuspend-usb-port/70283) and the problem now is gone.
Anyway I will google to find something how TinyUSB should handle autosuspend. In the new year. And if yes, report in discussions.
Thank you for insisting on that possibility. Have a happy new year.

closed 2022-12-31 by rkompass

robert-hh, 2022-12-31:
It's good that we have a solution. If you have a result for TinyUSB, report it as well in the TinyUSB repository. The maintainer there answers quite fast. And he may have knowledge about USB and autosuspend.

rkompass, 2022-12-31:
It occurs to me as if this is the corresponding circuitpython issue: [https://github.com/adafruit/circuitpython/pull/1737](https://github.com/adafruit/circuitpython/pull/1737). Does MP has this incorporated?
With circuitpython I din't have the problems...

robert-hh, 2022-12-31:
This PR adds two callback functions, which are empty. I could add them to mphalport.c, but that may not change anything. The callback functions are usually implemented in the TinyUSB lib as well with a default action and declared as weak links, so they can be overriden.

rkompass, 2022-12-31:
Seeing [https://github.com/adafruit/usb_descriptor/pull/7/files](https://github.com/adafruit/usb_descriptor/pull/7/files)
I wonder if just bit 5 of USB descriptor has to be set? But I'm lost searching for the place where that might happen.

robert-hh, 2022-12-31:
There is nothing like that in the MP files. Just tusb_config.h, and a call to tusb_init(). I made a firmware_TEENSY_CB.hex.
https://github.com/robert-hh/Shared-Stuff/blob/master/firmware_TEENSY_CB.hex
So you can try if that makes a difference.

rkompass, 2022-12-31:
Makes no difference. Should I ask dhalbert?

robert-hh, 2023-01-01:
> Makes no difference. 

Which was expected.

> Should I ask dhalbert?

Please do so.