micropython/micropython #10354: https not working for large amount of json data

abuvanth, 2022-12-28:
following code is working fine

```
import urequests as rt

url = "https://testme.free.beeceptor.com"

j = {"name":"abu"}

rt.post(url, json=j)
```

following code is not working

```
import urequests as rt

url = "https://testme.free.beeceptor.com"

j = {"hash": "JyuDWsdKFNbg-XoTcviJUWbkDwCXjJVKTOMuYUiZBAk", "sigs": [{"sig": "55b1a254cf98c10be27f13300b4227bf30f616129816b2e0a9d96fedf4ab383c6d3728501e054dc6976dbba6b57db2f12d08a9bd4554e2391fbab3abc0f17800"}], "cmd": "{\"payload\": {\"exec\": {\"code\": \"(free.cyberfly_devices.get-device \\\"f1063aee-7b0a-439c-9697-000770436977\\\")\", \"data\": {}}}, \"networkId\": \"testnet04\", \"signers\": [{\"pubKey\": \"d04bbd8f403e583248aa461896bd7518113f89b85c98f3d9596bbfbf30df0bcb\"}], \"meta\": {\"gasPrice\": 9.999998e-07, \"ttl\": 28800, \"sender\": \"not real\", \"creationTime\": 1671918855, \"gasLimit\": 80000, \"chainId\": \"1\"}, \"nonce\": \"1671918855\"}"}

rt.post(url, json=j)
```

both are working fine in http instead of https.

so that is related to https issue.

[image]

abuvanth, 2023-01-28:
this issue only reproducible in pico port

abuvanth, 2023-01-28:
working fine with esp32

dpgeorge, 2023-01-29:
Trying this out I don't see any issue.  Using the rp2 firmware at `v1.19.1-841-g3446d440f` all requests (http and https, short and long json above) return the same thing:
```
'\nHey ya! Great to see you here. Btw, nothing is configured for this request path. Create a rule and start building a mock API.\n'
```

abuvanth, 2023-01-29:
I think i am using old micropython codebase. need to pull the code.
closing the issue

closed 2023-01-29 by abuvanth