micropython/micropython #10366: mimxrt: Memory corruption at devices with large RAM.

robert-hh, 2022-12-29:
When executing two simple scripts one after the other, the second script fails. The error state differs, sometimes an assertion fault at different places (when compiled with DEBUG=1), sometimes 'just' a hard fault.  The sequence to cause the crash is:

1. Import the first script (strings.py). A copy of the script is below.
2. Run the first script with an argument e.g. >500.
3. Import the second script (numbers.py).
4. Run the second script. It will fail after a few iterations, either with an error message, or the board will just freeze (internal hard fault).

The sequence will run fine, if:
- Steps 2. and 3 are swapped: both scripts are imported at the same time and then executed one after the other.
- After a during step 2 a garbage collection is performed, either manually or because a gc.threshold() was set to e.g. 1 MByte.
- The argument in step is small (like 200), although that is not fully confirmed.

I can reproduce the phenomenon reliably on boards with a LARGE heap, like Seeed Arch Mix and the NXP Dev boards for MX1020, MX1052 and MX1170, which all provide a heap of >= 32MByte. I recall having seen this with a Teensy as well, but cannot reproduce it again. The test sequence touches almost nothing port specific, besides e.g. access to the flash during import.

Script one, strings.py:
```py
# store growing strings into a list

from micropython import mem_info

def run(n=1000):
    content = []
    for x in range(n):
        l=""
        for _ in range(50):
            l = l + "#"
        content.append(l)
    mem_info()
```
Script 2, numbers.py:
```py
# Perform a lot of float calculations

import micropython

def bm_001AdditionA(n):
    for idx in range(n):
        if idx % 50 == 0:
            a = 1.0
            b = 1.1
            c = 3.456789e-4
            d = 5.678901e4
            f = 1.5
        a += 1.01
        b = a + b
        c = c + a + a + a
        d = a + b + 0.9999
        e = a + b + c + d
    return (a, b, c, d, e)

def run(n=10):
    for _ in range(n):
        bm_001AdditionA(50000)
        micropython.mem_info()
```

Typical execution log:
```

MicroPython v1.19.1-782-g699477d12-dirty on 2022-12-29; Seeed ARCH MIX with MIMXRT1052DVL5B
Type "help()" for more information.
>>> import strings
>>> strings.run(500)
stack: 740 out of 23552
GC: total: 32785984, used: 1258928, free: 31527056
 No. of 1-blocks: 31658, 2-blocks: 8025, max blk sz: 264, max free sz: 1970441
>>> import numbers
>>> numbers.run()
stack: 716 out of 23552
GC: total: 32785984, used: 9261184, free: 23524800
 No. of 1-blocks: 531694, 2-blocks: 8029, max blk sz: 264, max free sz: 1470295
stack: 716 out of 23552
GC: total: 32785984, used: 17261216, free: 15524768
 No. of 1-blocks: 1031694, 2-blocks: 8030, max blk sz: 264, max free sz: 970293
stack: 716 out of 23552
GC: total: 32785984, used: 25261248, free: 7524736
 No. of 1-blocks: 1531694, 2-blocks: 8031, max blk sz: 264, max free sz: 470291
stack: 716 out of 23552
GC: total: 32785984, used: 483824, free: 32302160
 No. of 1-blocks: 29816, 2-blocks: 24, max blk sz: 64, max free sz: 1970248
Assertion 'def_args == NULL || def_args[1] == MP_OBJ_NULL || mp_obj_is_type(def_args[1], &mp_type_dict)' failed, at file ../../py/emitglue.c:181

```

rkompass, 2022-12-29:
On the teensy 4.1 with your binary  strings.run(510) is about the maximum that's possible: 520 already gives:
_MemoryError: memory allocation failed, allocating 4096 bytes_
After that numbers.run() completes.

robert-hh, 2022-12-29:
That's a controlled stop and not the problem I report here.

robert-hh, 2023-01-10:
Some more observations:
- At step 4 it is sufficient to call gc.collect() after importing numbers.py to let the board fail. Running the numbers.py script is not needed.
- A lot of RAM is not sufficient to cause the error. No ESP32 with SPIRAM failed in my tests, only mimxrt devices with SRAM, which starts at address 0x8000_0000.
- It does not make  a difference whether the numbers.py script is pre-compiled or not.

robert-hh, 2023-01-12:
One more observations:
Using the ARCH MIX board and the actual set-up, the limit for step 2 (run the strings script) to pass step for is 414. Using 415 cycles the board fails in step 4. mem_info(True) shows, that at 414 the highest used address is 0xff490 after step 2, while for 415 cycles it is 0xffe50. So there may be an issue with the 1MByte limit.

robert-hh, 2023-01-12:
I found the reason. It was the setting of  MICROPY_GC_STACK_ENTRY_TYPE as uint16_t in mpconfigboard.h. Without that line, the default of size_t is used, which is uint32_t. I'll put it into the next SP.

closed 2023-01-12 by robert-hh